ros/meta-ros #520: Build issues with gcc-7.1

bulwahn, 2017-06-16:
The following recipes currently fail compiling with the gcc-7.1 compiler, which is the default compiler since commit 03bb12008891cf1a023aaddb6547da6d41d0cab0 in openembedded-core [1]:

[ ] ecl-containers_0.61.17
[ ] amcl_1.12.13
[ ] assimp_3.0.1270.bb
[ ] sophus_0.9.1.bb

If you use meta-ros with a commit later than 03bb12008891cf1a023aaddb6547da6d41d0cab0, you can simply change to gcc-6 by adding `GCCVERSION = 6%` to the local.conf.

[1] http://cgit.openembedded.org/openembedded-core/commit/?id=03bb12008891cf1a023aaddb6547da6d41d0cab0

rojkov, 2017-06-16:
Submitted a fix [1] to amcl upstream.

[1] https://github.com/ros-planning/navigation/pull/587

bulwahn, 2017-06-16:
@rojkov I had fixed the amcl issue also, but did not commit it as quickly as you did. Thanks.

bulwahn, 2017-06-16:
For sophus, I found out that it fails with the current eigen library:
```
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/sophus/0.9.1-r0/recipe-sysroot/usr/include/eigen3/Eigen/src/Core/Assign.h:53:57: error: enum constant in boolean context [-Werror=int-in-bool-context]
|      MayLinearVectorize = MightVectorize && MayLinearize && DstHasDirectAccess
|                           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/sophus/0.9.1-r0/recipe-sysroot/usr/include/eigen3/Eigen/src/Core/Assign.h:57:41: error: enum constant in boolean context [-Werror=int-in-bool-context]
|      MaySliceVectorize  = MightVectorize && DstHasDirectAccess
|                           ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
| cc1plus: all warnings being treated as errors
```

There are two possible solutions:
1. Either this could already be solved in a recent version of Eigen, so this would be resolved by an eigen recipe update,
2. or we change the sophus recipe to reduce the int-in-bool-context error to a warning through adding an extra CC compiler flag in the recipe.

The first solution (update to the eigen recipe) is currently blocked by the fact that the updated version fails with the current cmake version, as it has now included a `find_host_program` macro that does not work in the openembedded-core tool chain.

rojkov, 2017-06-16:
`find_host_program` is only a part of the problem. The latest release of libeigen also has problems with defining traits with `enum` instead of `static const bool`. I've left a message to their mailing list, but it doesn't seem to go to the mailing list's archive. I think the only realistic solution at the moment is to disable the warning for sophus.

rojkov, 2017-06-19:
The discussion in libeigen's ML:
https://listengine.tuxfamily.org/lists.tuxfamily.org/eigen/2017/06/msg00009.html

PR #523 (2017-06-27): Fix gcc v7.1 compilation issues
resolves #520
commit 4f863ffa (2017-06-27): ecl-containers: fix gcc7.1 compilation issue

This commit resolves #520
closed 2017-06-27 by bulwahn