ros/meta-ros #522: ompl with c++11 and gcc6 problem

yangjiel, 2017-06-22:
The ompl fails to compile due to the c++11 problem and gcc6 problem in morty branch. I have extra patch for it, please have a look.

[0001-address-gcc6-build-error2.txt](https://github.com/bmwcarit/meta-ros/files/1096148/0001-address-gcc6-build-error2.txt)

yangjiel, 2017-06-23:
Actually there is one more bug behind it, the flann issue. I fixed all of them into one additional patch. The file name is not good and it is not from an actual commit. These are the link for the c++11 and flann issues. 
https://bitbucket.org/ompl/ompl/issues/206/cannot-compile-as-c-11
https://bitbucket.org/ompl/ompl/issues/221/cant-get-ompl-update_bindings-work

I know there is a gcc6 patch in your repo but that does not fix the problem. I am getting gcc6 problem in some machine.
[0001-address-gcc6-and-c++11-error.txt](https://github.com/bmwcarit/meta-ros/files/1096464/0001-address-gcc6-and-c.11-error.txt)

bulwahn, 2017-06-23:
I am wondering if we already addressed the reported issue, or why we did not detect this issue before with our testing routine.
Could you please provide the build configuration you are using?

I quickly checked that `moveit-planners-ompl` depends on `ompl`, and `moveit-planners-ompl` is part of `packagegroup-ros-world`. So our usual routine of building `core-image-ros-world` should detect compile issues in ompl, and we use gcc-6 for quite some time now (at least a couple of months, possibly even a year or more).

yangjiel, 2017-06-23:
I would like to provide you more detail. The attached files are the log of the buildbot machine, which is force compiling all recipes. It should provide environment and error information. the stdio contains the gcc6 problem and stdio2 contains c++11 problem. 

These problems are not exploded under normal environment. I guess the buildbot compile strictly or is forcing it to compile causing the problems.
[stdio.txt](https://github.com/bmwcarit/meta-ros/files/1098942/stdio.txt)
[stdio2.txt](https://github.com/bmwcarit/meta-ros/files/1098943/stdio2.txt)

yangjiel, 2017-06-25:
The cv-bridge seems not working as well. Here is the error:
...
| -- catkin 0.6.19
| -- Boost version: 1.61.0
| CMake Error at /home/gumstix/Buildbot/yocto-overo-morty/build/build/tmp/sysroots/overo/usr/share/OpenCV/OpenCVConfig.cmake:86 (include):
|   include could not find load file:
| 
|     /home/gumstix/Buildbot/yocto-overo-morty/build/build/tmp/sysroots/overo/usr/share/OpenCV/OpenCVModules_.cmake
| Call Stack (most recent call first):
|   CMakeLists.txt:16 (find_package)
...
I appreciate your help.

bulwahn, 2017-06-27:
@yangjiel There are quite a number of build environments and tools that use openembedded-core and poky as a basis and then build more scripts around that cross-compiling toolchain. Grasping your build logs, we have not come across the specific build environment you are using. Could you please explain how you create your whole build environment and point to more instructions on that specific build environment and the tools you use? Otherwise, we have probably little chance of re-producing the issue.

yangjiel, 2017-06-28:
We are using the yocto bitbake build engine to build the kernel (morty yocto linux) with necessary recipes like normal and the buildbot is no difference with the desktop. I believe everything we set is following the yocto standard. As I am responsible for maintaining the repository not building the environment, I couldn't provide you much information about the detail. If there is a way to check, you can ask me to do so. Sorry about that.

bulwahn, 2017-07-02:
I started a small testing campaign and build all recipes in meta-ros with the openembedded-core master and the poky pyro release. This already uncovered a number of failing recipes; hopefully, I can continue this testing campaign on the poky morty release in course of this week. Then, I should be able to reproduce the issue you are reporting.

bulwahn, 2017-07-09:
I can reproduce two build errors in cv-bridge and ompl on the poky morty release on this build configurations:
```
Build Configuration:
BB_VERSION        = "1.32.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal-4.8"
TARGET_SYS        = "i586-poky-linux"
MACHINE           = "qemux86"
DISTRO            = "poky"
DISTRO_VERSION    = "2.2.2"
TUNE_FEATURES     = "m32 i586"
TARGET_FPU        = ""
meta              
meta-poky         
meta-yocto-bsp    = "morty:fd7f2d644a94bbb951c0bed11af7deb35f1cfb22"
meta-oe           
meta-python       
meta-multimedia   = "morty:fe5c83312de11e80b85680ef237f8acb04b4b26e"
meta-ros          = "master:08516c36c67bf0edbc01073cb09eb41fc443fedb"
```
@yangjiel At first glance, these build errors are most probably those that you reported. Now that I can reproduce them, I am enthusiastic about reviewing and merging the patches that you proposed and which will resolve those build errors. Please open a pull request, and I will test, review, discuss and merge them. Thanks a lot for your help.

yangjiel, 2017-07-14:
@bulwahn Could you please test the vlc package as well under the same setup as you did before. There is a dependency problem on my build. It requires the libvdpau package but no recipe is providing it. If you have the same error I will try to fix that as well.

bulwahn, 2017-07-15:
@yangjiel I am on vacation and I don't have access to my build machine. But I am wondering if the issue is at all related to the meta-ros layer or if it is a general dependency problem in the poky morty release. If it is the latter, it is probably better a question to the yocto mailing list. If it is the former, you need to explain which package in meta-ros you try to build, why you need it to depend on vlc or if the build process just happens to use vlc although you don't want to use vlc.

yangjiel, 2017-07-17:
I apologize for the inaccurate information in my last post, the vlc package is from meta-openembedded, not thing to do with meta-ros. I will report the issue to them.

closed 2017-07-19 by yangjiel