ros/meta-ros #524: vision-opencv: compile error `'__fp16' does not name a type` on qemuarm

bulwahn, 2017-07-02:
On the qemuarm machine, both vision-opencv recipes, cv-bridge and image-geometry, fail with this compile error:

```
| cd /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/build/src && /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/recipe-sysroot-native/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-g++  -DPYTHON3 -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"cv_bridge\" -Dcv_bridge_EXPORTS -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/vision_opencv-1.11.15/cv_bridge/include -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/recipe-sysroot/usr/include -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/recipe-sysroot/usr/include/opencv -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/recipe-sysroot/opt/ros/indigo/include -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/vision_opencv-1.11.15/cv_bridge/src/. -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/recipe-sysroot/usr/include/python3.5m -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/recipe-sysroot-native/usr/lib/python3.5/site-packages/numpy/core/include  -march=armv5e -marm  --sysroot=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0=/usr/src/debug/cv-bridge/1.11.15-r0 -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/recipe-sysroot=  -fvisibility-inlines-hidden  -march=armv5e -marm  --sysroot=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/recipe-sysroot -fPIC   -o CMakeFiles/cv_bridge.dir/rgb_colors.cpp.o -c /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/vision_opencv-1.11.15/cv_bridge/src/rgb_colors.cpp
| In file included from /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/recipe-sysroot/usr/include/opencv2/core.hpp:52:0,
|                  from /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/recipe-sysroot/usr/include/opencv2/opencv.hpp:52,
|                  from /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/vision_opencv-1.11.15/cv_bridge/include/cv_bridge/rgb_colors.h:39,
|                  from /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/vision_opencv-1.11.15/cv_bridge/src/rgb_colors.cpp:36:
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/recipe-sysroot/usr/include/opencv2/core/cvdef.h:316:5: error: '__fp16' does not name a type
|      __fp16 h;
|      ^~~~~~
| In file included from /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/recipe-sysroot/usr/include/opencv2/core.hpp:52:0,
|                  from /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/recipe-sysroot/usr/include/opencv2/imgproc.hpp:46,
|                  from /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/recipe-sysroot/usr/include/opencv2/imgproc/imgproc.hpp:48,
|                  from /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/vision_opencv-1.11.15/cv_bridge/src/cv_bridge.cpp:43:
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/recipe-sysroot/usr/include/opencv2/core/cvdef.h:316:5: error: '__fp16' does not name a type; did you mean '__fmal'?
|      __fp16 h;
|      ^~~~~~
|      __fmal
| make[2]: *** [src/CMakeFiles/cv_bridge.dir/rgb_colors.cpp.o] Error 1
| make[2]: *** Waiting for unfinished jobs....
| make[2]: *** [src/CMakeFiles/cv_bridge.dir/cv_bridge.cpp.o] Error 1
| make[2]: Leaving directory `/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/build'
| make[1]: *** [src/CMakeFiles/cv_bridge.dir/all] Error 2
| make[1]: Leaving directory `/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/build'
| make: *** [all] Error 2
| WARNING: exit code 2 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/cv-bridge/1.11.15-r0/temp/log.do_compile.29890)
ERROR: Task (/home/lukas/work/repositories/github.com/bmwcarit/meta-ros/recipes-ros/vision-opencv/cv-bridge_1.11.15.bb:do_compile) failed with exit code '1'
```

```
| [ 66%] Building CXX object CMakeFiles/image_geometry.dir/src/stereo_camera_model.cpp.o
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/image-geometry/1.11.15-r0/recipe-sysroot-native/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-g++  -Dimage_geometry_EXPORTS -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/image-geometry/1.11.15-r0/recipe-sysroot/opt/ros/indigo/include -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/image-geometry/1.11.15-r0/recipe-sysroot/usr/include -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/image-geometry/1.11.15-r0/recipe-sysroot/usr/include/opencv -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/image-geometry/1.11.15-r0/vision_opencv-1.11.15/image_geometry/include  -march=armv5e -marm  --sysroot=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/image-geometry/1.11.15-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/image-geometry/1.11.15-r0=/usr/src/debug/image-geometry/1.11.15-r0 -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/image-geometry/1.11.15-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/image-geometry/1.11.15-r0/recipe-sysroot=  -fvisibility-inlines-hidden  -march=armv5e -marm  --sysroot=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/image-geometry/1.11.15-r0/recipe-sysroot -fPIC   -o CMakeFiles/image_geometry.dir/src/stereo_camera_model.cpp.o -c /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/image-geometry/1.11.15-r0/vision_opencv-1.11.15/image_geometry/s [... cut ...]
NATIVELSBSTRING   = "universal-4.8"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "qemuarm"
DISTRO            = "poky"
DISTRO_VERSION    = "2.3"
TUNE_FEATURES     = "arm armv5 thumb dsp"
TARGET_FPU        = "soft"
meta              
meta-poky         
meta-yocto-bsp    = "pyro:f0d128ea0dfc2c403ff53a1ac1db3521854b63d5"
meta-oe           
meta-python       
meta-multimedia   = "pyro:5e82995148a2844c6f483ae5ddd1438d87ea9fb7"
meta-ros          = "master:b0bf3da6d5543234bda7c87b1ea93f323e8ac469"
```

bulwahn, 2017-09-13:
The issue is caused by the opencv recipe. I did a further analysis and testing to pinpoint where in the git history of meta-openembedded, the problem was introduced and were it was resolved.

It was introduced with, i.e., changed from no error to this error in, 51fbd3cf7db14399bc1c69c4a5fd63f157fd209d, and it was resolved with, i.e., changed from error to no error in, 31fceb560c0b81cff3aa439e3f43fd7a04dc477f.

In the current upstream meta-openembedded repository, this issue is already resolved and needs no further action.
However, it would good if the poky branch provides a resolution to this issue as well. Probably, it only requires to backport a specific patch from the opencv upstream development that ultimately resolved this issue in the later version.

herb-kuta-lge, 2019-06-28:
@bulwahn < I was hoping not to have to support OpenEmbedded releases prior to thud. If this is OK, please close this issue.

shr-project, 2019-07-31:
I'm using this work around in recipes which are built against opencv-3.2:

# To prevent issues with opencv-3.2 as described in:
# https://github.com/bmwcarit/meta-ros/issues/524
# https://github.com/opencv/opencv/pull/7294
# https://github.com/OSSystems/meta-gstreamer1.0/issues/11
CXXFLAGS_append_arm = " -mfp16-format=ieee"

closed 2019-11-04 by herb-kuta-lge