ros/meta-ros #525: mavros: "undefined reference to `__atomic_{load,store}_8'" while linking on qemuarm, qemumips, qemuppc

bulwahn, 2017-07-02:
On the current openembedded-core master with qemuarm and on poky pyro with qemuarm, qemumips and qemuppc, the mavros recipe fails with:
```
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot-native/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-g++   -march=armv5e -marm  --sysroot=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0=/usr/src/debug/mavros/0.18.7-r0 -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot=  -fvisibility-inlines-hidden  -march=armv5e -marm  --sysroot=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot -std=c++14  -march=armv5e -marm  --sysroot=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0=/usr/src/debug/mavros/0.18.7-r0 -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot=  -fvisibility-inlines-hidden  -march=armv5e -marm  --sysroot=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -rdynamic CMakeFiles/mavros_node.dir/src/mavros_node.cpp.o  -o devel/lib/mavros/mavros_node -Wl,-rpath,/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/build/devel/lib:/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot/opt/ros/indigo/lib: devel/lib/libmavros.so /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot/opt/ros/indigo/lib/libclass_loader.so -lPocoFoundation -ldl /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot/opt/ros/indigo/lib/libroslib.so /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot/opt/ros/indigo/lib/librospack.so -lpython3.5m -lboost_program_options-mt -ltinyxml /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot/opt/ros/indigo/lib/libtf2_ros.so /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot/opt/ros/indigo/lib/libactionlib.so /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot/opt/ros/indigo/lib/libmessage_filters.so /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot/opt/ros/indigo/lib/libroscpp.so -lpthread -lboost_signals-mt -lboost_filesystem-mt /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot/opt/ros/indigo/lib/libxmlrpcpp.so /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot/opt/ros/indigo/lib/libtf2.so /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot/opt/ros/indigo/lib/libmavconn.so /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot/opt/ros/indigo/lib/librosconsole_bridge.so /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot/opt/ros/indigo/lib/librosconsole.so /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot/opt/ros/indigo/lib/librosconsole_log4cxx.so /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot/opt/ros/indigo/lib/librosconsole_backend_interface.so -llog4cxx -lboost_regex-mt /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot/opt/ros/indigo/lib/libeigen_conversions.so /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot/usr/lib/liborocos-kdl.so.1.3.0 /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot/opt/ros/indigo/lib/libroscpp_serialization.so /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot/opt/ros/indigo/lib/librostime.so /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot/opt/ros/indigo/lib/libcpp_common.so -lboost_system-mt -lboost_thread-mt -lboost_chrono-mt -lboost_date_time-mt -lboost_atomic-mt -lconsole_bridge -lboost_system-mt -lboost_thread-mt -lboost_chrono-mt -lboost_date_time-mt -lboost_atomic-mt -lconsole_bridge
| devel/lib/libmavros.so: undefined reference to `__atomic_load_8'
| devel/lib/libmavros.so: undefined reference to `__atomic_store_8'
| collect2: error: ld returned 1 exit status
| make[2]: *** [devel/lib/mavros/gcs_bridge] Error 1
| make[2]: Leaving directory `/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/build'
| make[1]: *** [CMakeFiles/gcs_bridge.dir/all] Error 2
| make[1]: *** Waiting for unfinished jobs....
| devel/lib/libmavros.so: undefined reference to `__atomic_load_8'
| devel/lib/libmavros.so: undefined reference to `__atomic_store_8'
| collect2: error: ld returned 1 exit status
| make[2]: *** [devel/lib/mavros/mavros_node] Error 1
| make[2]: Leaving directory `/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/build'
| make[1]: *** [CMakeFiles/mavros_node.dir/all] Error 2
| make[2]: Entering directory `/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/build'
| cd /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/build && /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/recipe-sysroot-native/usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/mavros-0.18.7/mavros /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/mavros-0.18.7/mavros /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux-gnueabi/mavros/0.18.7-r0/build /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/armv5e-oe-linux [... cut ...]
NATIVELSBSTRING   = "universal-4.8"
TARGET_SYS        = "powerpc-poky-linux"
MACHINE           = "qemuppc"
DISTRO            = "poky"
DISTRO_VERSION    = "2.3"
TUNE_FEATURES     = "m32 fpu-hard ppc7400 altivec"
TARGET_FPU        = ""
meta              
meta-poky         
meta-yocto-bsp    = "pyro:f0d128ea0dfc2c403ff53a1ac1db3521854b63d5"
meta-oe           
meta-python       
meta-multimedia   = "pyro:5e82995148a2844c6f483ae5ddd1438d87ea9fb7"
meta-ros          = "master:b0bf3da6d5543234bda7c87b1ea93f323e8ac469"
```

bulwahn, 2017-07-02:
@TSC21 Would you have time to look into this issue? Possibly you already resolved this issue upstream and we must just apply a patch from the upstream repository?

TSC21, 2017-07-03:
@bulwahn it's the first time I'm seeing this problem. Most probable is a GCC compatibility issue. I will recheck the current recipe but I believe that the more we upgrade MAVROS and don't keep up with the ROS version, more problems will come.

bulwahn, 2017-08-25:
I did some research and found out that:
For arm & mips architecture, the linker must explicitly be asked to link the atomic library (with `-latomic`).

Some discussions that point out the issue are:
  - https://github.com/PowerDNS/pdns/issues/4171
  - https://github.com/android-ndk/ndk/issues/104
  - https://mail.openvswitch.org/pipermail/ovs-discuss/2015-April/037233.html

More discussions on this error can be found by googling for `undefined reference to `__atomic_load_8'`.

I have a patch that resolves the issue and will provide that upstream at the mavros repository soon, discuss it, and then finally also provide a pull request here in meta-ros.

commit 25d97e4b (2017-09-05): mavros: also compile with arm, mips & ppc (resolves #525)
commit 859f5da5 (2017-09-05): Merge pull request #544 from bulwahn/assimp

resolve #525 and #526
closed 2017-09-05 by bulwahn