ros/meta-ros #527: Build status on poky pyro release

bulwahn, 2017-07-02:
Similar to #443, I have tested `bitbake core-image-ros-world -k` on the current poky
pyro release with all qemu machines in that yocto release.

The common target-machine independent build configuration is:
```
BB_VERSION        = "1.34.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal-4.8"
DISTRO            = "poky"
DISTRO_VERSION    = "2.3"
meta              
meta-poky         
meta-yocto-bsp    = "pyro:f0d128ea0dfc2c403ff53a1ac1db3521854b63d5"
meta-oe           
meta-python       
meta-multimedia   = "pyro:5e82995148a2844c6f483ae5ddd1438d87ea9fb7"
meta-ros          = "master:b0bf3da6d5543234bda7c87b1ea93f323e8ac469"
```

There are some errors in the recipes for some of those machines. The individual errors are reported in separate issues. Here, we just collect the list of issues:

- `qemux86`: No build errors.
- `qemux86-64`: `sophus` fails (#497).
- `qemuarm`: `cv-bridge` and `image-geometry` fail due to a common issue with the opencv header (#524).
- `qemuarm64`: No build errors.
- `qemumips`: `assimp` fails (#526) and `mavros` fails (#525)
- `qemumips64`: `assimp` fails (#526)
- `qemuppc`: `assimp` fails (#526) and `mavros` fails (#525)

bulwahn, 2017-09-25:
Lately, I tested `bitbake core-image-ros-world -k` with and without `ROS_USE_PYTHON3 = "yes"` on some current poky pyro release with all qemu machines in that yocto release.

The common target-machine independent build configuration is:
```
Build Configuration:
BB_VERSION        = "1.34.0"
BUILD_SYS         = "x86_64-linux"
DISTRO            = "poky"
DISTRO_VERSION    = "2.3.1"
meta              
meta-poky         
meta-yocto-bsp    = "pyro:d9b63154aa74f443bd5d8216670428ccb91a6ca5"
meta-oe           
meta-python       
meta-multimedia   = "pyro:5e82995148a2844c6f483ae5ddd1438d87ea9fb7"
```

As testing `core-image-ros-world` with all qemu machines took some days, the testing was first done without `ROS_USE_PYTHON3 = "yes"` on this commit, which was the latest commit at the time of testing:
```
meta-ros          = "master:859f5da5cb2320e67f63f826ec4b07ffd4af83a6"
```

Then, testing `core-image-ros-world` with all qemu machines with `ROS_USE_PYTHON3 = "yes"` was continued on this commit, which incidentally also already resolved an issue with python3 compared to the commit above:
```
meta-ros          = "master:3ff83184bd16a9ef13660dd5df08687a6c17a458"
```

All test builds with and without `ROS_USE_PYTHON3 = "yes"` were successful for all machines except for the qemuarm machine. On qemuarm, there was only one already known build error with `cv-bridge` and `image-geometry`, which fail due to a common issue with the opencv header (#524), which has already been resolved on the current master with an recipe update (cf. #524).

The target-machine specific build configurations are:
```
NATIVELSBSTRING   = "universal-4.8"
TARGET_SYS        = "i586-poky-linux"
MACHINE           = "qemux86"
TUNE_FEATURES     = "m32 i586"
TARGET_FPU        = ""
```
```
NATIVELSBSTRING   = "ubuntu-14.04"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "qemuarm"
TUNE_FEATURES     = "arm armv5 thumb dsp"
TARGET_FPU        = "soft"
```
```
NATIVELSBSTRING   = "universal-4.8"
TARGET_SYS        = "aarch64-poky-linux"
MACHINE           = "qemuarm64"
TUNE_FEATURES     = "aarch64"
TARGET_FPU        = ""
```
```
NATIVELSBSTRING   = "ubuntu-14.04"
TARGET_SYS        = "mips-poky-linux"
MACHINE           = "qemumips"
TUNE_FEATURES     = "o32 bigendian fpu-hard mips32r2"
TARGET_FPU        = ""
```
```
NATIVELSBSTRING   = "ubuntu-14.04"
TARGET_SYS        = "mips64-poky-linux"
MACHINE           = "qemumips64"
TUNE_FEATURES     = "n64 bigendian fpu-hard"
TARGET_FPU        = ""
```
```
NATIVELSBSTRING   = "universal-4.8"
TARGET_SYS        = "powerpc-poky-linux"
MACHINE           = "qemuppc"
TUNE_FEATURES     = "m32 fpu-hard ppc7400 altivec"
TARGET_FPU        = ""
```
```
NATIVELSBSTRING   = "universal-4.8"
TARGET_SYS        = "x86_64-poky-linux"
MACHINE           = "qemux86-64"
TUNE_FEATURES     = "m64 core2"
TARGET_FPU        = ""
```

herb-kuta-lge, 2019-06-28:
@bulwahn < I was hoping not to have to support OpenEmbedded releases prior to thud. If this is OK, please close this issue.

closed 2019-08-13 by herb-kuta-lge