ros/meta-ros #532: roscpp: "error: 'readv'/'writev' was not declared in this scope" in do_compile

bulwahn, 2017-07-12:
`bitbake roscpp` currently fails on openembedded-core with:
```
 /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0/recipe-sysroot-native/usr/bin/i586-oe-linux/i586-oe-linux-g++  -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"roscpp\" -Droscpp_EXPORTS -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0/build/devel/include -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0/ros_comm-1.11.21/clients/roscpp/include -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0/build/devel/include/ros -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0/recipe-sysroot/opt/ros/indigo/include -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0/recipe-sysroot/usr/include -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0/build  -m32 -march=i586  --sysroot=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0=/usr/src/debug/roscpp/1.11.21-r0 -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0/recipe-sysroot=  -fvisibility-inlines-hidden  -m32 -march=i586  --sysroot=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0/recipe-sysroot -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0/ros_comm-1.11.21/clients/roscpp/build/devel/include -fPIC   -Wall -Wextra -o CMakeFiles/roscpp.dir/src/libros/transport/transport_udp.cpp.o -c /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0/ros_comm-1.11.21/clients/roscpp/src/libros/transport/transport_udp.cpp
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0/ros_comm-1.11.21/clients/roscpp/src/libros/transport/transport_udp.cpp: In member function 'virtual int32_t ros::TransportUDP::read(uint8_t*, uint32_t)':
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0/ros_comm-1.11.21/clients/roscpp/src/libros/transport/transport_udp.cpp:407:21: error: 'readv' was not declared in this scope
|          num_bytes = readv(sock_, iov, 2);
|                      ^~~~~
[...]
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0/ros_comm-1.11.21/clients/roscpp/src/libros/transport/transport_udp.cpp: In member function 'virtual int32_t ros::TransportUDP::write(uint8_t*, uint32_t)':
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0/ros_comm-1.11.21/clients/roscpp/src/libros/transport/transport_udp.cpp:579:25: error: 'writev' was not declared in this scope
|      ssize_t num_bytes = writev(sock_, iov, 2);
|                          ^~~~~~
[...]
| make[2]: *** [CMakeFiles/roscpp.dir/src/libros/transport/transport_udp.cpp.o] Error 1
| make[2]: Leaving directory `/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0/build'
| make[1]: *** [CMakeFiles/roscpp.dir/all] Error 2
| make[1]: Leaving directory `/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0/build'
| make: *** [all] Error 2
| WARNING: exit code 2 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/roscpp/1.11.21-r0/temp/log.do_compile.4896)
ERROR: Task (/home/lukas/work/repositories/github.com/bmwcarit/meta-ros/recipes-ros/ros-comm/roscpp_1.11.21.bb:do_compile) failed with exit code '1'
```

The build configuration is:
```
Build Configuration:
BB_VERSION        = "1.34.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "ubuntu-14.04"
TARGET_SYS        = "i586-oe-linux"
MACHINE           = "qemux86"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "m32 i586"
TARGET_FPU        = ""
meta              = "master:b1c4661742d12cfd444043e597d01bd627ee4ca4"
meta-oe           
meta-multimedia   
meta-python       = "master:66b88f7729d24ef2e3469fcaca172f420c4d54be"
meta-ros          = "master:2b7704917c0443183ea1299b15543157fa0c4185"
```

bulwahn, 2017-07-12:
If I find time, I will start bisecting to find the recent commit that introduces the issue.

rojkov, 2017-07-19:
I've submitted a patch for this issue: https://github.com/ros/ros_comm/pull/1105

If it's acceptable I'll make a backport for meta-ros.

PR #534 (2017-07-24): roscpp: add missing header for writev()
resolves #532
commit 00003239 (2017-07-24): roscpp: add missing header for writev()

resolves #532
closed 2017-07-24 by bulwahn