ros/meta-ros #536: pcl: missing runtime dependency for dev-package

peter-fischer-professional, 2017-07-31:
I had a problem with the pcl recipe when I generated a cross compile toolchain: 
Compiling an application using the crosscompile toolchain failed because it used some pcl function that depended on eigen, which was not included in the toolchain's rootfs. I fixed the issue by adding: 
`RDEPENDS_${PN}-dev += " libeigen"`
to the pcl recipe. I thought you might want to include this or an equivalent fix in the repository since others might have the same problem.

bulwahn, 2017-07-31:
Did you already use the commits on pull request #519 to build the cross-compile tool chain?

peter-fischer-professional, 2017-07-31:
No, I actually had the issue back in April. Perhaps I'm missing something but I don't see the relation between this issue and #519 
To clarify, I did neither use any ros-recipes nor any sdk-recipe from the meta-ros layer but only the pcl-recipe.

bulwahn, 2017-07-31:
Well, I just wanted to understand if you build a tool chain for a ROS package, or if you just use the pcl recipe in the meta-ros layer. As you said, it is the latter. Please open a pull request with your commit, then we will review, test and merge it.

bulwahn, 2017-08-25:
I looked into it and wondered if there would be an alternative solution.
If I am not mistaken, the libeigen recipe is the peculiarity that in all its dependencies needs to be included as rdepends-dev dependency. So if bitbake allows that, it would be best if we could include some annotation in the libeigen recipe rather than all the recipes that depend on libeigen.

herb-kuta-lge, 2020-02-07:
@PeterProfessional < Can this issue be closed?

PeterProfessional, 2020-02-10:
The issue was not addressed yet (as far as I know), so I would not close it. I see that it is quite old and the chances that someone will invest the time for it grow somewhat thinner but maybe I'll find the time for a fix in the next 2 months.

shr-project, 2020-07-28:
@PeterProfessional any update on this? We don't plan to do any new changes to old legacy branch, if you don't plan to fix it yourself, then please close it.

robwoolley, 2023-11-17:
This appears to be against the old legacy branch.  Closing it as it is over 5 years old and we don't intend to support the legacy branch.

closed 2023-11-17 by robwoolley