ros/meta-ros #542: build failures with updated binutils 2.29

bulwahn, 2017-08-23:
Since the update of binutils to version 2.29 [1], `bitbake core-image-ros-world -k` fails with 4 recipes at` do_compile`:
- [ ] kdl-parser_1.11.14.bb
- [ ] industrial-robot-client_0.4.3.bb
- [ ] collada-urdf_1.11.14.bb
- [ ] cv-camera_0.1.0.bb

This issue also explains the CI build failures for the recent pull requests (#540, #541).
My educated guess is that the linker (`ld`) in the updated binutils now behaves slightly different, and hence, an undetected issue in the build scripts of these ROS packages have now become apparent.

[1] http://cgit.openembedded.org/openembedded-core/commit/?id=3bf990eb275f63190a2cf7253527d6d49fd93f1a

bulwahn, 2017-08-23:
@rojkov I will start looking into the `cv-camera` issue first, and report my findings here.

bulwahn, 2017-08-23:
I did have a first look at the `CMakeLists.txt` of `camera_info_manager` and `camera_calibration_parsers`, and I could not find any specialities in the setup. So possibly, the issue is really that `CATKIN_DEPENDS` does not properly work with the new linker in version 2.29?

rojkov, 2017-08-23:
@bulwahn Thanks! Even on my machine bisecting would take ages.

My initial guess is it could be that before the linker update the paths to libclass_loader.so and other libs were taken from liburdf's RPATH (AFAIK CMake sets RPATH for build time and removes it from the lib at install time). And the new linker ignores RPATH.

rojkov, 2017-08-23:
No, in both cases there's no rpath in liburdf.so. And the command for linker includes `-L/path/to/build/tmp-glibc/work/i586-oe-linux/kdl-parser/1.11.14-r0/recipe-sysroot/opt/ros/indigo/lib/`. That's weird.

rojkov, 2017-08-23:
In both cases command lines and environments looks to be the same. It seems that transitive dependencies are broken in meta-ros's `CATKIN_DEPENDS` indeed (i.e. class_loader -> pluginlib -> urdf -> kdl-parser). At least it builds ok when I add the missing libs explicitly.

rojkov, 2017-08-25:
It turned out my initial guess was closer to the reality: with this change [1] merged to binutils linker can't find the lib in the given RPATH, because it prepends the value of `-rpath` option with `--sysroot`. The effective path becomes

```
/data/cibot/deleteme/build/tmp-glibc/work/i586-oe-linux/kdl-parser/1.11.14-r0/recipe-sysroot/data/cibot/deleteme/build/tmp-glibc/work/i586-oe-linux/kdl-parser/1.11.14-r0/recipe-sysroot/opt/ros/indigo/lib
```
which is wrong.

With the patch reverted the package builds successfully again. I'm trying to find where this `-rpath` option comes from (cmake, yocto or catkin), but no luck so far.

[1] https://sourceware.org/ml/binutils/2017-03/msg00134.html

rojkov, 2017-08-29:
I've submitted a workaround in https://github.com/bmwcarit/meta-ros/pull/546 for this issue. But I'd like to keep it open because either correct -rpath or just -rpath-link should be used in the linker commands. 

Unfortunately CMake is not very helpful at explaining how the current -rpath ends up in the command.

herb-kuta-lge, 2019-11-14:
All 4 of these successfully build for **melodic** using `[thud-draft]`. @bulwahn < Can I close this issue?

closed 2019-12-03 by herb-kuta-lge