ros/meta-ros #547: libmavconn compile failed

yangjiel, 2017-08-29:
This failure first found at Aug 27 23:56, 2017 with **Morty** **Yocto** build
````
| In file included from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/../message.hpp:32:0,
|                  from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/common.hpp:16,
|                  from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/libmavconn/0.18.7-r0/mavros-0.18.7/libmavconn/include/mavconn/mavlink_dialect.h:23,
|                  from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/libmavconn/0.18.7-r0/mavros-0.18.7/libmavconn/include/mavconn/interface.h:37,
|                  from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/libmavconn/0.18.7-r0/mavros-0.18.7/libmavconn/src/mavlink_helpers.cpp:21:
| /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/../mavlink_helpers.h: In function 'uint8_t mavlink::mavlink_frame_char_buffer(mavlink::mavlink_message_t*, mavlink::mavlink_status_t*, uint8_t, mavlink::mavlink_message_t*, mavlink::mavlink_status_t*)':
| /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/../mavlink_helpers.h:781:54: error: 'MAVLINK_START_SIGN_STREAM' was not declared in this scope
|     MAVLINK_START_SIGN_STREAM(status->signing->link_id);
|                                                       ^
| /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/../mavlink_helpers.h:783:52: error: 'MAVLINK_END_SIGN_STREAM' was not declared in this scope
|     MAVLINK_END_SIGN_STREAM(status->signing->link_id);
|                                                     ^
| In file included from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/../message.hpp:32:0,
|                  from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/common.hpp:16,
|                  from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/libmavconn/0.18.7-r0/mavros-0.18.7/libmavconn/include/mavconn/mavlink_dialect.h:23,
|                  from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/libmavconn/0.18.7-r0/mavros-0.18.7/libmavconn/include/mavconn/interface.h:37,
|                  from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/libmavconn/0.18.7-r0/mavros-0.18.7/libmavconn/src/interface.cpp:22:
| /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/../mavlink_helpers.h: In function 'uint8_t mavlink::mavlink_frame_char_buffer(mavlink::mavlink_message_t*, mavlink::mavlink_status_t*, uint8_t, mavlink::mavlink_message_t*, mavlink::mavlink_status_t*)':
| /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/../mavlink_helpers.h:781:54: error: 'MAVLINK_START_SIGN_STREAM' was not declared in this scope
|     MAVLINK_START_SIGN_STREAM(status->signing->link_id);
|                                                       ^
| /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/../mavlink_helpers.h:783:52: error: 'MAVLINK_END_SIGN_STREAM' was not declared in this scope
|     MAVLINK_END_SIGN_STREAM(status->signing->link_id);
|                                                     ^
| In file included from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/../message.hpp:32:0,
|                  from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/common.hpp:16,
|                  from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/libmavconn/0.18.7-r0/mavros-0.18.7/libmavconn/include/mavconn/mavlink_dialect.h:23,
|                  from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/libmavconn/0.18.7-r0/mavros-0.18.7/libmavconn/include/mavconn/interface.h:37,
|                  from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/libmavconn/0.18.7-r0/mavros-0.18.7/libmavconn/include/mavconn/tcp.h:24,
|                  from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/libmavconn/0.18.7-r0/mavros-0.18.7/libmavconn/src/tcp.cpp:22:
| /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/../mavlink_helpers.h: In function 'uint8_t mavlink::mavlink_frame_char_buffer(mavlink::mavlink_message_t*, mavlink::mavlink_status_t*, uint8_t, mavlink::mavlink_message_t*, mavlink::mavlink_status_t*)':
| /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/../mavlink_helpers.h:781:54: error: 'MAVLINK_START_SIGN_STREAM' was not declared in this scope
|     MAVLINK_START_SIGN_STREAM(status->signing->link_id);
|                                                       ^
| /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/../mavlink_helpers.h:783:52: error: 'MAVLINK_END_SIGN_STREAM' was not declared in this scope
|     MAVLINK_END_SIGN_STREAM(status->signing->link_id);
|                                                     ^
| In file included from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/../message.hpp:32:0,
|                  from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/common.hpp:16,
|                  from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/libmavconn/0.18.7-r0/mavros-0.18.7/libmavconn/include/mavconn/mavlink_dialect.h:23,
|                  from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/libmavconn/0.18.7-r0/mavros-0.18.7/libmavconn/include/mavconn/interface.h:37,
|                  from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/libmavconn/0.18.7-r0/mavros-0.18.7/libmavconn/include/mavconn/udp.h:22,
|                  from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/libmavconn/0.18.7-r0/mavros-0.18.7/libmavconn/src/udp.cpp:22:
| /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/../mavlink_helpers.h: In function 'uint8_t mavlink::mavlink_frame_char_buffer(mavlink::mavlink_message_t*, mavlink::mavlink_status_t*, uint8_t, mavlink::mavlink_message_t*, mavlink::mavlink_status_t*)':
| /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/../mavlink_helpers.h:781:54: error: 'MAVLINK_START_SIGN_STREAM' was not declared in this scope
|     MAVLINK_START_SIGN_STREAM(status->signing->link_id);
|                                                       ^
| ERROR: oe_runmake failed
| /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/../mavlink_helpers.h:783:52: error: 'MAVLINK_END_SIGN_STREAM' was not declared in this scope
|     MAVLINK_END_SIGN_STREAM(status->signing->link_id);
|                                                     ^
| In file included from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/../message.hpp:32:0,
|                  from /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/sysroots/duovero/opt/ros/indigo/include/mavlink/v2.0/common/common.hpp:16,
[... cut ...]
| WARNING: /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/libmavconn/0.18.7-r0/temp/run.do_compile.54082:1 exit 1 from 'exit 1'
| ERROR: Function failed: do_compile (log file is located at /home/gumstix/Buildbot/yocto-duovero-morty/build/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/libmavconn/0.18.7-r0/temp/log.do_compile.54082)
NOTE: recipe libmavconn-0.18.7-r0: task do_compile: Failed
ERROR: Task (/home/gumstix/Buildbot/yocto-duovero-morty/build/poky/meta-ros/recipes-ros/mavros/libmavconn_0.18.7.bb:do_compile) failed with exit code '1'
````

bulwahn, 2017-08-30:
Thank you very much for reporting.

I believe the issue is that the mavlink recipe [1] uses `AUTOREV`, which means it always updates to most recent commit in the git repository. I think we should just tie it to a specific commit (of a specific released version), and make sure to update that in controlled fashion.
When mavlink and mavros was proposed as pull request and merged, I did not known how many users would actually use and rely on those recipes, so I noticed the AUTOREV line, but I did not start nitpicking about that detail.

[1] https://github.com/bmwcarit/meta-ros/blob/master/recipes-ros/mavlink/mavlink_git.bb

yangjiel, 2017-08-30:
No problem, it makes sense. AUTOREV is a good option for reducing the work of updating recipes though if they tested through the release. If the problem is fixed, please close this issue then I will know it. Thank you!

TSC21, 2017-08-30:
@yangjiel MAVROS and MAVlink packages have different release timing, so that's why sometimes we can find issues when the MAVLink package is not sync with upstream of MAVLink.

bulwahn, 2017-08-31:
In the commit
https://github.com/bmwcarit/meta-ros/commit/068d91088e2b9de110249e852e6e51bc28cf94ee, we tied mavlink to a specific version, so this issue is now resolved if you update to the current master or cherry-pick this commit.

closed 2017-08-31 by bulwahn