ros/meta-ros #548: build failures in `do_configure` for x86-64 architecture

bulwahn, 2017-08-31:
Currently, a large number of recipes fail in `do_configure` with the same error:
```
[...]
| -- Found PythonInterp: /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/genmsg/0.5.8-r0/recipe-sysroot-native/usr/bin/python-native/python (found version "2.7.13")
| -- Using PYTHON_EXECUTABLE: /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/genmsg/0.5.8-r0/recipe-sysroot-native/usr/bin/python-native/python
| -- Using default Python package layout
| -- Found PY_em: /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/genmsg/0.5.8-r0/recipe-sysroot-native/usr/lib/python2.7/site-packages/em.pyc
| -- Using empy: /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/genmsg/0.5.8-r0/recipe-sysroot-native/usr/lib/python2.7/site-packages/em.pyc
| -- Using CATKIN_ENABLE_TESTING: 0
| -- catkin 0.6.19
| rm: relocation error: /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/genmsg/0.5.8-r0/recipe-sysroot/usr/lib/libc.so.6: symbol __tunable_get_val, version GLIBC_PRIVATE not defined in file ld-linux-x86-64.so.2 with link time reference
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/genmsg/0.5.8-r0/recipe-sysroot-native/usr/bin/python-native/python: relocation error: /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/genmsg/0.5.8-r0/recipe-sysroot/usr/lib/libc.so.6: symbol __tunable_get_val, version GLIBC_PRIVATE not defined in file ld-linux-x86-64.so.2 with link time reference
| Traceback (most recent call last):
|   File "/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/genmsg/0.5.8-r0/build/catkin_generated/generate_cached_setup.py", line 22, in <module>
|     code = generate_environment_script('/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/genmsg/0.5.8-r0/build/devel/env.sh')
|   File "/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/genmsg/0.5.8-r0/recipe-sysroot-native/opt/ros/indigo/lib/python2.7/site-packages/catkin/environment_cache.py", line 61, in generate_environment_script
|     output = subprocess.check_output([env_script, sys.executable, '-c', python_code])
|   File "/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/genmsg/0.5.8-r0/recipe-sysroot-native/usr/lib/python2.7/subprocess.py", line 219, in check_output
|     raise CalledProcessError(retcode, cmd, output=output)
| subprocess.CalledProcessError: Command '['/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/genmsg/0.5.8-r0/build/devel/env.sh', '/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/genmsg/0.5.8-r0/recipe-sysroot-native/usr/bin/python-native/python', '-c', 'import os; print(dict(os.environ))']' returned non-zero exit status 127
| CMake Error at /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/genmsg/0.5.8-r0/recipe-sysroot-native/opt/ros/indigo/share/catkin/cmake/safe_execute_process.cmake:11 (message):
| 
|   execute_process(/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/genmsg/0.5.8-r0/recipe-sysroot-native/usr/bin/python-native/python
|   "/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/genmsg/0.5.8-r0/build/catkin_generated/generate_cached_setup.py")
|   returned error code 1
| Call Stack (most recent call first):
|   /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/genmsg/0.5.8-r0/recipe-sysroot-native/opt/ros/indigo/share/catkin/cmake/all.cmake:184 (safe_execute_process)
|   /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/genmsg/0.5.8-r0/recipe-sysroot-native/opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:20 (include)
|   CMakeLists.txt:3 (find_package)
| 
| 
| -- Configuring incomplete, errors occurred!
| See also "/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/genmsg/0.5.8-r0/build/CMakeFiles/CMakeOutput.log".
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/genmsg/0.5.8-r0/temp/log.do_configure.27982)
ERROR: Task (/home/lukas/work/repositories/github.com/bmwcarit/meta-ros/recipes-ros/genmsg/genmsg_0.5.8.bb:do_configure) failed with exit code '1'
```
(Here, the error message from `genmsg` is just one instance; there are many more errors for other recipes.)

The build configuration is:
```
Build Configuration:
BB_VERSION        = "1.35.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "ubuntu-14.04"
TARGET_SYS        = "x86_64-oe-linux"
MACHINE           = "qemux86-64"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "m64 core2"
TARGET_FPU        = ""
meta              = "master:62f1122ef166eba56441d669c6b3b3fe5f367418"
meta-oe           
meta-multimedia   
meta-python       = "master:28d2c9b4474844516a9e8328bb497cdf3bec88ef"
meta-ros          = "assimp:fadf0d6048dce4a51a456c199fefcd4a64e4d646"
```

rojkov, 2017-08-31:
Hm.. looks like the issue I fixed with https://github.com/bmwcarit/meta-ros/pull/521 for Yocto Refkit.

bulwahn, 2017-08-31:
I will check if the commit from #521 resolves this issue.

bulwahn, 2017-09-01:
I am currently bisecting to find the first bad commit.

bulwahn, 2018-07-10:
The issue only occurred with the usrmerge feature; I have not seen this issue re-appear since quite some time, so I will close this issue here for now.

closed 2018-07-10 by bulwahn