ros/meta-ros #553: festival-2.3-r0 do_compile: oe_runmake failed with No GNU_HASH in the elf binary

yahiafarghaly, 2017-09-16:
When i just do this IMAGE_INSTALL_append = " festival" . it promotes me with error of 
do_compile: oe_runmake failed and No GNU_HASH in the elf binary  

here is the log: 

> Build Configuration:
> BB_VERSION        = "1.34.0"
> BUILD_SYS         = "x86_64-linux"
> NATIVELSBSTRING   = "universal"
> TARGET_SYS        = "arm-poky-linux-gnueabi"
> MACHINE           = "raspberrypi3"
> DISTRO            = "poky"
> DISTRO_VERSION    = "2.3.2"
> TUNE_FEATURES     = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
> TARGET_FPU        = "hard"
> meta              
> meta-poky         
> meta-yocto-bsp    = "pyro:ce26a57e04ad65c02087629701d96448a44e73d5"
> meta-oe           
> meta-python       
> meta-multimedia   
> meta-networking   = "pyro:5e82995148a2844c6f483ae5ddd1438d87ea9fb7"
> meta-raspberrypi  = "pyro:c981f01a2cdb3299eb57920cffbb5dfd4248209a"
> meta-ros          = "master:3ff83184bd16a9ef13660dd5df08687a6c17a458"
> meta-mylayer      = "pyro:ce26a57e04ad65c02087629701d96448a44e73d5"
> Initialising tasks: 100% |############################################################################################################################| Time: 0:00:05
> NOTE: Executing SetScene Tasks
> NOTE: Executing RunQueue Tasks
> ERROR: festival-2.3-r0 do_compile: oe_runmake failed
> ERROR: festival-2.3-r0 do_compile: Function failed: do_compile (log file is located at /yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/festival/2.3-r0/temp/log.do_compile.111086)
> ERROR: Logfile of failure stored in: /yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/festival/2.3-r0/temp/log.do_compile.111086
> Log data follows:
> | DEBUG: SITE files ['endian-little', 'bit-32', 'arm-common', 'arm-32', 'common-linux', 'common-glibc', 'arm-linux', 'arm-linux-gnueabi', 'common']
> | DEBUG: Executing shell function do_compile
> | NOTE: make
> | ERROR: oe_runmake failed
> | config/config:43: /yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/festival/2.3-r0/recipe-sysroot/usr/share/speech-tools/config/config: No such file or directory
> | config/common_make_rules:67: /yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/festival/2.3-r0/recipe-sysroot/usr/share/speech-tools/config/rules/common_make_rules.mak: No such file or directory
> | Makefile:84: /yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/festival/2.3-r0/recipe-sysroot/usr/share/speech-tools/config/rules/top_level.mak: No such file or directory
> | Makefile:85: /yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/festival/2.3-r0/recipe-sysroot/usr/share/speech-tools/config/rules/install.mak: No such file or directory
> | make: *** No rule to make target '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/festival/2.3-r0/recipe-sysroot/usr/share/speech-tools/config/rules/install.mak'.  Stop.
> | ERROR: Function failed: do_compile (log file is located at /yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/festival/2.3-r0/temp/log.do_compile.111086)
> ERROR: Task (/yocto/poky/meta-ros/recipes-extended/festival/festival_2.3.bb:do_compile) failed with exit code '1'
> ERROR: speech-tools-2.3-r0 do_package_qa: QA Issue: No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/pda'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/design_filter'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/sigfilter'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/ols_test'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/ch_utt'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/sig2fv'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/siod'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/ngram_test'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/ch_wave'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/dp'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/wagon_test'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/ngram_build'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/fringe_client'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/viterbi'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/scfg_parse'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/wfst_train'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/scfg_test'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/pitchmark'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/na_play'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/ch_lab'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/ols'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/na_record'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/spectgen'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/wfst_run'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/scfg_train'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/xml_parser'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/align'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/bcat'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/tilt_analysis'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/ch_track'
> No GNU_HASH in the elf binary: '/yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/packages-split/speech-tools/usr/bin/wagon'
[... cut ...]
> ERROR: Logfile of failure stored in: /yocto/poky/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/speech-tools/2.3-r0/temp/log.do_package_qa.111098
> ERROR: Task (/yocto/poky/meta-ros/recipes-extended/speech-tools/speech-tools_2.3.bb:do_package_qa) failed with exit code '1'
> NOTE: Tasks Summary: Attempted 2083 tasks of which 2081 didn't need to be rerun and 2 failed.
> Summary: 2 tasks failed:
>   /yocto/poky/meta-ros/recipes-extended/festival/festival_2.3.bb:do_compile
>   /yocto/poky/meta-ros/recipes-extended/speech-tools/speech-tools_2.3.bb:do_package_qa
> Summary: There were 5 ERROR messages shown, returning a non-zero exit code.

tinoue, 2017-10-13:
For now I create festival_2.3.bbappend which has the following lines:
----
INSANE_SKIP_${PN} = "ldflags"
INSANE_SKIP_${PN}-dev = "ldflags"

herb-kuta-lge, 2019-07-02:
Commit 181946aac3733c1c8c483a8f24f7d733367c9f40 upgraded version to 2.4 .

shr-project, 2019-07-02:
The issue still exists with 2.4 version, fixing it in speech-tools is simple (just adding LDFLAGS to LINKFLAGS), but then festival fails to build (it reuses parts of speech-tools makefile configuration to build) and I haven't fixed this festival issue yet.

shr-project, 2019-07-03:
Fixed completely in https://github.com/ros/meta-ros/pull/634

closed 2019-07-04 by shr-project