ros/meta-ros #570: Some ROS packages are not ported to python3

bulwahn, 2017-12-19:
A number of packages contain scripts that seem to require python2.

Hence, `core-image-ros-world -k` (with ROS_USE_PYTHON3 = "yes") fails with a number of `do_package_qa` issues:
```
ERROR: camera-calibration-1.12.21-r0 do_package_qa: QA Issue: /opt/ros/indigo/lib/camera_calibration/cameracheck.py contained in package camera-calibration requires /usr/bin/python, but no providers found in RDEPENDS_camera-calibration? [file-rdeps]
ERROR: actionlib-1.11.9-r0 do_package_qa: QA Issue: /opt/ros/indigo/lib/actionlib/axclient.py contained in package actionlib requires /usr/bin/python, but no providers found in RDEPENDS_actionlib? [file-rdeps]
ERROR: tf2-kdl-0.5.16-r0 do_package_qa: QA Issue: /opt/ros/indigo/lib/tf2_kdl/test.py contained in package tf2-kdl requires /usr/bin/python, but no providers found in RDEPENDS_tf2-kdl? [file-rdeps]
ERROR: diagnostic-analysis-1.9.2-r0 do_package_qa: QA Issue: /opt/ros/indigo/lib/diagnostic_analysis/export_csv.py contained in package diagnostic-analysis requires /usr/bin/python, but no providers found in RDEPENDS_diagnostic-analysis? [file-rdeps]
ERROR: tf2-tools-0.5.16-r0 do_package_qa: QA Issue: /opt/ros/indigo/lib/tf2_tools/view_frames.py contained in package tf2-tools requires /usr/bin/python, but no providers found in RDEPENDS_tf2-tools? [file-rdeps]
ERROR: fake-localization-1.12.13-r0 do_package_qa: QA Issue: /opt/ros/indigo/lib/fake_localization/static_odom_broadcaster.py contained in package fake-localization requires /usr/bin/python, but no providers found in RDEPENDS_fake-localization? [file-rdeps]
ERROR: velodyne-pointcloud-1.3.0-r0 do_package_qa: QA Issue: /opt/ros/indigo/lib/velodyne_pointcloud/gen_calibration.py contained in package velodyne-pointcloud requires /usr/bin/python, but no providers found in RDEPENDS_velodyne-pointcloud? [file-rdeps]
ERROR: move-base-1.12.13-r0 do_package_qa: QA Issue: /opt/ros/indigo/lib/move_base/warner.py contained in package move-base requires /usr/bin/python, but no providers found in RDEPENDS_move-base? [file-rdeps]
```

rojkov, 2017-12-19:
The recommendation for ROS is to make all python scripts compatible with both python2 and python3.

The OE-Core developers don't believe that scripts can be compatible with python2 and python3 at the same time, so I had to add this workaround in Refkit [1].

[1]  https://github.com/intel/intel-iot-refkit/commit/9e236306be37d3b3368b35e0634dceb2090bd3ca

bulwahn, 2017-12-19:
Okay, I might just add that workaround in meta-ros as well, and check if the workaround resolves this issue.

commit 6ddc0cc5 (2018-02-23): python3: make python point to python3 (resolves #570)

This commit is just a copy of the patch that Dmitry Rozhkov
submitted to openembedded-core, but it was rejected there [1].

If only python3 is installed on an image nothing provides
/usr/bin/python even though many scripts compatible with
both python2 and python3 just state `#!/usr/bin/env python`
in their shebang line.

Make python and python3 recipes provide alternatives for
/usr/bin/python and /usr/bin/python-config. By default
python2 has higher priority over python3.

In meta-ros, we need this behaviour for ROS_USE_PYTHON3 = "yes"
bec [... cut]
closed 2018-02-23 by rojkov