ros/meta-ros #574: Pull request #573 broke my custom image

jayzhao93, 2018-01-15:
Hi everyone,
Here is the error log message i have got today, the new added socketcan recipe failed when compiling. Does anyone know how to fix it?

Thanks 
Jay

```
| ERROR: oe_runmake failed
| ~/build/build/tmp/sysroots/x86_64-linux/usr/libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/6.2.0/ld: CMakeFiles/socketcan_dump.dir/src/candump.cpp.o: undefined reference to symbol 'pthread_getspecific@@GLIBC_2.4'
| ~/build/build/tmp/sysroots/colibri-imx6/lib/libpthread.so.0: error adding symbols: DSO missing from command line
| collect2: error: ld returned 1 exit status
| make[2]: *** [CMakeFiles/socketcan_dump.dir/build.make:116: devel/lib/socketcan_interface/socketcan_dump] Error 1
| make[2]: Leaving directory '~/build/build/tmp/work/colibri_imx6-poky-linux-gnueabi/socketcan-interface/0.6.8-r0/build'
| make[1]: *** [CMakeFiles/Makefile2:172: CMakeFiles/socketcan_dump.dir/all] Error 2
| make[1]: Leaving directory '~/build/build/tmp/work/colibri_imx6-poky-linux-gnueabi/socketcan-interface/0.6.8-r0/build'
| make: *** [Makefile:131: all] Error 2
| WARNING: ~/build/build/tmp/work/colibri_imx6-poky-linux-gnueabi/socketcan-interface/0.6.8-r0/temp/run.do_compile.146847:1 exit 1 from 'exit 1'
| ERROR: Function failed: do_compile (log file is located at ~/build/build/tmp/work/colibri_imx6-poky-linux-gnueabi/socketcan-interface/0.6.8-r0/temp/log.do_compile.146847)
NOTE: recipe socketcan-interface-0.6.8-r0: task do_compile: Failed
ERROR: Task (~/build/poky/meta-ros/recipes-ros/ros-canopen/socketcan-interface_0.6.8.bb:do_compile) failed with exit code '1'
```

bulwahn, 2018-01-16:
I will try to reproduce the issue. Could you please provide your build configuration?

bhuvanchandra, 2018-01-16:
I could reproduce the build issue with socketcan-interface, [here](https://pastebin.com/J40deZ0K) is the build log.

Here is the fix for the build issue:
```
diff --git a/recipes-ros/ros-canopen/socketcan-interface_0.6.8.bb b/recipes-ros/ros-canopen/socketcan-interface_0.6.8.bb
index 94a7a43..87657e3 100644
--- a/recipes-ros/ros-canopen/socketcan-interface_0.6.8.bb
+++ b/recipes-ros/ros-canopen/socketcan-interface_0.6.8.bb
@@ -4,6 +4,7 @@ LICENSE = "LGPLv3"
 LIC_FILES_CHKSUM = "file://package.xml;beginline=10;endline=10;md5=b691248d2f70cdaeeaf13696ada5d47c"
 
 DEPENDS = "boost class-loader virtual/kernel"
+TARGET_CC_ARCH += "-pthread"
 
 require ros-canopen.inc
```

@jayzhao93 can you test it and confirm with the above fix ?

--
Bhuvan

bulwahn, 2018-01-16:
I could reproduce this as well. @bhuvanchandra I also tested your patch to the recipe and it resolves the issue; however, I would prefer if we can patch the CMakeLists.txt instead. I have seen this issue before and we could resolve that usually by adding a line in the CMakeLists.txt in the build script. You might find an example of that in the git history: just look at the file names of all the patch files that have been in the meta-ros repository.

bulwahn, 2018-01-16:
This link here, https://stackoverflow.com/questions/5395309/how-do-i-force-cmake-to-include-pthread-option-during-compilation, could point to second alternative solution to this issue.

bulwahn, 2018-01-16:
I actually have a working solution now and will provide it here soon.

jayzhao93, 2018-01-16:
the patch provide from @bhuvanchandra is working, thanks
Is there going to be another solution for this?

bulwahn, 2018-01-17:
I just submitted a patch to ros_canopen (https://github.com/ros-industrial/ros_canopen/pull/263). I will wait the next two days if this is merged or needs further discussion. Then, I will provide a pull request here with the submitted patch applied in the recipe.

bulwahn, 2018-01-31:
I did wait a bit longer without success, as my provided patch was not yet merged, but now I provide a pull request here nevertheless.

commit 5d16b796 (2018-02-05): socketcan-interface: patch CMakeLists.txt (resolves #574)
commit ae235635 (2018-02-05): Merge pull request #576 from bulwahn/ros-canopen-fixup

socketcan-interface: patch CMakeLists.txt (resolves #574)
closed 2018-02-05 by rojkov