ros/meta-ros #575: assimp 4.1.0: "error: lvalue required as unary '&' operand" on PPC

casey-ac, 2018-01-19:
assimp version 4.1 fails when building on a PowerPC e5500 system. 
`/home/casey/Documents/Sources/doomsday-2.0.0/doomsday/external/assimp/code/Bitmap.cpp: In instantiation of ‘std::size_t Assimp::Copy(uint8_t*, T&) [with T = short unsigned int; std::size_t = unsigned int; uint8_t = unsigned char]’:
/home/casey/Documents/Sources/doomsday-2.0.0/doomsday/external/assimp/code/Bitmap.cpp:95:50:   required from here
/home/casey/Documents/Sources/doomsday-2.0.0/doomsday/external/assimp/code/Bitmap.cpp:87:27: error: lvalue required as unary ‘&’ operand
         std::memcpy(data, &AI_BE(field), sizeof(field)); return sizeof(field);
                           ^
/home/casey/Documents/Sources/doomsday-2.0.0/doomsday/external/assimp/code/Bitmap.cpp: In instantiation of ‘std::size_t Assimp::Copy(uint8_t*, T&) [with T = unsigned int; std::size_t = unsigned int; uint8_t = unsigned char]’:
/home/casey/Documents/Sources/doomsday-2.0.0/doomsday/external/assimp/code/Bitmap.cpp:96:50:   required from here
/home/casey/Documents/Sources/doomsday-2.0.0/doomsday/external/assimp/code/Bitmap.cpp:87:27: error: lvalue required as unary ‘&’ operand
/home/casey/Documents/Sources/doomsday-2.0.0/doomsday/external/assimp/code/Bitmap.cpp: In instantiation of ‘std::size_t Assimp::Copy(uint8_t*, T&) [with T = int; std::size_t = unsigned int; uint8_t = unsigned char]’:
/home/casey/Documents/Sources/doomsday-2.0.0/doomsday/external/assimp/code/Bitmap.cpp:110:48:   required from here
/home/casey/Documents/Sources/doomsday-2.0.0/doomsday/external/assimp/code/Bitmap.cpp:87:27: error: lvalue required as unary ‘&’ operand
sdk/libgui/assimp/code/CMakeFiles/assimp.dir/build.make:518: recipe for target 'sdk/libgui/assimp/code/CMakeFiles/assimp.dir/Bitmap.cpp.o' failed
make[2]: *** [sdk/libgui/assimp/code/CMakeFiles/assimp.dir/Bitmap.cpp.o] Error 1
CMakeFiles/Makefile2:592: recipe for target 'sdk/libgui/assimp/code/CMakeFiles/assimp.dir/all' failed
make[1]: *** [sdk/libgui/assimp/code/CMakeFiles/assimp.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
`
Build/Target System: 
Ubuntu 16.04.3 LTS on PowerPC e5500 (A-Eon Cyrus Plus board based on Freescale P5020)

bulwahn, 2018-01-19:
@casey-ac Do you use the latest version of meta-ros?

I believe I had already fixed this issue in the past. That is at least what my commit message in https://github.com/bmwcarit/meta-ros/commit/5d71cc0abbc827719ffbbcb41db9cb5d71177401 suggests that I could fix https://github.com/bmwcarit/meta-ros/issues/526 with the upgrade.

Please provide your build configuration and we will try to help you.
(Are you sure you are build assimp 4.1 or is it a typo on assimp 3.1?)

casey-ac, 2018-01-19:
Apologies, I have no idea what I was doing last night, but that bug report was supposed to have been posted to https://github.com/assimp/assimp .

---Casey

I believe I had already fixed this issue in the past. That is at least what my commit message in 5d71cc0<https://github.com/bmwcarit/meta-ros/commit/5d71cc0abbc827719ffbbcb41db9cb5d71177401> suggests that I could fix #526<https://github.com/bmwcarit/meta-ros/issues/526> with the upgrade.

bulwahn, 2018-01-20:
No worries, then I will close the issue here.

closed 2018-01-20 by bulwahn