ros/meta-ros #577: syslog-ng: ExpansionError during parsing

bulwahn, 2018-01-31:
Since https://github.com/openembedded/openembedded-core/commit/0391fcad9103abca0796a068f957d0df63ab4776 ("classes/utils: remove compatibility functions"), `bitbake --parse-only syslog-ng` fails with:
```
ERROR: ExpansionError during parsing /[...]/openembedded.org/meta-openembedded/meta-oe/recipes-support/syslog-ng/syslog-ng_3.8.1.bb
Traceback (most recent call last):
bb.data_smart.ExpansionError: Failure expanding variable CONFIG_TLS, expression was ${@base_conditional( "DEBUG_BUILD", "1", " --disable-thread-tls", " --enable-thread-tls", d )} which triggered exception NameError: name 'base_conditional' is not defined
```

Currently, the meta-openembedded master, commit openembedded/meta-openembedded@6f536298c1afcaaff2771c7d0f242259d6b9fc1f ("fontforge: explicitly depend on gettext-native") does not properly parse with any openembedded-core commit since commit https://github.com/openembedded/openembedded-core/commit/0391fcad9103abca0796a068f957d0df63ab4776 ("classes/utils: remove compatibility functions").

So, meta-ros also does not parse properly due to its dependencies. We will just need to wait until meta-openembedded is updated and this issue is resolved.

bulwahn, 2018-01-31:
This also explains the current CI build error for pull request #576.

bulwahn, 2018-02-02:
We fortunately did not need to wait too long; http://git.openembedded.org/meta-openembedded/commit/?id=99aa19ff53922b61dee0c8b63ee7f664f90e9a91 resolves this issue. I will now resubmit the pending pull requests one by one to get the CI lighting up green again.

closed 2018-02-05 by bulwahn

ThatAdamsGuy, 2018-04-24:
I'm still getting this with master branch?