ros/meta-ros #584: ERROR: ExpansionError during parsing /home/jarvis/poky/meta-ros/recipes-extended/wxwidgets/wxwidgets_2.9.5.bb

jarviscrypter, 2018-03-26:
Hi all,

I am trying to build ros to my yocto image poky and i have followed ROS's official page documentation but when i run "bitbake roslaunch" or any of other ros package name i am getting the following error

since we found two links which helpus build ros and opencv bettter we followed them today and ended up with erros for ros, both are cross-compilation method
the respective errors i have posted a thread in the respective github and meta-ros thread

Do help me overcome this error

bulwahn, 2018-03-26:
I am sorry but I don't know what error you are reporting and how to reproduce this. The issue report seems to be incomplete.

jarviscrypter, 2018-03-26:
what else are you expecting with respect to the issue so that i can provide the required complete information

bulwahn, 2018-03-26:
Is this a duplicate of https://github.com/bmwcarit/meta-ros/issues/578? Are you using the current master of meta-ros? What is your build configuration?

jarviscrypter, 2018-03-26:
Basically i am following the Installation Guide: http://wiki.ros.org/hydro/Installation/OpenEmbedded to cross-compile meta-ros so that i can add the required packages to my my Build image
Yes that is the error i was referring to almost same type, but in #578 i dint get the rectification method
And yes, i am using meta-ros in master branch.
I'll send my build configuration in a while

jarviscrypter, 2018-03-26:
This is my build configuration:

Build Configuration:
BB_VERSION           = "1.36.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "i586-poky-linux"
MACHINE              = "qemux86"
DISTRO               = "poky"
DISTRO_VERSION       = "2.4.2"
TUNE_FEATURES        = "m32 i586"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "rocko:3d66766a49682d8e236390ebedae6ae59b97116c"
meta-oe              
meta-python          
meta-networking      = "master:4598769a96980b756b3489ac71ed46a075cfabd7"

bulwahn, 2018-03-26:
Hmm, do you actually need the wxwidgets recipe? If not, you can simply delete that file and the depending recipes in meta-ros and it should work. That is certainly a simple work around.
To actually repair the wxwidgets recipe, I would to see the specific error message, "ExpansionError" does not tell enough information; but I assume you can fix it by understanding what we did for #578 and the according commit #579.

jarviscrypter, 2018-03-26:
Sure thanks,

So finally what method would you suggest for installing ROS related
packages on to yocto?
Because when I directly included the meta-ros layer and then when I added
the recipes to core-extra-image-instsll and build the image ,the image was
successfully built but the ros packages like : rospy, roslib were not
importing in the python shell of poky.
And that is when I found the ros official website guidance,
So what would be the exact methodology that you'd suggest me to go ahead.
That is what is the exact procedure to follow to have Ros related packages
be included in poky.
These tips and suggestions would be of great value to my work.

Thanks

bulwahn, 2018-03-26:
I am sorry but I don't understand your question. I am the maintainer of meta-ros and the author of this documentation on the ROS wiki. I try to keep meta-ros running on the current master branch of openembedded-core and meta-openembedded, but some minor issues can always appear on "older" release versions of yocto due to backwards-incompatible changes that are simply unavoidable.

jarviscrypter, 2018-03-26:
Hi Lukas,

Sorry for the confusions in my questions, can i have your personal email-id
so i can give you a clear picture of mu current stance on meta-ros
implementation...?

Thank You

Rajath C S,
9964182112,
Bangalore - 560060

> I am sorry but I don't understand your question. I am the maintainer of
> meta-ros and the author of this documentation on the ROS wiki. I try to
> keep meta-ros running on the current master branch of [...]

Rajath C S,
9964182112,
Bangalore - 560060

wrote:

bulwahn, 2018-03-27:
As this is an open-source community project, I usually provide some basic support through github and discourse.ros.org. However, I assume that as a maintainer I can only provide some hints and the user needs to actually figure out the problem and resolve the issue in his/her build configuration.
I prefer public communication for the sake of better support in the community, and I only discuss very limited personal topics in private email threads. For this current issue, I don't see the need for   a private discussion. If you still feel that a private email thread would be required, you can find my email on my github profile or in the recent git commit messages.

bulwahn, 2018-04-03:
Question was resolved in non-public email.

closed 2018-04-03 by bulwahn

jarviscrypter, 2018-04-06:
1. i'd like to know does the process being followed in this link lead us to
installation of ROS-kinetic / indigo or others, hope i am clear..? that is
want to know if the process of including and compiling meta-ros layer for
yocto equal to install ros on ubuntu i mean in literal sense

http://wiki.ros.org/hydro/Installation/OpenEmbedded

2. Is the wxwidgets_2.9.5.bb recipe important because i am getting the
below error when 'bitbake roslaunch' is used

ERROR: ExpansionError during parsing
/home/jarvis/poky/meta-ros/recipes-extended/wxwidgets/wxwidgets_2.9.5.bb
| ETA:  0:02:07
Traceback (most recent call last):
bb.data_smart.ExpansionError: Failure expanding variable CXXFLAGS[:=],
expression was ${@oe_filter_out('-fvisibility-inlines-hidden', ' -O2 -pipe
-g -feliminate-unused-debug-types
-fdebug-prefix-map=/home/jarvis/poky/openembedded-core/build/tmp-glibc/work/i586-oe-linux/wxwidgets/2.9.5-r0=/usr/src/debug/wxwidgets/2.9.5-r0
-fdebug-prefix-map=/home/jarvis/poky/openembedded-core/build/tmp-glibc/work/i586-oe-linux/wxwidgets/2.9.5-r0/recipe-sysroot-native=
-fdebug-prefix-map=/home/jarvis/poky/openembedded-core/build/tmp-glibc/work/i586-oe-linux/wxwidgets/2.9.5-r0/recipe-sysroot=
-fvisibility-inlines-hidden', d)} which triggered exception NameError: name
'oe_filter_out' is not defined

Thanks

jarviscrypter, 2018-04-06:
Hi Lukas,

So, I tried the meta-ros inclusion again on my system and even this time i faced the similar error of wxwidgets_2.9.5.bb expansion error and as you suggested me in earlier comments i removed the respective recipe and performed the build again for which it threw the following error

"ERROR: No recipes available for:
  /home/jarvis/poky/meta-ros/recipes-devtools/python/python-nose_1.2.1.bbappend
  /home/jarvis/poky/meta-ros/recipes-devtools/python/python-numpy_1.7.0.bbappend
"
i will try including those python related recipes and build again, **i'd also like to attach the complete details of previous error as asked (since i had'nt sent it previously)**

`ERROR: ExpansionError during parsing /home/jarvis/poky/meta-ros/recipes-extended/wxwidgets/wxwidgets_2.9.5.bb                          | ETA:  0:02:07
Traceback (most recent call last):
bb.data_smart.ExpansionError: Failure expanding variable CXXFLAGS[:=], expression was ${@oe_filter_out('-fvisibility-inlines-hidden', ' -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/jarvis/poky/openembedded-core/build/tmp-glibc/work/i586-oe-linux/wxwidgets/2.9.5-r0=/usr/src/debug/wxwidgets/2.9.5-r0 -fdebug-prefix-map=/home/jarvis/poky/openembedded-core/build/tmp-glibc/work/i586-oe-linux/wxwidgets/2.9.5-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/jarvis/poky/openembedded-core/build/tmp-glibc/work/i586-oe-linux/wxwidgets/2.9.5-r0/recipe-sysroot=  -fvisibility-inlines-hidden', d)} which triggered exception NameError: name 'oe_filter_out' is not defined

Summary: There was 1 ERROR message shown, returning a non-zero exit code.
`
I'd like to know some solving pointers for this specific error

Thanks,
Rajath

jarviscrypter, 2018-04-06:
i understand that this isnt specific to ros queries but i do have posted
the same at the ROS mailing-list and to ROS maintainers and the issue is to
be yet resolved perfectly.

Thanks

Rajath C S,
9964182112,
Bangalore - 560060

On Fri, Apr 6, 2018 at 12:34 PM, Rajath C S <rajathcs.1996@gmail.com> wrote:

> 1. i'd like to know does the process being followed in this link lead us
> to installation of ROS-kinetic / indigo or others, hope i am clear..? that
> is want to know if the process of including and [...]

rojkov, 2018-04-06:
> `ERROR: ExpansionError during parsing /home/jarvis/poky/meta-ros/
> recipes-extended/wxwidgets/wxwidgets_2.9.5.bb | ETA: 0:02:07
> Traceback (most recent call last):
> bb.data_smart.ExpansionError: [...]
Can you spot the exception's message "NameError: name 'oe_filter_out' is
not defined"? This is the first pointer. I presume the prefix "oe_" stems
from "OpenEmbedded", so there's a chance that this is a function meant to
be defined in OpenEmbedded-Core. Try to find where this function is
documented. Usually Google Search helps with it. It might happen that the
function got deprecated and was removed from the core. In such case there
must be a replacement available.

Or it's your project that was configured improperly and the function was
not imported in the recipe for some reason. Then check what's required to
make the function imported.