ros/meta-ros #591: Application development work-flow with catkin (SDK)

flplv, 2018-04-27:
Hello guys!
I wonder how is the support for the application development work-flow regarding catkin and the yocto sdk?

I mean, what is the recommended approach from you guys? 

Just to clarify, the application development work-flow is this:
- a developer writes his application, that is an catkin package.
- it sources the yocto sdk
- calls catkin_make
- and the result is a cross compiled catkin package that he can deploy to the target hardware.

flplv, 2018-04-27:
@bulwahn My friend, I just saw you have a PR with SDK support for catkin: https://github.com/bmwcarit/meta-ros/pull/590, Thanks! 

May I ask, without the SDK, what is the flow you guys are following? Compile on target? Compile on build server only?
Best!

bulwahn, 2018-04-30:
Without the SDK support, the only supported use case in meta-ros is to write a recipe for each ROS package, cross-compile every package into an image and flash that image onto the target.

With the SDK support in #590, it should be possible to compile on the target and to use yocto SDK as you described above. Also, you find a good step-by-step documentation for the SDK feature in #492 (in a comment from dominiquehunziker on June 3rd, 2017).

We would like to merge #590, but we, the meta-ros maintainers, currently don't have the time to test the SDK support feature. You would really help us out if you could test the SDK feature and provide a short step-by-step explanation what you did and that it was successful. Then we can merge #590, as we know that it works and we have a test that we can reproduce in case we break it at some point. Once it is merged in master for indigo, I will also make it available for kinetic.

flplv, 2018-05-22:
We have tested the master of this branch with the catkin_make support in the SDK for a while by now, with ROSDISTRO changed to Kinetic and it is working fine.
Thanks a lot to the team here.

closed 2018-05-22 by flplv

AnneSt, 2018-08-27:
I build the sdk with ros-indigo and yocto-sumo, python 2.7 for raspberry. I had problems with python. The executable "python" is not found in SDK. Did you test the following [commit](https://github.com/bmwcarit/meta-ros/commit/c017174e78a347c3bf37186781b2ae516fb64678) with python2?
(See changes in  "recipes-ros/catkin/catkin-runtime_0.6.19.bb")

If I revert this change, I will have no problems.