ros/meta-ros #592: recent upgrade to boost 1.67.0 breaks ROS library builds

dominiquehunziker, 2018-05-05:
The recent upgrade to boost 1.67.0 in openembedded-core breaks the build due to a change in boost's date_time module. Namely, the constructor for sub-second durations now has a [check in place that the argument is of integral type](https://github.com/boostorg/date_time/blob/boost-1.67.0/include/boost/date_time/time_duration.hpp#L285-L286).
Before, the ROS code relied on an implicit conversion of floating point to integral type.

bulwahn, 2018-05-14:
After the first two preliminary patches were applied with #590, these four further packages need to be addressed:
```
Summary: 4 tasks failed:
  /home/lukas/repositories/github.com/bmwcarit/meta-ros/recipes-ros/actionlib/actionlib_1.11.13.bb:do_compile
  /home/lukas/repositories/github.com/bmwcarit/meta-ros/recipes-ros/bond-core/bondcpp_1.7.20.bb:do_compile
  /home/lukas/repositories/github.com/bmwcarit/meta-ros/recipes-ros/vision-opencv/cv-bridge_1.11.16.bb:do_configure
  /home/lukas/repositories/github.com/bmwcarit/meta-ros/recipes-devtools/yaml-cpp/yaml-cpp_0.5.3.bb:do_compile
```
(This is mainly for me to remember the current state or someone to check if we are aware of the problems users might encounter.)

bulwahn, 2018-05-14:
For the actionlib issue, this patch is already available upstream, which might solve the issue:
https://github.com/ros/actionlib/commit/677e952fcbfe49e6e4c5b835357f88740f49c6ff

shr-project, 2018-05-15:
See the fixes in:
https://github.com/shr-project/meta-ros/commits/jansa/master

I've few more changes in that branch to fix the build with gcc8 (from http://lists.openembedded.org/pipermail/openembedded-core/2018-May/150461.html).

I haven't sent them in PR, because I cannot properly test them in runtime, but feel free to test the changes from there and apply them as you wish.

bulwahn, 2018-05-17:
@shr-project I had a look at your patches, and now provided a pull request for boost-1.67 update issues. Could you possibly report back the `boost_python3 is not found` issue? To reproduce it, just switch on `ROS_USE_PYTHON3 = "yes"` and `bitbake cv-bridge`. I believe that is a general issue with the boost recipe.

bulwahn, 2018-05-17:
You need to include `ROS_USE_PYTHON3 = "yes"` in your local.conf, and then you should be able to see camera-calibration-parsers and cv-bridge failing in do_configure.

bulwahn, 2018-06-30:
The boost_python3 issue was reported in the yocto project bugzilla: https://bugzilla.yoctoproject.org/show_bug.cgi?id=12833

herb-kuta-lge, 2019-07-02:
No longer occurs in the [thud-draft] branch. @dominiquehunziker < Can this issue be closed?

closed 2019-11-04 by herb-kuta-lge