ros/meta-ros #596: Kinetic official support of meta-ros

flplv, 2018-05-22:
Hey guys, 
What is the official kinetic release of the meta-ros? Would it be https://github.com/bulwahn/meta-ros/tree/kinetic-experimental?

bulwahn, 2018-05-26:
There is no official release of kinetic yet, but I am working on that. I currently have a kinetic branch that is almost ready for being announced official. If you would like to help me with the last few task, let me know.

bulwahn, 2018-05-26:
By the way, I am currently on vacation. Once I am back in a week, I can share the current state of my branch with you.

krisklau, 2018-06-21:
Hi! I am also quite interested on getting kinetic up and running on our oe-build, and would be happy to help.

bulwahn, 2018-06-22:
Help is greatly appreciated. You can find my current state of work at this personal experimental branch:

https://github.com/bulwahn/meta-ros/tree/kinetic-experimental-v4-work-in-progress-v1

As a first step, please try to build it with `bitbake core-image-ros-world -k`

There are currently two open topics:
- the bullet recipe does not work correctly yet, and hence, map-server does not build.
- the whole branch is based on a before-sdk-support baseline, we would need to rebase that to master, i.e., I would need to know which patches shall be dropped and which patches are the new ones. If you want to do this, let me know and I will provide details. I tried already once and failed to do this and aborted it. Now I have a better idea what to do.

It might be that you need to cherry-pick some recent commit from master for gcc8 support.
I will probably a work-in-progress-v2 branch on the weekend, look out and be prepared for a new version of this branch.

flplv, 2018-07-11:
Thanks @bulwahn. Yes, we found the hard way that we had to update the versions :laughing: the commit is not in the fork yet and we only changed the few packages we use versions, thanks for pin pointing.

I do have interest in having a full ROS deployed soon, instead of only Core. Then your branch will be the way to go. I hope we get the opportunity to test most of packages from your Kinetic branch soon, with the sdk too.

Best.

John-Ingersoll, 2018-11-20:
It has been 6 Months since the last update on this issue.  Is the ongoing or been shelved?

flplv, 2018-11-20:
This is what we have running in our setup:

We forked meta-ros and merged the sdk support branch,
Then we did a quick and dirty settings to convert the fork to kinetic
Finally we bumped down the versions of the packages we use (core, msgs, and other considerable close to core packsges) (it is not in the fork)

If u’d like I can merge the missing parts of my fork so you could use the same setup. But it is faaaaaar from a full support. (And bumped versions are already outdated, ketics krep evolving on main ppas)

Best

bulwahn, 2019-01-18:
@kevswims Yes, I have that SDK support into Kinetic. It is a large WIP private branch: it is as always the usual problem, it is 90% done but then I got interrupted. I actually have put very little time in meta-ros lately due to work in other projects. However, as you ask for it, I will spend my free time on the weekend on work of this branch and then publish whatever I have at the end of this weekend.

kevswims, 2019-01-18:
Thanks @bulwahn We are working on a new project and I would really like to move to Kinetic on it since Indigo is quickly approaching EOL. I will be happy to help with testing and working on the Kinetic branch once you get it released.

superna9999, 2019-03-12:
@bulwahn Gentil Ping about this, can you provide the remaining work to do ? what are the packages not building yet ? I'm quite sure this can be done collaborately

krisklau, 2019-06-17:
How is this development going?

herb-kuta-lge, 2019-06-28:
> I would really like to move to Kinetic on it since Indigo is quickly approaching EOL. I will be happy to help with testing and working on the Kinetic branch once you get it released.

We are planning to support **melodic**, at least at first, as it's listed as the recommended ROS 1 distro by http://wiki.ros.org/Distributions. (See https://github.com/ros/meta-ros/wiki/Superflore-OE-Recipe-Generation-Scheme#milestones for our plans.) But if the community would rather have **kinetic** be supported ahead of **melodic**, we can certain change our plans.

krisklau, 2019-06-28:
Definitely melodic if we can move straight to that! I did not know of this new development work, very glad to see continued effort for ros/ros 2 in oe.

herb-kuta-lge, 2019-07-02:
@flplv < Can this issue be closed?

flplv, 2019-07-03:
I believe you can close the issue @herb-kuta-lge, regardless I am aware you are doing what is possible to release the official kinetic or melodic with sdk support. 

Note to future readers: Supporting full ROS can be a big endeavourer due to the amount of packages existent in the whole ROS ecosystem, my recommendation to those who want to have ROS at Yocto level is to try to reduce the amount of packages to cross build and deploy. On my experience we can already have a pretty stable build with SDK support to the core packages, even though the official release is not there yet.

herb-kuta-lge, 2019-07-03:
> my recommendation to those who want to have ROS at Yocto level is to try to reduce the amount of packages to cross build and deploy.

FYI, we're handling this by generating recipes for all the packages in the index -- see https://github.com/ros/meta-ros/wiki/Superflore-OE-Recipe-Generation-Scheme .

closed 2019-07-03 by herb-kuta-lge