ros/meta-ros #599: cv-bridge configuration errors

Tal-eigen, 2018-06-11:
Hello, 

I'm trying to build meta-ros (master branch), cv-bridge do_configure throws the following error:

```
ERROR: cv-bridge-1.10.18-r0 do_configure: Function failed: do_configure (log file is located at /media/talha/mybook/quad-yocto-build-krogoth/build/tmp/work/cortexa9hf-neon-poky-linux-gnueabi/cv-bridge/1.10.18-r0/temp/log.do_configure.1524)
ERROR: Logfile of failure stored in: /media/talha/mybook/quad-yocto-build-krogoth/build/tmp/work/cortexa9hf-neon-poky-linux-gnueabi/cv-bridge/1.10.18-r0/temp/log.do_configure.1524
Log data follows:
| DEBUG: Executing python function sysroot_cleansstate
| DEBUG: Python function sysroot_cleansstate finished
| DEBUG: Executing shell function do_configure
| NOTE: cmake.bbclass no longer uses OECMAKE_BUILDPATH.  The default behaviour is now out-of-tree builds with B=WORKDIR/build.
| -- The C compiler identification is GNU 5.3.0
| -- The CXX compiler identification is GNU 5.3.0
| -- Check for working C compiler: /media/talha/mybook/quad-yocto-build-krogoth/build/tmp/sysroots/x86_64-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-gcc
| -- Check for working C compiler: /media/talha/mybook/quad-yocto-build-krogoth/build/tmp/sysroots/x86_64-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-gcc -- works
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Check for working CXX compiler: /media/talha/mybook/quad-yocto-build-krogoth/build/tmp/sysroots/x86_64-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++
| -- Check for working CXX compiler: /media/talha/mybook/quad-yocto-build-krogoth/build/tmp/sysroots/x86_64-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++ -- works
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| -- Using CATKIN_DEVEL_PREFIX: /media/talha/mybook/quad-yocto-build-krogoth/build/tmp/work/cortexa9hf-neon-poky-linux-gnueabi/cv-bridge/1.10.18-r0/devel
| -- Using CMAKE_PREFIX_PATH: /media/talha/mybook/quad-yocto-build-krogoth/build/tmp/sysroots/udooqdl/opt/ros/indigo;/media/talha/mybook/quad-yocto-build-krogoth/build/tmp/sysroots/udooqdl/usr;/media/talha/mybook/quad-yocto-build-krogoth/build/tmp/sysroots/x86_64-linux/opt/ros/indigo;/media/talha/mybook/quad-yocto-build-krogoth/build/tmp/sysroots/x86_64-linux/usr
| -- This workspace overlays: /media/talha/mybook/quad-yocto-build-krogoth/build/tmp/sysroots/udooqdl/opt/ros/indigo;/media/talha/mybook/quad-yocto-build-krogoth/build/tmp/sysroots/x86_64-linux/opt/ros/indigo
| -- Found PythonInterp: /media/talha/mybook/quad-yocto-build-krogoth/build/tmp/sysroots/x86_64-linux/usr/bin/python-native/python (found version "2.7.11")
| -- Using PYTHON_EXECUTABLE: /media/talha/mybook/quad-yocto-build-krogoth/build/tmp/sysroots/x86_64-linux/usr/bin/python-native/python
| -- Using default Python package layout
| -- Found PY_em: /media/talha/mybook/quad-yocto-build-krogoth/build/tmp/sysroots/x86_64-linux/usr/lib/python2.7/site-packages/em.pyc
| -- Using empy: /media/talha/mybook/quad-yocto-build-krogoth/build/tmp/sysroots/x86_64-linux/usr/lib/python2.7/site-packages/em.pyc
| -- Using CATKIN_ENABLE_TESTING: 0
| -- catkin 0.6.19
| -- Boost version: 1.60.0
| -- Found the following Boost libraries:
| --   python
| -- Found PythonLibs: /media/talha/mybook/quad-yocto-build-krogoth/build/tmp/sysroots/udooqdl/usr/lib/libpython2.7.so (found version "2.7.11")
| -- Configuring done
| CMake Error in src/CMakeLists.txt:
|   Imported target "opencv_xphoto" includes non-existent path
| 
|     "/media/talha/mybook/quad-yocto-build-krogoth/build/tmp/sysroots/udooqdl/include/opencv"
| 
|   in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:
| 
|   * The path was deleted, renamed, or moved to another location.
| 
|   * An install or uninstall procedure did not complete successfully.
| 
|   * The installation package was faulty and references files it does not
|   provide.
| 
| 
| 
| CMake Error in src/CMakeLists.txt:
|   Imported target "opencv_xphoto" includes non-existent path
| 
|     "/media/talha/mybook/quad-yocto-build-krogoth/build/tmp/sysroots/udooqdl/include/opencv"
| 
|   in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:
| 
|   * The path was deleted, renamed, or moved to another location.
| 
|   * An install or uninstall procedure did not complete successfully.
| 
|   * The installation package was faulty and references files it does not
|   provide.
| 
| 
| 
| -- Generating done
| CMake Warning:
|   Manually-specified variables were not used by the project:
| 
|     CMAKE_INSTALL_BINDIR
|     CMAKE_INSTALL_DATAROOTDIR
|     CMAKE_INSTALL_INCLUDEDIR
|     CMAKE_INSTALL_LIBDIR
|     CMAKE_INSTALL_LIBEXECDIR
|     CMAKE_INSTALL_LOCALSTATEDIR
|     CMAKE_INSTALL_SBINDIR
|     CMAKE_INSTALL_SHAREDSTATEDIR
|     CMAKE_INSTALL_SYSCONFDIR
| 
| 
| -- Build files have been written to: /media/talha/mybook/quad-yocto-build-krogoth/build/tmp/work/cortexa9hf-neon-poky-linux-gnueabi/cv-bridge/1.10.18-r0/build
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /media/talha/mybook/quad-yocto-build-krogoth/build/tmp/work/cortexa9hf-neon-poky-linux-gnueabi/cv-bridge/1.10.18-r0/temp/log.do_configure.1524)
ERROR: Task 6 (/media/talha/mybook/quad-yocto-build-krogoth/sources/meta-ros/recipes-ros/vision-opencv/cv-bridge_1.10.18.bb, do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2299 tasks of which 2298 didn't need to be rerun and 1 failed.
No currently running tasks (1919 of 2307)

Summary: 1 task failed:
  /media/talha/mybook/quad-yocto-build-krogoth/sources/meta-ros/recipes-ros/vision-opencv/cv-bridge_1.10.18.bb, do_configure
Summary: There was 1 WARNING message shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```
i reverted to vision_opencv 1.10.8 to check if the issue gets fixed but it persists. The error is quite clear, the imported cv module opencv_xphoto has set a wrong include path.

```
 Imported target "opencv_xphoto" includes non-existent path
| 
|     "/media/talha/mybook/quad-yocto-build-krogoth/build/tmp/sysroots/udooqdl/include/opencv"
| 
```

This should be 
` "/media/talha/mybook/quad-yocto-build-krogoth/build/tmp/sysroots/udooqdl/include/opencv"`

Following is my yocto build environment:

```
Build Configuration:
BB_VERSION        = "1.30.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "udooqdl"
DISTRO            = "poky"
DISTRO_VERSION    = "2.1.3"
TUNE_FEATURES     = "arm armv7a vfp thumb neon       callconvention-hard       cortexa9"
TARGET_FPU        = "hard"
meta              
meta-poky         = "HEAD:444dc2e99b3c3967d9f83380c34bb99077a6ffa8"
meta-oe           
meta-multimedia   
meta-python       
meta-networking   = "HEAD:55c8a76da5dc099a7bc3838495c672140cedb78e"
meta-fsl-arm      = "HEAD:e2254e7b2ded0c2b66b1226f879b3a6d52037b2d"
meta-fsl-arm-extra = "HEAD:2c28e636ec15c2cfd49bc9cebe0bbbcfde95bc7b"
meta-fsl-demos    = "HEAD:a165068f8a0d1cf29aabe4b4053f28be1c2aa492"
meta-rust         = "krogoth:bedc4ad57c01a6e1c9ce11d36ab51b2f2a9e0dac"
meta-udoo         = "HEAD:a01cfce52c52fc942eab4375cd249795ca24c53f"
meta-ros          = "master:619eab0099fbb7bf71c7ddb43e4b50a0450ca40c"

```

I scanned through the opencv layer and could'nt locate where xphoto is being defined as a target.

bulwahn, 2018-06-11:
I agree, this looks very strange. Did you have a look if there are recent commits to the opencv recipe concerning xphoto in master, which might not be backported to your poky version?

Do you modify opencv in some way, e.g., with a PACKAGECONFIG?

Leistungsabfall, 2018-06-18:
I was facing the same problem with the Krogoth release. Looks like the issue is caused by an incompatibility between opencv 3.1 (provided by meta-openembedded) and cv-bridge (master):

https://github.com/bmwcarit/meta-ros/commit/909662efe72b59d94e74884b62a113831ee659e3 sets cmake variable `ANDROID` to `ON`.
When cv-bridge tries to find opencv using `<sysroot>/usr/share/OpenCV/OpenCVConfig.cmake` the `OpenCV_INSTALL_PATH` variable is set incorrectly since `ANDROID` is turned on: https://github.com/opencv/opencv/blob/92387b1ef8fad15196dd5f7fb4931444a68bc93a/cmake/templates/OpenCVConfig.cmake.in#L112-L127 (note the different amount of `../`)

This is how I fixed it locally:
**opencv_3.1.bbappend:**

```
SRC_URI += "file://fix-install-path.patch"

# Fix for https://github.com/bmwcarit/meta-ros/issues/599
FILESEXTRAPATHS_prepend := "${THISDIR}/files:"

do_install_append() {
    # Fix for https://github.com/bmwcarit/meta-ros/issues/522#issuecomment-310878031
    cp -f ${D}${datadir}/OpenCV/OpenCVModules.cmake ${D}${datadir}/OpenCV/OpenCVModules_.cmake
}
```

**fix-install-path.patch:**
```
Index: git/cmake/templates/OpenCVConfig.cmake.in
===================================================================
--- git.orig/cmake/templates/OpenCVConfig.cmake.in
+++ git/cmake/templates/OpenCVConfig.cmake.in
@@ -114,7 +114,7 @@ get_filename_component(OpenCV_CONFIG_PAT
 
 if(NOT WIN32 OR ANDROID)
   if(ANDROID)
-    set(OpenCV_INSTALL_PATH "${OpenCV_CONFIG_PATH}/../../..")
+    set(OpenCV_INSTALL_PATH "${OpenCV_CONFIG_PATH}/../..")
   else()
     set(OpenCV_INSTALL_PATH "${OpenCV_CONFIG_PATH}/../..")
   endif()
```

bulwahn, 2018-06-19:
Should we turn off the ANDROID option, or should we use the proposed solution for those patches here? I am happy to merge any change for this issue, if it is well described and explained with a proper rationale, and it does not break the build with the current master branch of openembedded-core and meta-openembedded. Just open a pull request and I will comment with my expectations.

herb-kuta-lge, 2019-07-02:
No longer occurs in the [thud-draft] branch. @Tal-seven < Can this issue be closed?

closed 2019-08-13 by herb-kuta-lge