ros/meta-ros #611: Error running roscore

jaronski, 2018-10-26:
hi all,
i am trying to run roscore on a yocto meta-ros bitbaked image. i exported everything from the README i added the ROS_USE_PYTHON3 = yes and LICENSE_FLAGS_WHITELIST = "commercial"
when i run roscore i get the error:
Traceback (most recent call last):
  File "/opt/ros/indigo/bin/roscore", line 69, in <module>
    import roslaunch
  File "/opt/ros/indigo/lib/python3.5/site-packages/roslaunch/__init__.py", line 51, in <module>
    from . import param_dump as roslaunch_param_dump
  File "/opt/ros/indigo/lib/python3.5/site-packages/roslaunch/param_dump.py", line 40, in <module>
    import roslaunch.config
  File "/opt/ros/indigo/lib/python3.5/site-packages/roslaunch/config.py", line 45, in <module>
    import rospkg.distro
  File "/usr/lib/python3.5/site-packages/rospkg/distro.py", line 44, in <module>
    import yaml
  File "/usr/lib/python3.5/site-packages/yaml/__init__.py", line 284
    class YAMLObject(metaclass=YAMLObjectMetaclass):
                              ^
SyntaxError: invalid syntax

Can anyone point me in a direction solving this?

thanks and best regards

herb-kuta-lge, 2020-02-07:
@jaronski < Can this issue be closed?

herb-kuta-lge, 2020-02-21:
"Silence implies consent" (see http://wiki.c2.com/?SilenceImpliesConsent).

closed 2020-02-21 by herb-kuta-lge

aarsht7, 2023-05-11:
I had the same problem as there were multiple versions of python present on the system.

Solution: `ln -sf /usr/bin/python3 /usr/local/bin/python`

You can put it in ~/.bashrc to avoid it in future