ros/meta-ros #613: bitbake yaml-cpp error

jaloo2018, 2018-11-02:
ERROR: yaml-cpp-0.6.2-r0 do_package: QA Issue: yaml-cpp: Files/directories were installed but not shipped in any package:
  /usr/lib/cmake
  /usr/lib/cmake/yaml-cpp
  /usr/lib/cmake/yaml-cpp/yaml-cpp-config.cmake
  /usr/lib/cmake/yaml-cpp/yaml-cpp-config-version.cmake
  /usr/lib/cmake/yaml-cpp/yaml-cpp-targets-release.cmake
  /usr/lib/cmake/yaml-cpp/yaml-cpp-targets.cmake
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
yaml-cpp: 6 installed and not shipped files. [installed-vs-shipped]
ERROR: yaml-cpp-0.6.2-r0 do_package: Fatal QA errors found, failing task.
ERROR: yaml-cpp-0.6.2-r0 do_package: Function failed: do_package
ERROR: Logfile of failure stored in: /home/jaloo/backup/s32v234/src/fsl-auto-yocto-bsp/build_s32v234evb_vsdk/tmp/work/aarch64-fsl-linux/yaml-cpp/0.6.2-r0/temp/log.do_package.25518
ERROR: Task (/home/jaloo/backup/s32v234/src/fsl-auto-yocto-bsp/sources/meta-ros/recipes-devtools/yaml-cpp/yaml-cpp_0.6.2.bb:do_package) failed with exit code '1

shr-project, 2018-11-02:
You're probably using old oe-core. /usr/lib/cmake is automatically packaged to PN-dev with recent oe-core releases.

aizelauna, 2019-03-12:
To workaround this issue, you can create the file _recipes-devtools/yaml-cpp/yaml-cpp_0.6.2.bbappend_ in your layer with this content:
`FILES_${PN}-dev += "/usr/lib/cmake/yaml-cpp"`

herb-kuta-lge, 2019-07-02:
No longer happens in the [thud-draft] branch. @jaloo2018 < Can this issue be closed?

closed 2019-08-13 by herb-kuta-lge