ros/meta-ros #618: roscpp, boost 1.69 and thud

OYTIS, 2019-02-11:
Greetings,

boost 1.69 has got it to thud branch of oe-core, so roscpp is broken now (see https://github.com/bmwcarit/meta-ros/issues/615). Is there any chance compatibility with thud can be restored, or do we have to downgrade to sumo now?

Thanks!

aizelauna, 2019-02-27:
I confirm the issue. I have the same error on my side.

aizelauna, 2019-02-27:
As a workaround, I patched _roscpp_ to remove signals from _find_package_ directive. _roscpp_ uses now signals2 which one is a header only component:
https://stackoverflow.com/questions/27252213/cmake-can-not-find-boostsignals2

shr-project, 2019-03-18:
FYI: the upgrade will probably be reverted very soon from thud:
http://lists.openembedded.org/pipermail/openembedded-core/2019-March/280203.html

This issue is still valid for oe-core master (which will soon became warrior).

shr-project, 2019-07-31:
Fixed in legacy branch.

PR #621 (2019-07-03): tf, tf2, message-filters, roscpp: remove signals from find_package() calls to support boost 1.69
boost has been updated in Yocto thud branch to 1.69. This boost update removes the deprecated library boost::signals. 
Since several versions, it is replaced by boost::signals2 which is a header only library.
Ros code already uses signals2 library but it still has links to the older library in its CMake files. This commit removes those references as they are now useless and more importantly source of error at configure time.

This PR should fix #618  issue.
closed 2019-07-31 by shr-project