ros/meta-ros #619: rosrun / roslaunch issue

Tal-seven, 2019-02-14:
Hi, I have been able to successfully deploy meta-ros on an udoo-quad board. Following is the config of my poky distro:

```
Build Configuration:
BB_VERSION        = "1.30.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "udooqdl"
DISTRO            = "poky"
DISTRO_VERSION    = "2.1.3"
TUNE_FEATURES     = "arm armv7a vfp thumb neon       callconvention-hard       cortexa9"
TARGET_FPU        = "hard"
meta              
meta-poky         = "HEAD:73cc31c11a9d75a2f4204a9a8c6832c6e505a86a"
meta-oe           
meta-multimedia   
meta-python       
meta-networking   = "HEAD:55c8a76da5dc099a7bc3838495c672140cedb78e"
meta-fsl-arm      = "HEAD:e2254e7b2ded0c2b66b1226f879b3a6d52037b2d"
meta-fsl-arm-extra = "HEAD:2c28e636ec15c2cfd49bc9cebe0bbbcfde95bc7b"
meta-fsl-demos    = "HEAD:a165068f8a0d1cf29aabe4b4053f28be1c2aa492"
meta-udoo         = "HEAD:a01cfce52c52fc942eab4375cd249795ca24c53f"
meta-ros          = "kinetic-experimental-v4-work-in-progress-v1:c8a3fc6aeb03c8efe8c507ca05ce967577bf8594"
```

I have tested running default meta-ros packages nodes (robot_pose_ekf) and they work without any issues. I tried to create my own test package following the tutorial found here:

https://github.com/vmayoral/diving-meta-ros

This is the test package 

https://github.com/Tal-seven/test_launch

and this is the recipe that i added to meta-ros/recipes-ros layer

```
SUMMARY = "A test ros node to experiment with yocto meta-ros and ubuntu desktop ros"
SECTION = "devel"

LICENSE = "MIT"
LIC_FILES_CHKSUM = "file://package.xml;md5=5926e6607c8f5a1205dc9e75af582544"

inherit catkin

DEPENDS = "roscpp std-msgs"

SRCREV = "13d9e051b320ac0bd3f45cb1f5e86ba7c26bc805"
SRC_URI = "git://github.com/Tal-seven/test_launch.git"

S = "${WORKDIR}/git"

BB_STRICT_CHECKSUM = "0" 
```

When i install the package using opkg install or by creating a full image and try to launch the node. 

`roslaunch test_launch test_node.launch`

I get the following error

`ERROR: cannot launch node of type [test_launch/test_launch]: can't locate node [test_launch] in package [test_launch]`

I have test the same package with an x86 platform and it runs fine. 

I am not experienced with yocto/embedded linux development. So any help would be great. Regards

Tal-seven, 2019-02-16:
I have found an adhoc fix for now by manually installing and copying the missing executable and pkgconfig files to  ${ros_libdir}, below is the fix. 

```
do_install_append(){
  install -m 0744 ${S}/../devel/lib/pkgconfig/test_launch.pc ${D}${ros_libdir}/pkgconfig
  mkdir -p ${D}${ros_libdir}/test_launch
  install -m 0744 ${S}/../devel/lib/test_launch/test_launch ${D}${ros_libdir}/test_launch
}

//Not doing this results in a Bad RPATH error during package qa
do_after_install()  {
  cd ${D}${ros_libdir}/test_launch
  chrpath -d test_launch
}

addtask do_after_install after do_install before do_package
```

bulwahn, 2019-02-28:
Your CMakeLists.txt does not install your package properly; please look into CMake and catkin documentation and tutorials how to properly install a ROS package. When you done that properly, there is no need for your adhoc fix in the recipe and the recipe is back to the very simple state as most other recipes in meta-ros.

herb-kuta-lge, 2020-02-07:
@Tal-seven < Can this issue be closed?

Tal-seven, 2020-02-08:
Yes thank you

closed 2020-02-08 by Tal-seven