ros/meta-ros #620: Project rostime unable to find lpthread

GSlzr, 2019-02-25:
I followed all the steps described in the wiki, using ubuntu 16.04, and when running `bitbake core-image-ros-roscore` it fails with the following error:
`Project 'rostime' tried to find library '-lpthread'.  The library is neither a target nor built/installed properly.  Did you compile project 'cpp_common'? Did you find_package() it before the subdirectory containing
its code is included?` 

Am I missing a step when configuring the setup?

bulwahn, 2019-02-28:
Could you please provide your full Build Configuration? Then, I might be able to reproduce the issue you encountered.

GSlzr, 2019-02-28:
After running the `oe-init-build-env` I edited the `bblayers.conf` and `local.conf`.
`bblayers.conf`
`# LAYER_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
LCONF_VERSION = "7"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  <workspace_path>/openembedded-core/meta \
  <workspace_path>/openembedded-core/meta-openembedded/meta-oe \
  <workspace_path>/openembedded-core/meta-openembedded/meta-python \
  <workspace_path>/openembedded-core/meta-openembedded/meta-multimedia \
  <workspace_path>/openembedded-core/meta-ros \
  "
`
`local.conf` I kept the generated one and added at the bottom
`BB_NUMBER_THREADS = "1"
PARALLEL_MAKE = "-j1"`

bulwahn, 2019-02-28:
When you start bitbake, it shows the "Build configuration" right at the start. This includes all information that I would need to have a fair chance to reproduce it. Can you provide that?

GSlzr, 2019-02-28:
Here it is:

`Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-16.04"
TARGET_SYS           = "i586-oe-linux"
MACHINE              = "qemux86"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "m32 i586"
TARGET_FPU           = ""
meta                 = "master:d2b1a889ef8fb9e6a2fa3d9bfc3eaf6113db9b1f"
meta-oe              
meta-python          
meta-multimedia      = "master:2b5beadcdf371b925ba10be94f7b5f7767881f71"
meta-ros             = "master:72068b17e4192b51e09c8dc633805a35edac8701"
`

bulwahn, 2019-02-28:
Thanks, that is what I need.

GSlzr, 2019-03-12:
@bulwahn Hi, do you have any news? Is there anyway I can help with this?

Thanks!

counter-logic, 2019-04-07:
To fix it temporarily, look for all files that thorws error 'CMake Error at /home/...Config.cmake'. Edit the .cmake file and manually change -'lpthread' to 'pthread'. Also you might need to fork https://github.com/aizelauna/meta-ros to make this work as a related PR about boost library has not been merged yet. Otherwise change catkin version to 0.7.15+ according to my PR #624.

shr-project, 2019-07-31:
Fixed in legacy branch.

closed 2019-07-31 by shr-project