ros/meta-ros #622: ROS 2 recipes for crystal

huntjw, 2019-03-21:
Any plans to update ROS 2 recipes for the crystal release?

windelbouwman, 2019-06-11:
I'm not sure how this would work, but ROS 2 was officially released. Is it an idea to create specific versions of the ros2 recipes? At this point there are several ros2 recipes with no version, but a git version. Should we create pinned versions? Also, it might be good to have package groups for each ros release.

This is the release of ros at this point:

https://index.ros.org/doc/ros2/Installation/Dashing/

herb-kuta-lge, 2019-06-28:
Recipes for **crystal** are now available -- see https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions. Please let us know if you encounter any problems.

closed 2019-06-28 by herb-kuta-lge