ros/meta-ros #623: configuration error for catkin_runtime

d-wang03, 2019-04-02:
I have encountered a strange error on catkin-runtime and nativesdk-catkin-runtime during do_configure(), which says "/lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.28' not found"

This error first occurred after update openembedded-core to the latest, on either master or thud branch.

The detail Info are:
ERROR: Logfile of failure stored in: /home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/temp/log.do_configure.11840
Log data follows:
| DEBUG: Executing shell function do_configure
| NOTE: cmake.bbclass no longer uses OECMAKE_BUILDPATH.  The default behaviour is now out-of-tree builds with B=WORKDIR/build.
| -- The C compiler identification is GNU 8.3.0
| -- The CXX compiler identification is GNU 8.3.0
| -- Check for working C compiler: /home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/recipe-sysroot-native/usr/bin/x86_64-oesdk-linux/x86_64-oesdk-linux-gcc
| -- Check for working C compiler: /home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/recipe-sysroot-native/usr/bin/x86_64-oesdk-linux/x86_64-oesdk-linux-gcc -- works
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Check for working CXX compiler: /home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/recipe-sysroot-native/usr/bin/x86_64-oesdk-linux/x86_64-oesdk-linux-g++
| -- Check for working CXX compiler: /home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/recipe-sysroot-native/usr/bin/x86_64-oesdk-linux/x86_64-oesdk-linux-g++ -- works
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| -- Using CATKIN_DEVEL_PREFIX: /home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/devel
| -- Using CMAKE_PREFIX_PATH: /home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/recipe-sysroot/usr/local/oecore-x86_64/sysroots/x86_64-oesdk-linux/opt/ros/indigo;/home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/recipe-sysroot/usr/local/oecore-x86_64/sysroots/x86_64-oesdk-linux/usr;/home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/recipe-sysroot-native/usr/local/oecore-x86_64/sysroots/x86_64-oesdk-linux/opt/ros/indigo;/home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/recipe-sysroot-native/usr/local/oecore-x86_64/sysroots/x86_64-oesdk-linux/usr
| -- Found PythonInterp: /home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/recipe-sysroot-native/usr/bin/python3-native/python3 (found version "3.7.2")
| -- Using PYTHON_EXECUTABLE: /home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/recipe-sysroot-native/usr/bin/python3-native/python3
| -- Using default Python package layout
| -- Found PY_em: /home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/recipe-sysroot-native/usr/lib/python3.7/site-packages/em.py
| -- Using empy: /home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/recipe-sysroot-native/usr/lib/python3.7/site-packages/em.py
| -- Using CATKIN_ENABLE_TESTING: 0
| -- catkin 0.6.19
| /home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/recipe-sysroot-native/usr/bin/python3-native/python3: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/recipe-sysroot/usr/local/oecore-x86_64/sysroots/x86_64-oesdk-linux/usr/lib/libpython3.7m.so.1.0)
| /home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/recipe-sysroot-native/usr/bin/python3-native/python3: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/recipe-sysroot/usr/local/oecore-x86_64/sysroots/x86_64-oesdk-linux/usr/lib/libpython3.7m.so.1.0)
| Traceback (most recent call last):
|   File "/home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/build/catkin_generated/generate_cached_setup.py", line 22, in <module>
|     code = generate_environment_script('/home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/devel/env.sh')
|   File "/home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/catkin-0.6.19/cmake/../python/catkin/environment_cache.py", line 61, in generate_environment_script
|     output = subprocess.check_output([env_script, sys.executable, '-c', python_code])
|   File "/home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/recipe-sysroot-native/usr/lib/python3.7/subprocess.py", line 395, in check_output
|     **kwargs).stdout
|   File "/home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/recipe-sysroot-native/usr/lib/python3.7/subprocess.py", line 487, in run
|     output=stdout, stderr=stderr)
| subprocess.CalledProcessError: Command '['/home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/devel/env.sh', '/home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/recipe-sysroot-native/usr/bin/python3-native/python3', '-c', 'import os; print(dict(os.environ))']' returned non-zero exit status 1.
| CMake Error at cmake/safe_execute_process.cmake:11 (message):
|
|   execute_process(/home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/recipe-sysroot-native/usr/bin/python3-native/python3
|   "/home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/build/catkin_generated/generate_cached_setup.py")
|   returned error code 1
| Call Stack (most recent call first):
|   cmake/all.cmake:187 (safe_execute_process)
|   CMakeLists.txt:8 (include)
|
|
| -- Configuring incomplete, errors occurred!
| See also "/home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/build/CMakeFiles/CMakeOutput.log".
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /home/supervisor/diwang/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-catkin-runtime/0.6.19-r0/temp/log.do_configure.11840)
ERROR: Task (virtual:nativesdk:/home/supervisor/diwang/meta-ros/recipes-ros/catkin/catkin-runtime_0.6.19.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 822 tasks of which 444 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  virtual:nativesdk:/home/supervisor/diwang/meta-ros/recipes-ros/catkin/catkin-runtime_0.6.19.bb:do_configure
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

d-wang03, 2019-04-03:
I think I found the reason. when executing python scripts during do_configure(), the python interpreter search its dynamic libraries first in  "recipe-sysroot/usr/local/oecore-x86_64/sysroots/x86_64-oesdk-linux/usr/lib", which is the target sysroots, and linked to the incorrect libpython3.7m.so.1.0 and libc.so.6.  The correct library path is "recipe-sysroot-native/usr/local/oecore-x86_64/sysroots/x86_64-oesdk-linux/usr/lib", which is put behind.

This order was set by CMAKE_PREFIX_PATH, in catkin.bbclass. The origin is ：
-DCMAKE_PREFIX_PATH='${STAGING_DIR_HOST}${ros_prefix};${STAGING_DIR_HOST}${prefix};${STAGING_DIR_NATIVE}${ros_prefix};${STAGING_DIR_NATIVE}${prefix}' 

changing it to 
-DCMAKE_PREFIX_PATH='${STAGING_DIR_HOST}${ros_prefix};${STAGING_DIR_NATIVE}${prefix};${STAGING_DIR_HOST}${prefix};${STAGING_DIR_NATIVE}${ros_prefix}'

will fix the problem.

However, is there any side effects in the solution? Please help!

shr-project, 2019-07-03:
Yes, I'm seeing the same issue when the glibc version used host and target doesn't match:

`/OE/build/oe-core/tmp-glibc/work/core2-64-oe-linux/catkin/0.6.19-r0/devel/env.sh /OE/build/oe-core/tmp-glibc/work/core2-64-oe-linux/catkin/0.6.19-r0/recipe-sysroot-native/usr/bin/python-native/python -C 'import os; print(dict(os.environ))'
/OE/build/oe-core/tmp-glibc/work/core2-64-oe-linux/catkin/0.6.19-r0/recipe-sysroot-native/usr/bin/python-native/python: /lib/x86_64-linux-gnu/libc.so.6: version 'GLIBC_2.28' not found (required by /OE/build/oe-core/tmp-glibc/work/core2-64-oe-linux/catkin/0.6.19-r0/recipe-sysroot/usr/lib/libpython2.7.so.1.0)
/OE/build/oe-core/tmp-glibc/work/core2-64-oe-linux/catkin/0.6.19-r0/recipe-sysroot-native/usr/bin/python-native/python: /lib/x86_64-linux-gnu/libm.so.6: version 'GLIBC_2.29' not found (required by /OE/build/oe-core/tmp-glibc/work/core2-64-oe-linux/catkin/0.6.19-r0/recipe-sysroot/usr/lib/libpython2.7.so.1.0)`
`

changing prefix to ros_prefix for STAGING_DIR_NATIVE doesn't look like correct fix, looking at the default value:
`catkin/0.6.19-r0/devel/_setup_util.py:        CMAKE_PREFIX_PATH = '/OE/build/oe-core/tmp-glibc/work/core2-64-oe-linux/catkin/0.6.19-r0/recipe-sysroot/opt/ros/indigo;/OE/build/oe-core/tmp-glibc/work/core2-64-oe-linux/catkin/0.6.19-r0/recipe-sysroot/usr;/OE/build/oe-core/tmp-glibc/work/core2-64-oe-linux/catkin/0.6.19-r0/recipe-sysroot-native/opt/ros/indigo;/OE/build/oe-core/tmp-glibc/work/core2-64-oe-linux/catkin/0.6.19-r0/recipe-sysroot-native/usr'.split(';')`

changing the last entry to /opt/ros/indigo doesn't make much sense (as there is nothing in catkin/0.6.19-r0/recipe-sysroot-native/opt)

To find out what's wrong we would need to figure out how this variable then being used by catkin, maybe it sets LD_LIBRARY_PATH somewhere based on that and that trips loading the libpython from target sysroot instead of native one.

It will also probably pass if you do build for different architecture (e.g. qemuarm instead of qemux86-64 which is matching with your host arch, then the libpython in target RSS will be skipped as incompatible and the right one in native RSS probably used). I'll check this later.

shr-project, 2019-07-04:
I've verified that building it for qemuarm in the same setup works fine.

herb-kuta-lge, 2020-02-07:
@d-wang03 < Can this issue be closed?

closed 2020-02-21 by herb-kuta-lge