ros/meta-ros #632: Use ros2 meta layer without mcf script

windelbouwman, 2019-07-01:
Hi,

I tried to use the ros2 meta layer as released recently. I want to use these generated recipes without using the mcf script. I want to use it by including the meta-ros layer into my bblayers.conf.

My question is: what is the reason why DISTRO must be set in bblayers.conf? Equally the ROS_DISTRO variable must be set in this file. Can this be modified such that these variables must be set in the local.conf file where the other settings are located?

In general, it would be nice to add a description to the wiki how to use the layer without the mcf script.

herb-kuta-lge, 2019-07-01:
> Can this be modified such that these variables must be set in the local.conf file where the other settings are located?

We were unable to think a way to do this given the design goals of allowing multiple ROS distributions to be built from the same branch and allowing webOS OSE as a `DISTRO`. If someone can, we'd be happy to adopt it.

> In general, it would be nice to add a description to the wiki how to use the layer without the mcf script.

We encourage the use of `mcf` because it provides a mechanism to pin the commits that are fetched from the repos and it ensures that the layers appear in `BBLAYERS` in the correct order. However, if someone from the community wishes to contribute and maintain instructions for using the layer without `mcf`, we'd be happy to include them.

windelbouwman, 2019-07-01:
Okay, that's interesting. The reason for me not wanting to use mcf is that I already have a vendor yocto system, and one can only have one framework at a time.

So my reasoning on using the layer with only variables is this:
- include the layer in bblayers.conf
- Set both DISTRO and ROS_DISTRO variables.
- Include packages into the image like rclpy
- bitbake some-image

Am I missing something?

windelbouwman, 2019-07-01:
How would one contribute such a guide? I cannot access the wiki page.

windelbouwman, 2019-07-01:
> We were unable to think a way to do this given the design goals of allowing multiple ROS distributions to be built from the same branch and allowing webOS OSE as a `DISTRO`. If someone can, we'd be happy to adopt it.

I think a solution for this problem would be to create sublayers. For example, within one branch, one could have the `meta-ros`, `meta-ros2` and `meta-ros-common` layers as seperate folders. A similar strategy can be observed in the meta-openembedded repository, which also contains several sublayers (https://git.openembedded.org/meta-openembedded/tree/)

herb-kuta-lge, 2019-07-01:
> I already have a vendor yocto system

If you could provide a pointer to its documentation, we could determine to make it do what `mcf` does.

> Set both DISTRO and ROS_DISTRO variables

Is there some reason you can't set them in `conf/bblayers.conf`?

> How would one contribute such a guide? I cannot access the wiki page.

Yeah, there doesn't seem to be a way to issue pull-requests for the GitHub wiki pages... Would it work to clone the wiki repo (using `git clone git@github.com:ros/meta-ros.wiki`) and then post the output from `git format-patch -1` once you've committed the changes locally?

> For example, within one branch, one could have the `meta-ros`, `meta-ros2` and `meta-ros-common` layers as seperate folders

How would this allow the support of all of the active distros of a `ROS_VERSION` in the same branch, eg, for ROS 2, `bouncy`, `crystal` and `dashing`?

windelbouwman, 2019-07-02:
> How would this allow the support of all of the active distros of a `ROS_VERSION` in the same branch, eg, for ROS 2, `bouncy`, `crystal` and `dashing`?

Okay, maybe it makes sense to create those sublayers: `meta-ros2-dashing`, `meta-ros2-crystal` etc. So fore each ROS release create a layer which can then be included. To me it makes more sense to select a certain ROS version through the inclusion of a layer then the setting of a variable. What essentially is happening now, is that a set of recipes is included given a variable. Given that a layer is a set of recipes, to me it makes sense to just use this concept of layers instead of enabling / disabling some recipes depending on variable.

herb-kuta-lge, 2019-07-02:
> To me it makes more sense to select a certain ROS version through the inclusion of a layer then the setting of a variable.

I'm convinced. @shr-project, @lokesku < What do you think? There would still need to be `meta-ros-common`, `meta-ros1`, and `meta-ros2` (sub)layers, but now that we're using `mcf`, having to include multiple layers to build a ROS distro doesn't introduce any additional (explicit) complexity for folks new to OE.

shr-project, 2019-07-02:
@herb-kuta-lge I like the idea of sublayers, even with the ros distro included (so meta-ros2-dashing instead and meta-ros2-crystal instead of just meta-ros).

davidlinn, 2019-07-03:
I'm doing a similar thing @windelbouwman . Trying to add meta-ros dashing as a layer to an existing project.

What I've done so far is build the existing project normally to generate sysroots, added meta-ros as a layer, added ROS_DISTRO = "dashing" and DISTRO = "ros2" to bblayers.conf, and re-run our normal build procedure, correcting dependency errors one-by-one. Adding -k to the bitbake command enables a build that continues through errors so that you can fix more than one dependency error at one time.

How are you going about this?

shr-project, 2019-07-03:
@davidlinn What kind of dependency errors are you seeing?

windelbouwman, 2019-07-03:
@davidlinn I did the exact same thing. Points to note:

- the order of the layers matter. I noticed that the meta-ros layer adds a newer version of python, so depending on your layer position, you might have the wrong version. The same holds for other packages.
- Second thing I did was switch from the poky distribution to the openembbeded-core distribution. Since the mcf script generated a set of layers for openembedded-core, and it's not yet set in stone what we're gonna use, I switched to that.

After I did this, the image build fine without patching (I included rclpy as IMAGE_INSTALL_append).

tfoote, 2019-07-11:
> How would one contribute such a guide? I cannot access the wiki page.

Yeah, this is a general issue with GitHub wikis. I would recommend moving the content into a doc folder on the repository and setup github pages to automatically render that folder instead of using the wiki. This will allow the documentation to go through the same review processes as the code.

lokesku, 2019-07-11:
?That sounds like a good idea. We will consider this in next iterations.

davidlinn, 2019-07-13:
@shr-project @herb-kuta-lge When compiling poco-vendor for aarch64 we get the following errors:

poky/build/tmp-glibc/work/aarch64-agl-linux/poco-vendor/1.2.0-1-r0/build/poco-1.8.0.1-release-prefix/src/poco-1.8.0.1-release/Foundation/src/zconf.h:445:14: fatal error: sys/types.h: No such file or directory
 #    include <sys/types.h>      /* for off_t */ 

 poky/build/tmp-glibc/work/aarch64-agl-linux/poco-vendor/1.2.0-1-r0/recipe-sysroot-native/usr/lib/aarch64-agl-linux/gcc/aarch64-agl-linux/7.3.0/include-fixed/limits.h:194:61: error: no include path in which to search for limits.h
 #include_next <limits.h>  /* recurse down to the real one */ 

Do you know how to resolve these? We cannot make changes to recipes as these are fetched from an external git as tar ball.

shr-project, 2019-07-13:
I'll try to reproduce it, but you can make changes to the source code easily with a .bbappend in your layer and a .patch file added to SRC_URI.

shr-project, 2019-07-15:
I've tried in local build and the poco from RSS was correctly detected for me:
work/aarch64-oe-linux/poco-vendor/1.2.0-1-r0/temp/log.do_configure
shows:
-- Found Poco 1.9.0

@davidlinn Please check your log.do_configure and exact build configuration (preferably in separate issue ticket).

vijaypalvai, 2019-07-16:
> And it's fetching this external one only if it didn't find the poco from RSS, why it's doing that should be investigated first, I'll have a look.
Could you please provide some more details on RSS, where do we look to see "why it didnot find poco from RSS"

FYI. we are seeing this error on compiling poco-vendor and it pulls "poco-1.8.0.1-release-prefix" 
the following lines shows it pulled from git page as follows

SRC_URI = "https://github.com/ros2-gbp/poco_vendor-release/archive/release/dashing/poco_vendor/1.2.0-1.tar.gz;downloadfilename=${ROS_SP}.tar.gz"
SRC_URI[md5sum] = "9561ff6bb7c46768531444787daa2bd8"
SRC_URI[sha256sum] = "ac61c044b172c2cd144fb22df5a466afefff5b980abfb5e245de67b24d867ed9"
S = "${WORKDIR}/poco_vendor-release-release-dashing-poco_vendor-1.2.0-1"

windelbouwman, 2019-07-16:
I think two of the most troublesome packages are indeed yaml-vendor and poco-vendor. These packages try to download something from the internet, which is not handy in a yocto build. The proper strategy for those two packages, would be to depend upon the poco and yaml packages provided by yocto itself.

windelbouwman, 2019-08-15:
Minor update: I succeeded in building ROS dashing onto the warrior release. The code lives here: https://github.com/windelbouwman/meta-ros2-generator

Points to note:
- I removed the libyaml-vendor by putting it in ros2_base.bbclass file. This removes libyaml-vendor usages from all generated packages. I did the same for poco-vendor and the connext and opensplice middlewares
- Some automatically generated recipes required some tuning. I did this with bbappend files instead of include files.
- Setting of the DISTRO is not required in the bblayers.conf file. You can even build without a DISTRO set.
- The dashing distribution is a moving target, it is more like the dashing branch, I noticed an update in the packages in the rosdistro repository.

herb-kuta-lge, 2019-12-12:
We're reached the milestone mentioned in my previous post. Please see the announcement [here](https://discourse.ros.org/t/published-revision-15-of-openembedded-build-instructions-meta-ros-milestone-8-reached/11903).

herb-kuta-lge, 2019-12-12:
@windelbouwman < Can this issue now be closed?

windelbouwman, 2019-12-12:
Hi @herb-kuta-lge , nice job! I would have to test this release, but I will probably not test this in the near future, since I now have a working setup, and I do not want to touch it. In my opinion this issue should not yet be closed, since the ROS_DISTRO still must be set. We can leave this issue open until milestone 12 I propose.

On the other hand, please feel free to close this issue as you see fit. Again: nice job, and thanks for making this happen!

herb-kuta-lge, 2019-12-20:
> We can leave this issue open until milestone 12 I propose.

closed 2020-09-21 by shr-project