ros/meta-ros #635: Build core-image-ros-world qa failure for eigen-conversions

RutujaShirali, 2019-07-24:
I am trying to bitbake core-image-ros-world using meta-ros from master. I keep seeing this error:

ERROR: eigen-conversions-1.11.9-r0 do_package_qa: QA Issue: eigen-conversions: The compile log indicates that host include and/or library paths were used.

Is anyone else also seeing this? Any ideas on how to resolve this?

herb-kuta-lge, 2019-07-29:
Which ROS distro are you targeting? If it's Indigo Igloo, then please use the `legacy` branch. `master` is in the process of upgraded to be able to build the current ROS 1 and ROS 2 distros (Crystal Clemmys, Dashing Diademata, and eventually Melodic Morenia and Bouncy Bolson) from generated recipes.  Draft versions of these changes are in the [`thud-draft`](https://github.com/ros/meta-ros/tree/thud-draft) branch. Please see this [wiki page](https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions) for instructions on how to build using the branch.

herb-kuta-lge, 2019-11-17:
`eigen-conversions` successfully builds for **melodic** using `[thud-draft]`. @RutujaShirali  < Can I close this issue?

herb-kuta-lge, 2019-12-03:
"Silence implies consent" (see http://wiki.c2.com/?SilenceImpliesConsent).

closed 2019-12-03 by herb-kuta-lge

m0oz, 2020-07-09:
this issue persists, when using the legacy branch

```
bitbake eigen-conversions
```
Fails with 
```bash
ERROR: eigen-conversions-1.11.9-r0 do_package_qa: QA Issue: eigen-conversions: The compile log indicates that host include and/or library paths were used.
         Please check the log '.../build_fb/tmp/work/arch/eigen-conversions/1.11.9-r0/temp/log.do_compile' for more information. [compile-host-path]
```

m0oz, 2020-07-09:
It seems like the recipe orocos-kdl_1.3.0 is messing with the include path