ros/meta-ros #637: diagnostic-aggregator-1.9.3-r0 do_compile: Function failed:generic_analyzer.h:49:10: fatal error: pluginlib/class_list_macros.hpp: No such file or directory |  #include <pluginlib/class_list_macros.hpp>

jiangfenghua123, 2019-08-20:
/home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++  -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"diagnostic_aggregator\" -Ddiagnostic_aggregator_EXPORTS -I/home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/diagnostics-1.9.3/diagnostic_aggregator/include -I/home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/recipe-sysroot/opt/ros/kinetic/include -I/home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/recipe-sysroot/opt/ros/kinetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I/home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/recipe-sysroot/usr/include -I/home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/diagnostics-1.9.3/diagnostic_aggregator/gtest-1.7.0/include  -march=armv8-a -mtune=cortex-a57.cortex-a53  --sysroot=/home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0=/usr/src/debug/diagnostic-aggregator/1.9.3-r0 -fdebug-prefix-map=/home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/recipe-sysroot=  -fvisibility-inlines-hidden  -march=armv8-a -mtune=cortex-a57.cortex-a53  --sysroot=/home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/recipe-sysroot -fPIC   -o CMakeFiles/diagnostic_aggregator.dir/src/aggregator.cpp.o -c /home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/diagnostics-1.9.3/diagnostic_aggregator/src/aggregator.cpp
| In file included from /home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/diagnostics-1.9.3/diagnostic_aggregator/src/generic_analyzer.cpp:37:0:
| /home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/diagnostics-1.9.3/diagnostic_aggregator/include/diagnostic_aggregator/generic_analyzer.h:49:10: fatal error: pluginlib/class_list_macros.hpp: No such file or directory
|  #include <pluginlib/class_list_macros.hpp>
|           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| compilation terminated.
| CMakeFiles/diagnostic_aggregator.dir/build.make:113: recipe for target 'CMakeFiles/diagnostic_aggregator.dir/src/generic_analyzer.cpp.o' failed
| make[2]: *** [CMakeFiles/diagnostic_aggregator.dir/src/generic_analyzer.cpp.o] Error 1
| make[2]: *** Waiting for unfinished jobs....
| In file included from /home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/diagnostics-1.9.3/diagnostic_aggregator/include/diagnostic_aggregator/ignore_analyzer.h:42:0,
|                  from /home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/diagnostics-1.9.3/diagnostic_aggregator/src/ignore_analyzer.cpp:37:
| /home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/diagnostics-1.9.3/diagnostic_aggregator/include/diagnostic_aggregator/generic_analyzer.h:49:10: fatal error: pluginlib/class_list_macros.hpp: No such file or directory
|  #include <pluginlib/class_list_macros.hpp>
|           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| compilation terminated.
| CMakeFiles/diagnostic_aggregator.dir/build.make:161: recipe for target 'CMakeFiles/diagnostic_aggregator.dir/src/ignore_analyzer.cpp.o' failed
| make[2]: *** [CMakeFiles/diagnostic_aggregator.dir/src/ignore_analyzer.cpp.o] Error 1
| In file included from /home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/diagnostics-1.9.3/diagnostic_aggregator/src/analyzer_group.cpp:37:0:
| /home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/diagnostics-1.9.3/diagnostic_aggregator/include/diagnostic_aggregator/analyzer_group.h:54:10: fatal error: pluginlib/class_loader.hpp: No such file or directory
|  #include "pluginlib/class_loader.hpp"
|           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
| compilation terminated.
| CMakeFiles/diagnostic_aggregator.dir/build.make:89: recipe for target 'CMakeFiles/diagnostic_aggregator.dir/src/analyzer_group.cpp.o' failed
| make[2]: *** [CMakeFiles/diagnostic_aggregator.dir/src/analyzer_group.cpp.o] Error 1
| In file included from /home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/diagnostics-1.9.3/diagnostic_aggregator/include/diagnostic_aggregator/discard_analyzer.h:42:0,
|                  from /home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/diagnostics-1.9.3/diagnostic_aggregator/src/discard_analyzer.cpp:37:
| /home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/diagnostics-1.9.3/diagnostic_aggregator/include/diagnostic_aggregator/generic_analyzer.h:49:10: fatal error: pluginlib/class_list_macros.hpp: No such file or directory
|  #include <pluginlib/class_list_macros.hpp>
|           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| compilation terminated.
| CMakeFiles/diagnostic_aggregator.dir/build.make:137: recipe for target 'CMakeFiles/diagnostic_aggregator.dir/src/discard_analyzer.cpp.o' failed
| make[2]: *** [CMakeFiles/diagnostic_aggregator.dir/src/discard_analyzer.cpp.o] Error 1
| In file included from /home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/diagnostics-1.9.3/diagnostic_aggregator/include/diagnostic_aggregator/aggregator.h:56:0,
|                  from /home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/diagnostics-1.9.3/diagnostic_aggregator/src/aggregator.cpp:37:
| /home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/diagnostics-1.9.3/diagnostic_aggregator/include/diagnostic_aggregator/analyzer_group.h:54:10: fatal error: pluginlib/class_loader.hpp: No such file or directory
|  #include "pluginlib/class_loader.hpp"
|           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
| compilation terminated.
| CMakeFiles/diagnostic_aggregator.dir/build.make:185: recipe for target 'CMakeFiles/diagnostic_aggregator.dir/src/aggregator.cpp.o' failed
| make[2]: *** [CMakeFiles/diagnostic_aggregator.dir/src/aggregator.cpp.o] Error 1
| make[2]: Leaving directory '/home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/build'
| CMakeFiles/Makefile2:976: recipe for target 'CMakeFiles/diagnostic_aggregator.dir/all' failed
| make[1]: *** [CMakeFiles/diagnostic_aggregator.dir/all] Error 2
| make[1]: Leaving directory '/home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/build'
| Makefile:132: recipe for target 'all' failed
| make: *** [all] Error 2
| WARNING: exit code 2 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/train/disk/wzh/H3/build/tmp/work/aarch64-poky-linux/diagnostic-aggregator/1.9.3-r0/temp/log.do_compile.10868)

jiangfenghua123, 2019-08-20:
not indigo distribution,For the kinetic distribution,

herb-kuta-lge, 2019-08-20:
@jiangfenghua123 < Which branch of `meta-ros` are you using?

jiangfenghua123, 2019-08-21:
For the kinetic distribution, you can use the experimental local branch at https://github.com/bulwahn/meta-ros/tree/kinetic-experimental-v3-alpha1. 

> @jiangfenghua123 < Which branch of `meta-ros` are you using?

For the kinetic distribution, you can use the experimental local branch at https://github.com/bulwahn/meta-ros/tree/kinetic-experimental-v3-alpha1.

herb-kuta-lge, 2019-08-29:
Can you migrate to **melodic**? If so, note that support for it will land soon in the `[thud-draft]` branch -- see [Milestone 6](https://github.com/ros/meta-ros/wiki/Superflore-OE-Recipe-Generation-Scheme#milestone-6-ros-core-with-webos-ose-on-qemux86-and-raspberrypi3-ros-1-melodic-and-oe-core-thud) of _Superflore OE Recipe Generation Scheme_.

herb-kuta-lge, 2019-09-10:
FYI: **Melodic** support has landed in `[thud-draft]` and it builds diagnostic-aggregator 1.9.3 successfully.

@jiangfenghua123 < Can this issue be closed?

closed 2019-11-04 by herb-kuta-lge