ros/meta-ros #638: OpenEmbedded warrior support

manuelvolk, 2019-08-23:
Hello everyone,
is there already a roadmap when meta-ros should be available for openembedded warrior? 
Are you already working on it?
What should be done to help?

herb-kuta-lge, 2019-08-29:
A milestone for initial support of OpenEmbedded **warrior** appears [here](https://github.com/ros/meta-ros/wiki/Superflore-OE-Recipe-Generation-Scheme#milestone-9-initial-support-for-oe-core-warrior), having been added in the recently published revision 41 of _Superflore OE Recipe Generation Scheme_.

sprhawk, 2019-10-30:
@herb-kuta-lge I got my own support for warrior in [my fork](https://github.com/sprhawk/meta-ros/tree/warrior-draft)
I want to make a pull request but there is no warrior branch for now.

herb-kuta-lge, 2019-11-04:
Thanks for offering to share your work. We have still not reached the [**warrior** support milestone](https://github.com/ros/meta-ros/wiki/Superflore-OE-Recipe-Generation-Scheme#milestone-9-initial-support-for-oe-core-warrior). Please be patient.

shr-project, 2020-01-13:
Warrior support available in https://github.com/ros/meta-ros/pull/658.

All changes are backwards compatible, so thud and warrior branches might be identical for a while.

shr-project, 2020-07-04:
Warrior support was merged with #658. 5 months ago, closing this ticket.

closed 2020-07-04 by shr-project