ros/meta-ros #639: Where should ROS packages be located on OpenEmbedded images?

herb-kuta-lge, 2019-08-24:
Should ROS packages be located under `/opt/ros/<ROS_DISTRO>` or folded into the rest of the OS under `/usr`? The current `meta-ros` implementation for **crystal** and **dashing** on `[thud-draft]` does the latter, but choosing the former would seem to guarantee that ROS 1 packages can be added to ROS 2 distros plus they would be where existing ROS users expect to find them. However, it would be, in some sense, a breaking change and would violate the [Filesystem Hierarchy Standard](http://refspecs.linuxfoundation.org/FHS_3.0/fhs/ch03s13.html) because ROS is not an add-on, but fundamental to the operation of the image. In spite of this, we favor `/opt/ros/<ROS_DISTRO>`. We would like feedback from the ROS community on this, especially from the current users of `meta-ros`, who would be impacted by the change.

fkromer, 2019-08-24:
`/opt/ros/<ROS_DISTRO>` seems to be a reasonable choice.

herb-kuta-lge, 2019-08-28:
> Also OSRF has registered ros with LANANA as a provider to be FHS compatible.

I finally found why this is so. The last sentence of 3.13.2. under <http://refspecs.linuxfoundation.org/FHS_3.0/fhs/ch03s13.html> states:

> Distributions may install and otherwise manage software in /opt under an appropriately registered subdirectory.

herb-kuta-lge, 2019-09-05:
So the decision is change where the ROS packages are located to be under `/opt/ros/<ROS_DISTRO>`. This will be done as part of the implementation of `meta-ros` layer version 3 -- see [Milestone 8](https://github.com/ros/meta-ros/wiki/Superflore-OE-Recipe-Generation-Scheme#milestone-8-implement-layer-version-3).

closed 2019-09-05 by herb-kuta-lge