ros/meta-ros #641: Would it be better to build ros-desktop, ros-rqt, ros-rviz, ros-qt5?

sprhawk, 2019-10-30:
Hi I am new to ROS,

I just set up meta-ros for my yocto build. I saw packagegroup-ros-world doesn't includes ros-desktop, ros-rqt, ros-rviz, ros-qt5, 

Would  they make ros dev easier on device local? or they are complete useless on device?

dkargin, 2019-11-03:
qt5 and rviz require quite a lot of things to be provided from BSP packages, like opengl/gles/egl drivers, x11/wayland support and so on. And every BSP package has their own way of providing these requirements. There are no guarantees that BSP provider did it in a good and compatible way, and there were numerous problems with them. So it was decided that all 'graphical' recipes are excluded. 

So packagegroup-ros-world contains packages which can be run in a headless mode and proven to be buildable on any platform (mostly, heh).

herb-kuta-lge, 2019-11-04:
We have a [milestone](https://github.com/ros/meta-ros/wiki/Superflore-OE-Recipe-Generation-Scheme#milestone-12-all-ros-2-packages-on-qemux86-and-raspberrypi3-ros-2-crystal-dashing-and-oe-core-thud-warrior) to support building all ROS 2 packages, including the graphical ones, but it will be a while before we get to it.

@sprhawk < Please re-open if you feel your issue hasn't been addressed.

closed 2019-11-04 by herb-kuta-lge