ros/meta-ros #642: meta-ros purposes

sancelot, 2019-11-12:
Hi,
If I understand this repository is dedicated to build ros for openembedded.

I am thinking about a way to build  ROS in a more generic way using cmake to download  needed packages and build them, in order being independt of the platform. 

Is it a good start , from here?

Regards,
S.Ancelot

gavanderhoorn, 2019-11-12:
Context: [ROS setup with cmake](https://answers.ros.org/question/337420) on ROS Answers.

herb-kuta-lge, 2019-11-14:
> Is it a good start , from here?

It could be if you are OK with using [**bitbake**](https://www.yoctoproject.org/docs/current/bitbake-user-manual/bitbake-user-manual.html) to download and build the packages instead of **cmake**.

herb-kuta-lge, 2019-12-03:
@sancelot < Can this issue be closed?

sancelot, 2019-12-03:
Yes, you can.

closed 2019-12-03 by herb-kuta-lge