ros/meta-ros #662: ecl-linear-algebra: fails in do_compile on x86-64 architecture

agorgeousday, 2020-01-13:
bitbake ecl-linear-algebra fails during compiling with:

```
[...]
| -- The C compiler identification is GNU 7.3.0
| -- The CXX compiler identification is GNU 7.3.0
| -- Check for working C compiler: ./build/tmp/work/aarch64-poky-linux/ecl-linear-algebra/0.62.1-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc
| -- Check for working C compiler: ./build/tmp/work/aarch64-poky-linux/ecl-linear-algebra/0.62.1-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc -- works
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Check for working CXX compiler: ./build/tmp/work/aarch64-poky-linux/ecl-linear-algebra/0.62.1-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++
| -- Check for working CXX compiler: ./build/tmp/work/aarch64-poky-linux/ecl-linear-algebra/0.62.1-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++ -- works
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| -- Using CATKIN_DEVEL_PREFIX: ./build/tmp/work/aarch64-poky-linux/ecl-linear-algebra/0.62.1-r0/devel
| -- Using CMAKE_PREFIX_PATH: ./build/tmp/work/aarch64-poky-linux/ecl-linear-algebra/0.62.1-r0/recipe-sysroot/opt/ros/melodic;./build/tmp/work/aarch64-poky-linux/ecl-linear-algebra/0.62.1-r0/recipe-sysroot/usr;./build/tmp/work/aarch64-poky-linux/ecl-linear-algebra/0.62.1-r0/recipe-sysroot-native/opt/ros/melodic;./build/tmp/work/aarch64-poky-linux/ecl-linear-algebra/0.62.1-r0/recipe-sysroot-native/usr
| -- This workspace overlays: ./build/tmp/work/aarch64-poky-linux/ecl-linear-algebra/0.62.1-r0/recipe-sysroot/opt/ros/melodic;./build/tmp/work/aarch64-poky-linux/ecl-linear-algebra/0.62.1-r0/recipe-sysroot-native/opt/ros/melodic
| -- Found PythonInterp: ./build/tmp/work/aarch64-poky-linux/ecl-linear-algebra/0.62.1-r0/recipe-sysroot-native/usr/bin/python-native/python (found version "2.7.13")
| -- Using PYTHON_EXECUTABLE: ./build/tmp/work/aarch64-poky-linux/ecl-linear-algebra/0.62.1-r0/recipe-sysroot-native/usr/bin/python-native/python
| -- Using default Python package layout
| -- Found PY_em: ./build/tmp/work/aarch64-poky-linux/ecl-linear-algebra/0.62.1-r0/recipe-sysroot-native/usr/lib/python2.7/site-packages/em.pyc
| -- Using empy: ./build/tmp/work/aarch64-poky-linux/ecl-linear-algebra/0.62.1-r0/recipe-sysroot-native/usr/lib/python2.7/site-packages/em.pyc
| -- Using CATKIN_ENABLE_TESTING: 0
| -- catkin 0.7.14
| -- Performing Test CXX11_COMPILER_FOUND
| -- Performing Test CXX11_COMPILER_FOUND - Success
| -- Configuring done
| CMake Error in src/lib/CMakeLists.txt:
|   Imported target "Sophus::Sophus" includes non-existent path
| 
|     "/usr/include/eigen3"
| 
|   in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:
| 
|   * The path was deleted, renamed, or moved to another location.
| 
|   * An install or uninstall procedure did not complete successfully.
| 
|   * The installation package was faulty and references files it does not
|   provide.
|   ERROR: Function failed: do_configure
```

This is on Arch Linux using the current cmake package version `3.8.2`, sophus `1.0.1` and boost `1.64.0`.

shr-project, 2020-01-13:
What meta-ros branch and ros release are you using?

ecl-linear-algebra/0.62.1 doesn't exist in current thud-drafts nor in legacy-v2 branch

There is 0.62.2 version in melodic now, so maybe you have some older revision from somewhere else? Please try to reproduce with current thud-draft branch.

agorgeousday, 2020-01-14:
> What meta-ros branch and ros release are you using?
> 
> ecl-linear-algebra/0.62.1 doesn't exist in current thud-drafts nor in legacy-v2 branch
> 
> There is 0.62.2 version in melodic now, so maybe you have some older revision from somewhere else? Please try to reproduce with current thud-draft branch.

The ROS version I used is `Melodic`
https://github.com/bulwahn/meta-ros/tree/melodic-experimental-v1

shr-project, 2020-01-14:
Then please report it in https://github.com/bulwahn/meta-ros, but last change in that repository was over 2 years ago, I doubt that it's still supported.

closed 2020-01-14 by shr-project