ros/meta-ros #666: kdl-parser: fails in do_compile on arch linux architecture

agorgeousday, 2020-01-14:
bitbake kdl-parser fails during compiling with:
```
| DEBUG: Executing shell function do_compile
| NOTE: VERBOSE=1 cmake --build ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/build -- -j 24
| ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/recipe-sysroot-native/usr/bin/cmake -H./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/kdl_parser-1.13.0/kdl_parser -B./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/build --check-build-system CMakeFiles/Makefile.cmake 0
| ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/recipe-sysroot-native/usr/bin/cmake -E cmake_progress_start ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/build/CMakeFiles ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/build/CMakeFiles/progress.marks
| make -f CMakeFiles/Makefile2 all
| make[1]: Entering directory './build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/build'
| make -f CMakeFiles/kdl_parser.dir/build.make CMakeFiles/kdl_parser.dir/depend
| make[2]: Entering directory './build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/build'
| cd ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/build && ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/recipe-sysroot-native/usr/bin/cmake -E cmake_depends "Unix Makefiles" ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/kdl_parser-1.13.0/kdl_parser ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/kdl_parser-1.13.0/kdl_parser ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/build ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/build ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/build/CMakeFiles/kdl_parser.dir/DependInfo.cmake --color=
| Dependee "./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/build/CMakeFiles/kdl_parser.dir/DependInfo.cmake" is newer than depender "./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/build/CMakeFiles/kdl_parser.dir/depend.internal".
| Dependee "./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/build/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/build/CMakeFiles/kdl_parser.dir/depend.internal".
| Scanning dependencies of target kdl_parser
| make[2]: Leaving directory './build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/build'
| make -f CMakeFiles/kdl_parser.dir/build.make CMakeFiles/kdl_parser.dir/build
| make[2]: Entering directory './build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/build'
| [ 25%] Building CXX object CMakeFiles/kdl_parser.dir/src/kdl_parser.cpp.o
| ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++  -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"kdl_parser\" -Dkdl_parser_EXPORTS -I./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/recipe-sysroot/usr/include -I./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/kdl_parser-1.13.0/kdl_parser/include -I./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/recipe-sysroot/usr/include/eigen3 -I/usr/include/eigen3 -I./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/recipe-sysroot/opt/ros/melodic/include -I./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/recipe-sysroot/opt/ros/melodic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp  -march=armv8-a -mtune=cortex-a53  --sysroot=./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/recipe-sysroot  -O2 -Os -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0=/usr/src/debug/kdl-parser/1.13.0-r0 -fdebug-prefix-map=./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/recipe-sysroot-native= -fdebug-prefix-map=./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/recipe-sysroot=  -fvisibility-inlines-hidden  -march=armv8-a -mtune=cortex-a53  --sysroot=./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/recipe-sysroot -fPIC   -std=c++11 -o CMakeFiles/kdl_parser.dir/src/kdl_parser.cpp.o -c ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/kdl_parser-1.13.0/kdl_parser/src/kdl_parser.cpp
| ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/kdl_parser-1.13.0/kdl_parser/src/kdl_parser.cpp: In function 'bool kdl_parser::treeFromXml(const tinyxml2::XMLDocument*, KDL::Tree&)':
| ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/kdl_parser-1.13.0/kdl_parser/src/kdl_parser.cpp:187:35: error: no matching function for call to 'urdf::Model::initXml(const tinyxml2::XMLDocument*&)'
|    if (!robot_model.initXml(xml_doc)) {
|                                    ^
| In file included from ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/kdl_parser-1.13.0/kdl_parser/src/kdl_parser.cpp:42:0:
| ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/recipe-sysroot/opt/ros/melodic/include/urdf/model.h:55:8: note: candidate: bool urdf::Model::initXml(TiXmlElement*)
|    bool initXml(TiXmlElement *xml);
|         ^~~~~~~
| ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/recipe-sysroot/opt/ros/melodic/include/urdf/model.h:55:8: note:   no known conversion for argument 1 from 'const tinyxml2::XMLDocument*' to 'TiXmlElement*'
| ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/recipe-sysroot/opt/ros/melodic/include/urdf/model.h:57:8: note: candidate: bool urdf::Model::initXml(TiXmlDocument*)
|    bool initXml(TiXmlDocument *xml);
|         ^~~~~~~
| ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/recipe-sysroot/opt/ros/melodic/include/urdf/model.h:57:8: note:   no known conversion for argument 1 from 'const tinyxml2::XMLDocument*' to 'TiXmlDocument*'
| CMakeFiles/kdl_parser.dir/build.make:65: recipe for target 'CMakeFiles/kdl_parser.dir/src/kdl_parser.cpp.o' failed
| make[2]: *** [CMakeFiles/kdl_parser.dir/src/kdl_parser.cpp.o] Error 1
| make[2]: Leaving directory './build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/build'
| CMakeFiles/Makefile2:70: recipe for target 'CMakeFiles/kdl_parser.dir/all' failed
| make[1]: *** [CMakeFiles/kdl_parser.dir/all] Error 2
| make[1]: Leaving directory './build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/build'
| Makefile:132: recipe for target 'all' failed
| make: *** [all] Error 2
| WARNING: exit code 2 from a shell command.
| ERROR: Function failed: do_compile (log file is located at ./build/tmp/work/aarch64-poky-linux/kdl-parser/1.13.0-r0/temp/log.do_compile.20984)

```
This is on Arch Linux using the current ROS version `Melodic`. The issue however occurs since kdl-parser in meta-ros at [kdl-parser_1.13.0.bb](https://github.com/bulwahn/meta-ros/tree/melodic-experimental-v1/recipes-ros/kdl-parser)

shr-project, 2020-01-14:
Then please report it in https://github.com/bulwahn/meta-ros, but last change in that repository was over 2 years ago, I doubt that it's still supported.

closed 2020-01-14 by shr-project

agorgeousday, 2020-01-14:
Since it doesn’t support create new issues at https://github.com/ros/meta-ros

If I porting ROS version Melodic on arch linux architecture. May I know which meta-ros branch is proper version for Melodic?

shr-project, 2020-01-14:
> Since it doesn’t support create new issues at https://github.com/bulwahn/meta-ros and that repository mentions I could list issues here.

It's a fork of older version of this repository and it mentions issues here, because it has the same README file as this repository.

> If I porting ROS version Melodic on arch linux architecture. May I know which meta-ros branch is proper version for Melodic?

What you mean by "arch linux"? aarch64 architecture? Because "arch linux" is also Linux distribution https://www.archlinux.org/.

thud-draft branch is currently the latest one and it supports newest melodic, see the documentation:
https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions

agorgeousday, 2020-01-15:
> What you mean by "arch linux"? aarch64 architecture? Because "arch linux" is also Linux distribution https://www.archlinux.org/.
Yes, I mean "aarch64" architecture.

> thud-draft branch is currently the latest one and it supports newest melodic, see the documentation:
> https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions
Thanks for sharing the info, I will try it later.