ros/meta-ros #667: Bitbake error - Nothing RPROVIDES 'nativesdk-libnsl2' (but virtual:nativesdk:/dev_ROS3/meta-ros/meta-ros-backports-warrior/recipes-devtools/python/python3_3.7.2.bb DEPENDS on)

agorgeousday, 2020-01-15:
I am on thud-draft branch, building a ros-image-world on "aarch64" architecture. My meta layers include the following:
```
1.meta
2.meta-poky
3.meta-yocto-bsp
4.meta-renesas/meta-rcar-gen3
5.meta-rcar/meta-rcar-gen3-adas
6.${META_ROS1_MELODIC_LAYER}
7.${META_ROS1_LAYER}
8.${META_ROS_COMMON_LAYER}
9.${META_ROS_BACKPORTS_WARRIOR_LAYER}
10.${META_PYTHON_LAYER} 
11.${META_OE_LAYER}
```

However, my bitbake does not go beyond the following
```
Parsing recipes: 100% |#####################################################################################################################################| Time: 0:00:14
Parsing of 3830 .bb files complete (0 cached, 3830 parsed). 4688 targets, 274 skipped, 11 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'nativesdk-libnsl2' (but virtual:nativesdk:/dev_ROS3/meta-ros/meta-ros-backports-warrior/recipes-devtools/python/python3_3.7.2.bb DEPENDS on or otherwise requires it). Close matches:
  nativesdk-libsdl
  nativesdk-libxslt
  nativesdk-libxml2
NOTE: Runtime target 'nativesdk-python3-git' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['nativesdk-python3-git', 'nativesdk-python3', 'nativesdk-libnsl2']
ERROR: Required build target 'ros-image-world' has no buildable providers.
Missing or unbuildable dependency chain was: ['ros-image-world', 'buildtools-tarball', 'nativesdk-python3-git', 'nativesdk-python3', 'nativesdk-libnsl2']
```

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
What could be a possible fix? Any suggestions would be appreciated.

shr-project, 2020-01-15:
buildtools-tarbal isn't part of regular ros-image-world we're building, is one of your extra layers adding dependency on it?

nativesdk-libnsl2 is available since oe-core commit d3589298f5ae0bdedbc1f265ed964841a9d11cfd which is included in our builds, are you using some much older revision of thud or what branches are you actually using?

Also ros-image-world currently doesn't build with melodic (because of other issues), see "Building Images" section in https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions

agorgeousday, 2020-01-16:
@shr-project 
Thanks for sharing the information. We actually source our own oe-init-build-env, but it is lack of nativesdk-libnsl2. I copied libnsl2_git.bb to our oe-core, and it seems issue fixed.

However, building a **ros-image-core** does not go beyond the following
```
Loading cache: 100% |#######################################################################################################################################| Time: 0:00:00
Loaded 4691 entries from dependency cache.
Parsing recipes: 100% |#####################################################################################################################################| Time: 0:00:03
Parsing of 3831 .bb files complete (3830 cached, 1 parsed). 4694 targets, 274 skipped, 11 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'libmodbus' (but /dev_ROS3/meta-ros/meta-ros1-melodic/recipes-core/packagegroups/packagegroup-ros-world-melodic.bb, /dev_ROS3/meta-ros/meta-ros1-melodic/generated-recipes/ipr-extern/ipr-extern_0.8.8-1.bb RDEPENDS on or otherwise requires it)
libmodbus was skipped: Recipe is blacklisted: Must specify PREFERRED_VERSION
libmodbus was skipped: Recipe is blacklisted: Must specify PREFERRED_VERSION
libmodbus was skipped: Recipe is blacklisted: Must specify PREFERRED_VERSION
libmodbus was skipped: Recipe is blacklisted: Must specify PREFERRED_VERSION
NOTE: Runtime target 'libmodbus' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['libmodbus']
NOTE: Runtime target 'packagegroup-ros-world-melodic' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-ros-world-melodic', 'libmodbus']
NOTE: Runtime target 'packagegroup-ros-world' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-ros-world', 'packagegroup-ros-world-melodic', 'libmodbus']
NOTE: Runtime target 'ros-image-world' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['ros-image-world', 'packagegroup-ros-world', 'packagegroup-ros-world-melodic', 'libmodbus']
ERROR: Required build target 'ros-image-core' has no buildable providers.
Missing or unbuildable dependency chain was: ['ros-image-core', 'ros-image-world', 'packagegroup-ros-world', 'packagegroup-ros-world-melodic', 'libmodbus']
```

shr-project, 2020-01-16:
> ERROR: Required build target 'ros-image-core' has no buildable providers.
> Missing or unbuildable dependency chain was: ['ros-image-core', 'ros-image-world', 'packagegroup-ros-world', 'packagegroup-ros-world-melodic', 'libmodbus']

I don't know what changes you did, but ros-image-core shouldn't depend on ros-image-world.

agorgeousday, 2020-01-22:
@shr-project, 

Thanks for your advise, I will close this issue first.

closed 2020-01-22 by agorgeousday