ros/meta-ros #674: How best to incorporate Wind River's work?

herb-kuta-lge, 2020-02-06:
Wind River has done some work on `meta-ros` and published it in these repos:
- <https://github.com/wind-river/meta-robot>
- <https://github.com/wind-river/wr-ros-demo>

How should it be added to `meta-ros`?

tfoote, 2020-02-07:
That's great to see. It would be great to find a way to integrate any changes or extensions they need to support their use cases into superflore so that it could be in the default meta-ros. It looks like they've been able to go a little deeper but narrower to get the TurtleBot running. I'll reach out to them and see if they have some suggestions.

rcwoolley, 2021-02-18:
I have created a new layer wr-ros whose purpose is to build the meta-ros "zeus" branch on Wind River Linux LTS 19 (based on Yocto Project "zeus" release).

The layer is posted here: [https://github.com/wind-river/wr-ros]

Many of the recipes are just overriding the bitbake QA checks to allow files in non-standard locations to Yocto.   However,  I was able to get rviz2 to successfully compile.   The bulk of this effort is to build the PyQt5 tools natively to enable cross-compilation.

I have verified that the rclcpp and rclpy examples run.   Next steps include re-integrate the Turtlebot3 drivers and fixing a dynamic loading problem with rviz_ogre_vendor that makes it look for RenderSystemGL.so.1.12.6 in the wrong directory (/usr/opt/rviz_ogre_vendor/lib/OGRE/)

Please let me know how I can best work with the meta-ros team to upstream these changes.

robwoolley, 2021-02-18:
I have created a new layer wr-ros whose purpose is to build the meta-ros "zeus" branch on Wind River Linux LTS 19 (based on Yocto Project "zeus" release).

The layer is posted here: [https://github.com/wind-river/wr-ros]

Many of the recipes are just overriding the bitbake QA checks to allow files in non-standard locations to Yocto. However, I was able to get rviz2 to successfully compile. The bulk of this effort is to build the PyQt5 tools natively to enable cross-compilation.

I have verified that the rclcpp and rclpy examples run. Next steps include re-integrate the Turtlebot3 drivers and fixing a dynamic loading problem with rviz_ogre_vendor that makes it look for RenderSystemGL.so.1.12.6 in the wrong directory (/usr/opt/rviz_ogre_vendor/lib/OGRE/)

Please let me know how I can best work with the meta-ros team to upstream these changes.