ros/meta-ros #681: Catkin build enviroment not available after building meta-ros from legacy branch

herr-kumar, 2020-02-24:
I am trying to use legacy version of meta-ros (gives out indigo package). I do not automatically get the catkin build enviroment (setup.sh, setup.bash, etc.) inside the SDK or the image.

Should I add something in the build/conf/local.conf file (I tried defining EXTRA_IMAGE_FEATURES += "ros-implicit-workspace") or should I use some patch to get this done. Any help would be really appreciated.

Thanks in advance,
Naveen

herb-kuta-lge, 2020-02-25:
@herr-kumar < Is there any chance you could migrate to melodic?

herr-kumar, 2020-02-26:
@herb-kuta-lge Thank you for the message. I am using the Indigo version (legacy branch) because the board I am using uses Yocto-Rocko.
Is it not possible to use Indigo and have the environment setup.

Can you let me know if these below settings are enough for set-up:
export ROS_ROOT=/opt/ros
export ROS_DISTRO=indigo
export ROS_PACKAGE_PATH=/opt/ros/indigo/share
export PATH=$PATH:/opt/ros/indigo/bin
export LD_LIBRARY_PATH=/opt/ros/indigo/lib
export PYTHONPATH=/opt/ros/indigo/lib/python2.7/site-packages
export ROS_MASTER_URI=http://localhost:11311
export CMAKE_PREFIX_PATH=/opt/ros/indigo
touch /opt/ros/indigo/.catkin

Also, regarding SDK cross-compiling to my target board, can you tell me how to do it?

herb-kuta-lge, 2020-02-27:
@herr-kumar < Sorry, but I have no knowledge of how to use what's on the `[legacy]` branch. Perhaps there's someone else in the community that can help.

herr-kumar, 2020-03-05:
@herb-kuta-lge May I know which branch of meta-ros should I checkout for melodic version of ROS?

shr-project, 2020-03-05:
@herr-kumar thud-draft, warrior-draft, zeus-draft or master-draft depending on which Yocto release you're using.

shr-project, 2020-07-28:
I'm going to assume that it works OK for you, please create new issue ticket if you find some issues later.

"Silence implies consent" (see http://wiki.c2.com/?SilenceImpliesConsent).

closed 2020-07-28 by shr-project