ros/meta-ros #683: ROS2 recipe for Foxy release

hyang5, 2020-04-10:
Is there any plan to support Foxy release?

herb-kuta-lge, 2020-04-16:
Yes! Our plan is to add support for it in [Milestone 14](https://github.com/ros/meta-ros/wiki/Superflore-OE-Recipe-Generation-Scheme#milestone-14-support-ros-2-foxy).

shr-project, 2020-07-04:
It was merged just now, please try it and let us know.

shr-project, 2020-07-28:
I'm going to assume that it works OK for you, please create new issue ticket if you find some issues later.

"Silence implies consent" (see http://wiki.c2.com/?SilenceImpliesConsent).

Today the foxy support was also added to branches for old OE releases thud, warrior and zeus.

closed 2020-07-28 by shr-project

hyang5, 2020-07-29:
> I'm going to assume that it works OK for you, please create new issue ticket if you find some issues later.
> 
> "Silence implies consent" (see http://wiki.c2.com/?SilenceImpliesConsent).
> 
> Today the foxy support was also added to branches for old OE releases thud, warrior and zeus.

Thanks. It works.