ros/meta-ros #684: build error timeout for foonathan-memory, libyaml-vendor and tinydir-vendor

avinashiitk, 2020-04-20:
I am getting below meta-ros build error.(output of **bitbake ros-image-core**)
Sharing my meta-ros configuration and build error.

-I do not doubt my "http_proxy" setting.
-"bitbake core-image-minimal" is working fine.

I also tried setting proxy at "~/.subversion/servers" but its not helping.
Error2 and Error3 below I can overcome with work around of doing wget or .zip and .tar.gz files.
however I could not find any workaround for Error1.

Error1:
foonathan-memory_git.bb..
-- Checking flags for C++17 - Found: -std=c++17
| -- Checking 13 features
| CMake Error at cmake/comp/comp_base.cmake:288 (message):
|   error downloading feature file cpp11_lang/alignas.cmake: "Couldn't connect
|   to server".  Check spelling of feature.
|
|     Trying 151.101.192.133:443...

Error2:
libyaml-vendor_1.0.0-1.bb
src='https://github.com/yaml/libyaml/archive/10c907871f1ccd779c7fccf6b81a62762b5c4e7b.zip'
| CMake Error at libyaml-10c9078-stamp/download-libyaml-10c9078.cmake:159 (message):
|   Each download failed!
|
|     error: downloading 'https://github.com/yaml/libyaml/archive/10c907871f1ccd779c7fccf6b81a62762b5c4e7b.zip' failed

Error3:
tinydir-vendor_1.1.1-1.bb
 -- Using src='https://github.com/cxong/tinydir/archive/1.2.4.tar.gz'
| CMake Error at tinydir-1.2.4-stamp/download-tinydir-1.2.4.cmake:159 (message):
|   Each download failed

Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "x86_64-oe-linux"
MACHINE              = "qemux86-64"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "m64 core2"
TARGET_FPU           = ""
DISTRO_NAME          = "OpenEmbedded"
ROS_DISTRO           = "eloquent"
ROS_VERSION          = "2"
ROS_PYTHON_VERSION   = "3"
meta-raspberrypi     = "zeus:85b75588fdedf72242870628f7c22f034d308740"
meta-ros2-eloquent
meta-ros2
meta-ros-common      = "zeus-draft:840ee20985c501555d46308ac0c16b67ee4ae253"
meta-python
meta-oe              = "zeus:e855ecc6d35677e79780adc57b2552213c995731"
meta                 = "zeus:f9739e0f58f5d8a5dc01fdb0efaa778af4edd671"

avinashiitk, 2020-04-20:
Note: I am following steps at "https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions" to reproduce above build error.

avinashiitk, 2020-04-20:
[yocto_build.log](https://github.com/ros/meta-ros/files/4504961/yocto_build.log)

shr-project, 2020-04-20:
{quote}
-I do not doubt my "http_proxy" setting.
{quote}

Are you able to download those archives manually? Without http_proxy being set? 

do_compile/do_configure shouldn't access network (as everything should be fetched with bitbake fetcher and respect PREMIRROR setting) - that's one of the reasons why we've changed those ExternalProjects calls with standalone recipe building the external dependency - but not in all cases and this might be side effect of that.

avinashiitk, 2020-04-21:
> Are you able to download those archives manually? Without http_proxy being set?

without http_proxy I am not able to download. 
I tried below 3 experiments but still I am facing same error.
experiment1:
add proxy settings to ~/.wgetrc 
(http_proxy = http://proxy-chain.intel.com:911/
  https_proxy = http://proxy-chain.intel.com:912)

experiment2:
add below options to all three .bb (bitbake files)
do_confiugre() {
export http_proxy=${http_proxy}
export https_proxy=${https_proxy}
}

Experiment3:
Add proxy settings to ~/.subversion/servers

Network timeout Errors we are seeing in below 3 bb files.
meta-ros2-eloquent/recipes-devtools/foonathan-memory/foonathan-memory_git.bb:do_configure
meta-ros2-eloquent/generated-recipes/libyaml-vendor/libyaml-vendor_1.0.0-1.bb:do_compile
meta-ros2-eloquent/generated-recipes/tinydir-vendor/tinydir-vendor_1.1.1-1.bb:do_compile

regards
Avinash

shr-project, 2020-04-23:
Please check your proxy configuration with curl as suggested here:
https://cmake.org/pipermail/cmake/2015-October/061850.html

shr-project, 2020-04-23:
With
```
do_compile_prepend() {
    bbwarn "http_proxy: $http_proxy"
}
```
added to `meta-ros/meta-ros2-eloquent/recipes-bbappends/libyaml-vendor/libyaml-vendor_%.bbappend`
```
$ export http_proxy=fail.for.me:80
$ bitbake -c compile libyaml-vendor
...
WARNING: libyaml-vendor-1.0.0-1-r0 do_compile: http_proxy: 
```

If I add `export http_proxy` before the do_compile_prepend, then it's set correctly:
```
WARNING: libyaml-vendor-1.0.0-1-r0 do_compile: http_proxy: fail.for.me:80
```
but curl doesn't seem to be using it (or at least doesn't fail).

When trying with curl-native (which will be used because cmake-native depends on it):
```
tmp-glibc/work/core2-64-oe-linux/libyaml-vendor/1.0.0-1-r0/recipe-sysroot-native/usr/bin/curl -v https://github.com/yaml/libyaml/archive/10c907871f1ccd779c7fccf6b81a62762b5c4e7b.zip
*   Trying 192.30.255.113:443...
* TCP_NODELAY set
* Connected to github.com (192.30.255.113) port 443 (#0)
* ALPN, offering http/1.1
* error setting certificate verify locations:
  CAfile: /jenkins/home/workspace/jansa/ros2-dashing-zeus/ros2-dashing-zeus/tmp-glibc/work/x86_64-linux/curl-native/7.66.0-r0/recipe-sysroot-native/etc/ssl/certs/ca-certificates.crt
  CApath: none
* Closing connection 0
curl: (77) error setting certificate verify locations:
  CAfile: /jenkins/home/workspace/jansa/ros2-dashing-zeus/ros2-dashing-zeus/tmp-glibc/work/x86_64-linux/curl-native/7.66.0-r0/recipe-sysroot-native/etc/ssl/certs/ca-certificates.crt
  CApath: none
```

That was because I was setting only http_proxy, not https_proxy, with https_proxy I can see the failure:
```
tmp-glibc/work/core2-64-oe-linux/libyaml-vendor/1.0.0-1-r0/recipe-sysroot-native/usr/bin/curl -q -I -k -v https://github.com/yaml/libyaml/archive/10c907871f1ccd779c7fccf6b81a62762b5c4e7b.zip
* Uses proxy env variable https_proxy == 'http://fail.for.me:80'
*   Trying 31.192.224.240:80...
* TCP_NODELAY set
* Connected to fail.for.me (31.192.224.240) port 80 (#0)
* allocate connect buffer!
* Establish HTTP proxy tunnel to github.com:443
> CONNECT github.com:443 HTTP/1.1
> Host: github.com:443
> User-Agent: curl/7.66.0
> Proxy-Connection: Keep-Alive
> 
* Recv failure: Connection reset by peer
* Received HTTP code 0 from proxy after CONNECT
* CONNECT phase completed!
* Closing connection 0
curl: (56) Recv failure: Connection reset by peer
```

Please notice that some of these are failing in do_compile (e.g. libyaml-vendor), not do_configure, so you need to make sure your proxy is exported there.

Long term solution would be to get rid of these ExternalProject_Add like we did in some cases:
https://github.com/ros/meta-ros/commit/e3585e5828afe256f29bf270c25afd016809fb84
https://github.com/ros/meta-ros/commit/01b7f28dcd4c9f43b9ec909ec74b1bc233d5bb84
or at least to fetch the ExternalProject sources in normal SRC_URI with bitbake fetcher.

shr-project, 2020-04-23:
Please try with the 3 commits added now for milestone 12 and let me know if something else is still failing for you.

If it works, then the same should be added at least to dashing as well.

avinashiitk, 2020-05-06:
Thanks for patch. Closing this issue.

closed 2020-05-06 by avinashiitk