ros/meta-ros #685: Failed to build  robot-state-publisher

clementlgl, 2020-04-21:
I'm trying to build `ros-image-world` on Zeus for a raspberrypi4-64.

For setup the build environnement I have following the wiki (https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions)

The compilation of the `robot-state-publisher` produces this error. 
```
DEBUG: Executing shell function do_compile
NOTE: VERBOSE=1 cmake --build /home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/build --target all -- -j 20
[1/4] /home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot-native/usr/bin/aarch64-oe-linux/aarch64-oe-linux-g++   -I/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/git/include -I/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot/usr/include/eigen3 -mcpu=cortex-a72+crc+crypto  --sysroot=/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0=/usr/src/debug/robot-state-publisher/2.2.5-1-r0                      -fdebug-prefix-map=/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0=/usr/src/debug/robot-state-publisher/2.2.5-1-r0                      -fdebug-prefix-map=/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -mcpu=cortex-a72+crc+crypto  --sysroot=/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot   -Wall -Wextra -std=gnu++14 -MD -MT CMakeFiles/robot_state_publisher.dir/src/joint_state_listener.cpp.o -MF CMakeFiles/robot_state_publisher.dir/src/joint_state_listener.cpp.o.d -o CMakeFiles/robot_state_publisher.dir/src/joint_state_listener.cpp.o -c /home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/git/src/joint_state_listener.cpp
FAILED: CMakeFiles/robot_state_publisher.dir/src/joint_state_listener.cpp.o 
/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot-native/usr/bin/aarch64-oe-linux/aarch64-oe-linux-g++   -I/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/git/include -I/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot/usr/include/eigen3 -mcpu=cortex-a72+crc+crypto  --sysroot=/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0=/usr/src/debug/robot-state-publisher/2.2.5-1-r0                      -fdebug-prefix-map=/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0=/usr/src/debug/robot-state-publisher/2.2.5-1-r0                      -fdebug-prefix-map=/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -mcpu=cortex-a72+crc+crypto  --sysroot=/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot   -Wall -Wextra -std=gnu++14 -MD -MT CMakeFiles/robot_state_publisher.dir/src/joint_state_listener.cpp.o -MF CMakeFiles/robot_state_publisher.dir/src/joint_state_listener.cpp.o.d -o CMakeFiles/robot_state_publisher.dir/src/joint_state_listener.cpp.o -c /home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/git/src/joint_state_listener.cpp
/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/git/src/joint_state_listener.cpp: In function 'int main(int, char**)':
/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/git/src/joint_state_listener.cpp:204:8: error: 'kdl_parser' has not been declared
  204 |   if (!kdl_parser::treeFromUrdfModel(model, tree)) {
      |        ^~~~~~~~~~
[2/4] /home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot-native/usr/bin/aarch64-oe-linux/aarch64-oe-linux-g++   -I/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/git/include -I/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot/usr/include/eigen3 -mcpu=cortex-a72+crc+crypto  --sysroot=/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0=/usr/src/debug/robot-state-publisher/2.2.5-1-r0                      -fdebug-prefix-map=/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0=/usr/src/debug/robot-state-publisher/2.2.5-1-r0                      -fdebug-prefix-map=/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -mcpu=cortex-a72+crc+crypto  --sysroot=/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot   -Wall -Wextra -std=gnu++14 -MD -MT CMakeFiles/joint_state_listener.dir/src/joint_state_listener.cpp.o -MF CMakeFiles/joint_state_listener.dir/src/joint_state_listener.cpp.o.d -o CMakeFiles/joint_state_listener.dir/src/joint_state_listener.cpp.o -c /home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/git/src/joint_state_listener.cpp
FAILED: CMakeFiles/joint_state_listener.dir/src/joint_state_listener.cpp.o 
/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot-native/usr/bin/aarch64-oe-linux/aarch64-oe-linux-g++   -I/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/git/include -I/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot/usr/include/eigen3 -mcpu=cortex-a72+crc+crypto  --sysroot=/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0=/usr/src/debug/robot-state-publisher/2.2.5-1-r0                      -fdebug-prefix-map=/home/bionic/claigl [... cut ...]
/home/bionic/claigle/Dev/Yocto/raspberrypi/ROS/BUILD-ros2-dashing-zeus/work/aarch64-oe-linux/robot-state-publisher/2.2.5-1-r0/git/src/joint_state_listener.cpp:204:8: error: 'kdl_parser' has not been declared
  204 |   if (!kdl_parser::treeFromUrdfModel(model, tree)) {
      |        ^~~~~~~~~~
ninja: build stopped: subcommand failed.
WARNING: exit code 1 from a shell command.
```

The compilation works with `rapsberrypi4` machine.
I think there's an incompatibility of this package with a 64-bit system.

Someone's already encountered the problem 

Thank you 
Clément

shr-project, 2020-04-21:
I've just built it for raspberrypi4-64 without an issue:
```
Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-oe-linux"
MACHINE              = "raspberrypi4-64"
DISTRO               = "ros2"
DISTRO_VERSION       = "2.dashing.20200319.5"
TUNE_FEATURES        = "aarch64 cortexa72 crc crypto"
TARGET_FPU           = ""
DISTRO_NAME          = "Robot Operating System 2 (ROS 2) with OpenEmbedded Linux"
ROS_DISTRO           = "dashing"
ROS_VERSION          = "2"
ROS_PYTHON_VERSION   = "3"
meta-raspberrypi     = "zeus:0e05098853eea77032bff9cf81955679edd2f35d"
meta-ros2-dashing    
meta-ros2            
meta-ros-common      
meta-ros-backports-dunfell = "zeus-draft:897334d61ea48c95512e945e0db6b536fba093d8"
meta-python          
meta-oe              = "zeus:9e60d30669a2ad0598e9abf0cd15ee06b523986b"
meta                 = "zeus:0743dcd0cd70dee87a49d3bcd017168352e60982"

```

clementlgl, 2020-04-21:
Thank you @shr-project 

```
Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "aarch64-oe-linux"
MACHINE              = "raspberrypi4-64"
DISTRO               = "ros2"
DISTRO_VERSION       = "2.dashing.20191213.3"
TUNE_FEATURES        = "aarch64 cortexa72 crc crypto"
TARGET_FPU           = ""
DISTRO_NAME          = "Robot Operating System 2 (ROS 2) with OpenEmbedded Linux"
ROS_DISTRO           = "dashing"
ROS_VERSION          = "2"
ROS_PYTHON_VERSION   = "3"
meta-raspberrypi     = "zeus:85b75588fdedf72242870628f7c22f034d308740"
meta-ros2-dashing    
meta-ros2            
meta-ros-common      = "zeus-draft:840ee20985c501555d46308ac0c16b67ee4ae253"
meta-python          
meta-oe              = "zeus:e855ecc6d35677e79780adc57b2552213c995731"
meta                 = "zeus:f9739e0f58f5d8a5dc01fdb0efaa778af4edd671"
```

I don't understand why my `DISTRO_VERSION` is different .. 
Maybe that's the problem.

shr-project, 2020-04-21:
I'm using newer version from https://github.com/ros/meta-ros/pull/679 and https://github.com/ros/meta-ros/pull/675 but checked that there were no changes to kdl-parser nor robot-state-publisher in them (and I'm pretty use it was building OK for me with older revisions as well).

Can you upload log.do_configure and log.do-compile files from ${WORKDIR}/temp?

clementlgl, 2020-04-21:
It's working. I just don't know why...
I'd clean the recipe (-c cleanall) several times and then recompile it, but always the same problem. I finally deleted the log file and it works. I've never encountered this bug on Yocto.

To confirm, I'll compile in a clean environment.

Sorry for the inconvenience

closed 2020-04-21 by clementlgl