ros/meta-ros #687: ros2-eloquent-thud didn't build for me

wpd, 2020-05-19:
Here are the steps I took:
```
$  git clone -b build-draft --single-branch https://github.com/ros/meta-ros.git build
$ mkdir conf
$ ln -snf ../conf build/.
$ cp build/files/ros2-eloquent-thud.mcf conf/
$ build/scripts/mcf -f conf/ros2-eloquent-thud.mcf 
$ unset BDIR BITBAKEDIR BUILDDIR OECORELAYERCONF OECORELOCALCONF OECORENOTESCONF OEROOT TEMPLATECONF
$ source openembedded-core/oe-init-build-env 
$ mkdir -p /home/wpd/ros2-separate-directory
```
Edited local.conf as per the instructions, and setting
```
MACHINE ?= "qemux86"
ROS_COMMON_ARTIFACTS = "/home/wpd/ros2-separate-directory"
```
then
```
$ bitbake ros-image-core
```
which ultimately failed with:
```
| ERROR: oe_runmake failed
| WARNING: /home/wpd/ros2-separate-directory/BUILD-ros2-eloquent-thud/work/x86_64-linux/asio-native/1.12.1-r0/temp/run.do_compile.29742:1 exit 1 from 'exit 1'
| ERROR: Function failed: do_compile (log file is located at /home/wpd/ros2-separate-directory/BUILD-ros2-eloquent-thud/work/x86_64-linux/asio-native/1.12.1-r0/temp/log.do_compile.29742)
ERROR: Task (virtual:native:/home/wpd/ros2sundial/test/meta-openembedded/meta-oe/recipes-support/asio/asio_1.12.1.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 5523 tasks of which 5522 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  virtual:native:/home/wpd/ros2sundial/test/meta-openembedded/meta-oe/recipes-support/asio/asio_1.12.1.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```
I'm going to go repeat this experiment, but I thought I would ask about it in parallel.  It seems very odd to me that a meta-openembedded recipe should have failed in this manner.

shr-project, 2020-05-19:
Please share /home/wpd/ros2-separate-directory/BUILD-ros2-eloquent-thud/work/x86_64-linux/asio-native/1.12.1-r0/temp/log.do_compile.29742 without seeing the error it's hard to guess what might went wrong for you.

wpd, 2020-05-20:
Sure.  Here they are.
Knowing that this works for you, I'll continue to poke around and see what's up on my build syste,.
Thanks for the help.
--wpd

[run.do_compile.29742.gz](https://github.com/ros/meta-ros/files/4654109/run.do_compile.29742.gz)
[log.do_compile.29742.gz](https://github.com/ros/meta-ros/files/4654110/log.do_compile.29742.gz)

shr-project, 2020-05-20:
What host OS are you using?

It's failing with
```
undefined reference to `boost::chrono::steady_clock::now()'
sender.cpp:(.text._ZNK4asio6detail11timer_queueINS0_18chrono_time_traitsIN5boost6chrono12steady_clockENS_11wait_traitsIS5_EEEEE18wait_duration_usecElcollect2: error: ld returned 1 exit status
blocking_token_tcp_client.cpp:(.text.startup+0x41c):collect2: error: ld returned 1 exit status
```

Are you using any other layers which might remove chrono from BOOST_LIBS?

You can also try to disable boost support in native asio
```
PACKAGECONFIG_remove_pn-asio-native = "boost"
```
in your conf/local.conf

wpd, 2020-05-20:
I'm building on a Ubuntu 16.04 LTS host.
I'll add the line you suggest and try that.
Thank you.
--wpd

wpd, 2020-05-20:
Hi shr-project.  I tried adding the PACKAGCONFIG line you suggested to local.conf and it still didn't build.
I am not using adding or removing any layers other than those added by the mcf tool with the ros2-eloquent-thud.mcf file.

I will work on debugging this myself, but it would be helpful to confirm that this particular configuration is known to build successfully on a Ubuntu 16.04 LTS system.

Thanks for your help.

--wpd

shr-project, 2020-05-20:
We use only 18.04 ubuntu (and for some builds 20.04).

Can you check with 18.04 e.g. in docker?

wpd, 2020-05-20:
Ok.  That's good to know.
I know that Eloquent and Dashing required 18.04, but I figured that since Yocto thud only lists 16.04 as its supported release, this would build on 16.04.
I'll spin up an 18.04 container and try again.
Thanks.

--wpd

wpd, 2020-05-20:
It looks like it built asio fine in my 18.04 lxd container.
Sadly, I didn't allocate enough disk space for the container months ago when I created it.
Time to go download some more disk space :-)

Aside, (trying to use my "I want to be helpful" voice and hoping that this doesn't come out in my "I sound snarky" voice), it would be nice if the build instructions wiki stated that the instructions were known to work on Ubuntu 18.04 (and whatever other releases on which you test them).  You could also add a sentence that they have been seen NOT work on Ubuntu 16.04.  If I had read the first sentence, then I would have immediately spun up an 18.04 container when my build failed.  If I had seen the second sentence, I wouldn't have even bothered trying to build on my 16.04 host.

But again, thanks for the help.
--wpd

shr-project, 2020-05-20:
I'm still surprised that it didn't build on 16.04, but digging a bit deeper boost::chrono::steady_clock::now() is guarded by BOOST_CHRONO_HAS_CLOCK_STEADY and boost's config.hpp sets it like this:
```
# if defined( BOOST_CHRONO_POSIX_API )
#   define BOOST_CHRONO_HAS_PROCESS_CLOCKS
#   include <time.h>  //to check for CLOCK_REALTIME and CLOCK_MONOTONIC and _POSIX_THREAD_CPUTIME
#   if defined(CLOCK_MONOTONIC)
#      define BOOST_CHRONO_HAS_CLOCK_STEADY
#   endif
...
```

So this is where boost-native from 16.04 and 18.04 gets different because OE uses glibc from host. But quick check for CLOCK_MONOTONIC show that it should be enabled in both 16.04 and 18.04. Maybe there is another difference which I don't see just by looking at config.hpp.

But in the end it looks like asio issue, it should check
BOOST_CHRONO_HAS_CLOCK_STEADY before trying to use it (as documented in https://theboostcpplibraries.com/boost.chrono)

But considering that Yocto 2.6 Thud is already out of support and asio-native isn't used widely, it's probably not worth fixing it there.

Ubuntu 18.04 is mentioned as prerequisite for meta-ros builds, see:
https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions
it should in theory work on other distributions, but we don't have resources to test on many of them, that's why bug reports like this are useful :).

wpd, 2020-05-20:
Apparently, the wiki _does_ mention bionic (18.04).  I just missed that.  Sorry for all the SPAM.
Now I just need to figure out why I get the following error when I attempt to build in my 18.04 lxd container (ros-image-core:do_image is failing):
```
| Copying files into the device: __populate_fs: Could not allocate block in ext2 filesystem while writing file "math2html.cpython-37.pyc"
| mkfs.ext4: Could not allocate block in ext2 filesystem while populating file system
```
It probably indicates that I'm out of disk space in my container...We'll see.

wpd, 2020-05-20:
Adding 100GB of disk space to my container didn't help.
In order to produce an image, I had to add:
```
IMAGE_ROOTFS_SIZE="262144"
```
to my `local.conf` file.  I sure wish I knew why I needed to add this when you don't.

Oh well.  I now have a buildable system.  Now it's time to play with it! :-)

--wpd

shr-project, 2020-05-20:
I'll fix asio to build correctly with 16.04.

Here is what I've found, the root of the issue is this:
https://github.com/chriskohlhoff/asio/blob/asio-1-12-1/asio/include/asio/detail/config.hpp#L480

In 16.04 __GXX_EXPERIMENTAL_CXX0X__ isn't enabled by default in g++ 5.4.0 and default __cplusplus 199711L, because of that ASIO_HAS_STD_CHRONO doesn't get defined and even when ASIO_HAS_BOOST_CHRONO is defined later, it doesn't add boost_chrono library when linking the icmp and timeouts examples, which results in that linking failure you were seeing. Fixing the examples/Makefile.am seems to work OK.

g++ 7.5.0 in 18.04 defines __GXX_EXPERIMENTAL_CXX0X__  and __cplusplus is 201402L (used in newer asio with https://github.com/chriskohlhoff/asio/commit/8bedc5aa3be1c47a6ab55d95f5ec8a5dce84f79c).

The PACKAGECONFIG option I've suggested earlier didn't work for you, because the boost PACKAGECONFIG isn't in thud's version of asio (sorry I was looking at newer meta-oe), it was added in Yocto 2.7 warrior with:
https://github.com/openembedded/meta-openembedded/commit/923d7082df3c70a597b05b5543e06cd8c00e3e7e
https://github.com/chriskohlhoff/asio/commit/8bedc5aa3be1c47a6ab55d95f5ec8a5dce84f79c
backporting PACKAGECONFIG works as well in 16.04

shr-project, 2020-05-20:
I've just rebuilt eloquent for qemux86 (with newer thud-draft with milestone 12 changes - but probably not affecting the image size much).

And yes, the image is bigger than the default IMAGE_ROOTFS_SIZE, but nothing was complaining here:

```
# $IMAGE_ROOTFS_SIZE [3 operations]
#   set /jenkins/mjansa/build/ros/ros2-eloquent-thud/openembedded-core/meta/conf/documentation.conf:220
#     [doc] "Defines the size in Kbytes for the generated image."
#   set /jenkins/mjansa/build/ros/ros2-eloquent-thud/openembedded-core/meta/conf/bitbake.conf:782
#     [_defaultval] "65536"
#   set? /jenkins/mjansa/build/ros/ros2-eloquent-thud/openembedded-core/meta/recipes-core/images/core-image-minimal.bb:11
#     "8192"
# pre-expansion value:
#   "8192"
IMAGE_ROOTFS_SIZE="8192"
```

```
# $IMAGE_FSTYPES [3 operations]
#   append /jenkins/mjansa/build/ros/ros2-eloquent-thud/openembedded-core/meta/conf/machine/include/qemu.inc:16
#     "tar.bz2 ext4"
#   set /jenkins/mjansa/build/ros/ros2-eloquent-thud/openembedded-core/meta/conf/documentation.conf:209
#     [doc] "Formats of root filesystem images that you want to have created."
#   set? /jenkins/mjansa/build/ros/ros2-eloquent-thud/openembedded-core/meta/conf/bitbake.conf:767
#     "tar.gz"
# pre-expansion value:
#   " tar.bz2 ext4"
IMAGE_FSTYPES=" tar.bz2 ext4"
#
```

```
grep IMAGESIZE buildhistory/images/qemux86/glibc/ros-image-core-eloquent/image-info.txt
IMAGESIZE = 240732
```

```
-rw-r--r-- 1 mjansa mjansa 310M May 20 16:42 tmp-glibc/deploy/images/qemux86/ros-image-core-eloquent-qemux86-20200520230606.rootfs.ext4
```

Please share
tmp-glibc/work/qemux86-oe-linux/ros-image-core/1.0-r0/temp/log.do_image_ext4
mine looks OK:
```
DEBUG: Executing python function set_image_size
DEBUG: 312951.600000 = 240732 * 1.300000
DEBUG: 317047.600000 = max(312951.600000, 8192)[312951.600000] + 1
DEBUG: 317048.000000 = int(317047.600000)
DEBUG: 317048 = aligned(317048)
DEBUG: returning 317048
DEBUG: Python function set_image_size finished
DEBUG: Executing python function extend_recipe_sysroot
NOTE: Direct dependencies ...
NOTE: Installed into sysroot: []
NOTE: Skipping as already exists....
DEBUG: Python function extend_recipe_sysroot finished
DEBUG: Executing shell function do_image_ext4
DEBUG: Executing dd if=/dev/zero of=/jenkins/mjansa/build/ros/ros2-eloquent-thud/tmp-glibc/work/qemux86-oe-linux/ros-image-core/1.0-r0/deploy-ros-image-core-image-complete/ros-image-core-eloquent-qemux86-20200520230606.rootfs.ext4 seek=317048 count=0 bs=1024
0+0 records in
0+0 records out
0 bytes copied, 0.000224589 s, 0.0 kB/s
DEBUG: Actual Rootfs size:  240732      /jenkins/mjansa/build/ros/ros2-eloquent-thud/tmp-glibc/work/qemux86-oe-linux/ros-image-core/1.0-r0/rootfs
DEBUG: Actual Partion size: 324657152
DEBUG: Executing mkfs.ext4 -F -i 4096 /jenkins/mjansa/build/ros/ros2-eloquent-thud/tmp-glibc/work/qemux86-oe-linux/ros-image-core/1.0-r0/deploy-ros-image-core-image-complete/ros-image-core-eloquent-qemux86-20200520230606.rootfs.ext4 -d /jenkins/mjansa/build/ros/ros2-eloquent-thud/tmp-glibc/work/qemux86-oe-linux/ros-image-core/1.0-r0/rootfs
mke2fs 1.44.3 (10-July-2018)
64-bit filesystem support is not enabled.  The larger fields afforded by this feature enable full-strength checksumming.  Pass -O 64bit to rectify.
Discarding device blocks:   1024/317048^H^H^H^H^H^H^H^H^H^H^H^H^H             ^H^H^H^H^H^H^H^H^H^H^H^H^Hdone
Creating filesystem with 317048 1k blocks and 79560 inodes
Filesystem UUID: b043df19-0476-4970-b62e-f7b65b6343b9
Superblock backups stored on blocks:
        8193, 24577, 40961, 57345, 73729, 204801, 221185

Allocating group tables:  0/39^H^H^H^H^H     ^H^H^H^H^Hdone
Writing inode tables:  0/39^H^H^H^H^H     ^H^H^H^H^Hdone
Creating journal (8192 blocks): done
Copying files into the device: done
Writing superblocks and filesystem accounting information:  0/39^H^H^H^H^H     ^H^H^H^H^Hdone

       12161 inodes used (15.29%, out of 79560)
          18 non-contiguous files (0.1%)
          44 non-contiguous directories (0.4%)
             # of inodes with ind/dind/tind blocks: 0/0/0
             Extent depth histogram: 11586/22
      235176 blocks used (74.18%, out of 317048)
           0 bad blocks
           0 large files

       10349 regular files
        1131 directories
           0 character device files
           0 block device files
           0 fifos
           6 links
         672 symbolic links (545 fast symbolic links)
           0 sockets
------------
       12158 files
DEBUG: Shell function do_image_ext4 finished
```

wpd, 2020-05-21:
Here you go.
[log.do_image.26934.gz](https://github.com/ros/meta-ros/files/4659975/log.do_image.26934.gz)

wpd, 2020-05-21:
Oops.  Let's try that again.
[log.do_image_ext4.gz](https://github.com/ros/meta-ros/files/4659986/log.do_image_ext4.gz)

shr-project, 2020-05-21:
Strange, your image should be half size compared to mine, but still the partition should be big enough (153676k partition 121860k rootfs)

```
DEBUG: Actual Rootfs size:  121860      /home/ubuntu/separate-directory/BUILD-ros2-eloquent-thud/work/qemux86-oe-linux/ros-image-core/1.0-r0/rootfs
DEBUG: Actual Partion size: 157364224
```

Maybe it's because your image is so much smaller there isn't enough overhead, I'll rebuild with old thud-draft branch, to see what's missing in the image (but it built for me before as well, so it might be something missing only in our image).

biggest 20 packages in mine:
```
head -n 20 buildhistory/images/qemux86/glibc/ros-image-core-eloquent/installed-package-sizes.txt
24991   KiB     cpp
17239   KiB     python3-numpy
8539    KiB     python3-core
7103    KiB     systemd
6795    KiB     python3-pip
6131    KiB     udev-hwdb
5231    KiB     python3-misc
5143    KiB     libfastrtps1
4681    KiB     python3-lxml
3965    KiB     bluez5
3922    KiB     libc6
3814    KiB     python3-distutils
3608    KiB     libglib-2.0-0
3262    KiB     libpython3.7m1.0
3048    KiB     python3-docutils
2789    KiB     libmpfr6
2741    KiB     libcrypto1.1
2643    KiB     python3-setuptools
2524    KiB     shared-mime-info
2198    KiB     python3-pydoc
```

wpd, 2020-05-21:
I thought I would test your change... especially since it would mean that I could go back to building on my development host instead of in a container (which makes things like grabbing log files from builds somewhat...interesting).
But, I see that your commit is on the thud-draft-ms12 branch of shr-project/meta-ros whereas the ros2-eloquent-thud.mcf file clones the thud-draft branch of ros/meta-ros.
I can go in and pull down and switch my meta-ros to your branch, but I wonder how many cans of how many worms I'd be pulling down with that.

I think I'll wait until the movie comes out. :-)

It seems to me that "ms12" must have something to do with "milestone 12".  When might we expect to see ms12 available?  Is there a timeline published somewhere?

In the mean time, as I now have a set of procedure for building this, I'm happy to close out this issue, unless you would like to keep it open for some reason.

Thanks again for your help.
--wpd

shr-project, 2020-05-21:
You can cherry-pick just that one commit, it doesn't depend on anything else from Milestone 12.

The milestones are defined in the wiki:
https://github.com/ros/meta-ros/wiki/Superflore-OE-Recipe-Generation-Scheme#milestone-12-implement-layer-version-3-phase-2
and you can see the changes also here in pull requests, I hope the changes for MS12 will be merged soon, but it doesn't depend on me.

wpd, 2020-05-21:
That's a good idea.  I'll do that.
Do you want me to close this issue now?  Or leave it open for you to close sometime in the future?

--wpd

shr-project, 2020-05-21:
I did 2 more builds with current thud-draft in 18.04 and 16.04 and both are in 200+MB range

16.04: 222428  tmp-glibc/work/qemux86-oe-linux/ros-image-core/1.0-r0/rootfs/

If you're happy with your small image, then we can close this :).

closed 2020-05-21 by wpd