ros/meta-ros #688: pkg-config no found yaml

carlo0120, 2020-05-20:
branch zeus-draft

bitbake core-image-sato

Run-time dependency yaml-0.1 found: NO (tried pkgconfig)
meson.build:54:0: ERROR: Dependency "yaml-0.1" not found, tried pkgconfig

poky/meta/recipes-devtools/libmodulemd/libmodulemd_git.bb:do_configure) failed with exit code '1'

shr-project, 2020-05-20:
What DISTRO are you using? My guess is poky with rpm?

zeus-draft builds yaml version from ros:
```
meta-ros2-dashing/conf/ros-distro/include/dashing/ros-distro-preferred-providers.inc:PREFERRED_PROVIDER_libyaml = "libyaml-vendor"
meta-ros2-dashing/conf/ros-distro/include/dashing/ros-distro-preferred-providers.inc:PREFERRED_PROVIDER_libyaml-dev = "libyaml-vendor-dev"
meta-ros2-dashing/conf/ros-distro/include/dashing/ros-distro-preferred-providers.inc:PREFERRED_PROVIDER_libyaml-dbg = "libyaml-vendor-dbg"
meta-ros2-dashing/conf/ros-distro/include/dashing/ros-distro-recipe-blacklist.inc:PNBLACKLIST[libyaml] ?= "Provided by libyaml-vendor"
```

I've already blacklisted it in newer zeus-draft for Milestone 12:
https://github.com/shr-project/meta-ros/commit/66a93a8b4e
which unfortunately means that you need to use ipk package manager (like ROS and webOS DISTRO does).

carlo0120, 2020-05-20:
ROS_DISTRO := "dashing"

vi poky/meta/recipes-devtools/libmodulemd/libmodulemd/0002-modulemd-v1-meson.build-do-not-generate-gir-or-gtkdo.patch

-yaml = dependency('yaml-0.1')
+yaml = dependency('yaml-0.1????')

What is the solution?

shr-project, 2020-05-20:
What OE DISTRO are you using? With ROS or webOS there is no dependency on libmodulemd.

The proper solution is to enable ros_opt_prefix for ROS2 as well (like ROS1 already does) and then let yaml-vendor to be installed for ROS components in /opt without affecting yaml in /usr prefix. But that's quite difficult to implement, it's still on our roadmap, probably after Milestone 14.

carlo0120, 2020-05-20:
git://git.yoctoproject.org/poky
git://git.openembedded.org/meta-openembedded

machine jetson nano
https://github.com/madisongh/meta-tegra.git

carlo0120, 2020-05-29:
The requirement is ROS2 operation on Xserver.
Running core-image-sato for the Xserver installs libmodulemd.
Is there a way to uninstall libmodulemd?
In yocto, do you use the Ros2 GUI unused?

shr-project, 2020-07-28:
I'm going to assume that the referenced pull requests work OK for you and everything was merged today, please create new issue ticket if you find some issues later.

"Silence implies consent" (see http://wiki.c2.com/?SilenceImpliesConsent).

closed 2020-07-28 by shr-project