ros/meta-ros #689: meta-ros yocto dunfell support status

avinashiitk, 2020-05-27:
I don't see /remotes/origin/dunfell-draft branch with meta-ros. 
Do we have plans to support dunfell (Yocto 3.1) on meta-ros?

shr-project, 2020-05-27:
Yes, it's already implemented in:
https://github.com/shr-project/meta-ros/tree/dunfell-draft
and .mcf files in:
https://github.com/shr-project/meta-ros/tree/build-draft-ms12

Hopefully it will be merged here soon, there is pull request for master-draft https://github.com/ros/meta-ros/pull/680, when it's merged to master, dunfell branch will be created as well.

avinashiitk, 2020-06-01:
Thanks for information. Do you have timeline, when dunfell-draft will get merged and branch will get created?
-Avinash

shr-project, 2020-07-04:
It was merged just now, please try it and let us know.

shr-project, 2020-07-28:
I'm going to assume that it works OK for you, please create new issue ticket if you find some issues later.

"Silence implies consent" (see http://wiki.c2.com/?SilenceImpliesConsent).

closed 2020-07-28 by shr-project