ros/meta-ros #701: [dunfell] [meta-ros2-foxy] Compatibility with libyaml & dnf

victor-uavia, 2020-07-08:
Hello, 

In the process of integrating meta-ros in my current yocto distro, I encountered some problems due to some recipes being blacklisted by the `meta-ros` layer, most notably:
- `dnf` (due to a problem with `libmodulemd-v1` not able to build due to problem with `libyaml`): I want to have a `rpm` based distro
- `gstreamer1.0-plugins-bad`: seems to be a license problem in your case, I did not encountered any _yocto_ error but I am not sure that this is relevant for you

I was finally able to overcome the problem with `libmodulemd-v1` since in my case I only needed to patch the `libmodulemd-v1-native` version of the recipe.

The following patch is what I need to build the `meta-ros2-foxy` layer, which enable `libyaml`, `libmodulemd-v1` and its dependencies and `gstreamer-plugins-bad`.
If some part of this is of any interrest for you, I would like to have some remark in order for this to be upstreamed. 

```
diff --git a/meta-ros-common/conf/ros-distro/include/ros-world-recipe-blacklist.inc b/meta-ros-common/conf/ros-distro/include/ros-world-recipe-blacklist.inc
index 00e8bec..b5dff21 100644
--- a/meta-ros-common/conf/ros-distro/include/ros-world-recipe-blacklist.inc
+++ b/meta-ros-common/conf/ros-distro/include/ros-world-recipe-blacklist.inc
@@ -5,7 +5,6 @@
 ROS_WORLD_SKIP_GROUPS += "world-license-faad"
 # faad2 is enabled only for rpi MACHINEs in meta-raspberrypi/recipes-multimedia/gstreamer/gstreamer1.0-plugins-bad_%.bbappend
 # but we're blacklisting it for all to keep this simpler
-PNBLACKLIST[gstreamer1.0-plugins-bad] ?= "${@bb.utils.contains('ROS_WORLD_SKIP_GROUPS', 'world-license-faad', 'depends on faad2 which has a restricted license not whitelisted in LICENSE_FLAGS_WHITELIST', '', d)}"
 PNBLACKLIST[gstd] ?= "${@bb.utils.contains('ROS_WORLD_SKIP_GROUPS', 'world-license-faad', 'depends on gstreamer1.0-plugins-bad which depends on faad2 which has a restricted license not whitelisted in LICENSE_FLAGS_WHITELIST', '', d)}"
 PNBLACKLIST[gst-shark] ?= "${@bb.utils.contains('ROS_WORLD_SKIP_GROUPS', 'world-license-faad', 'depends on gstreamer1.0-plugins-bad which depends on faad2 which has a restricted license not whitelisted in LICENSE_FLAGS_WHITELIST', '', d)}"
 PNBLACKLIST[packagegroup-meta-multimedia] ?= "${@bb.utils.contains('ROS_WORLD_SKIP_GROUPS', 'world-license-faad', 'rdepends on gstd, gst-shark which depend on gstreamer1.0-plugins-bad which depends on faad2 which has a restricted license not whitelisted in LICENSE_FLAGS_WHITELIST', '', d)}"
@@ -39,10 +38,6 @@ PNBLACKLIST[wiringpi] ?= "${@bb.utils.contains('ROS_WORLD_SKIP_GROUPS', 'world-i
 PNBLACKLIST[packagegroup-rpi-test] ?= "${@bb.utils.contains_any('ROS_WORLD_SKIP_GROUPS', ['world-issues', 'python2'], 'world-issues: Depends on blacklisted packages like wiringpi and recipes from meta-multimedia we do not depend on: bigbuckbunny-480p, bigbuckbunny-720p, bigbuckbunny-1080p and wireless-regdb from meta-networking; python2: Depends on blacklisted python2', '', d)}"
 PNBLACKLIST[kernel-selftest] ?= "${@bb.utils.contains('ROS_WORLD_SKIP_GROUPS', 'world-issues', 'Fails to build with linux-yocto kernel: WARNING: clang >= 6.0  with bpf support is needed with kernel 4.18+ so either install it and add it to HOSTTOOLS, or add clang-native from meta-clang to dependency and then /bin/sh: 1: llc: not found + /bin/sh: 1: clang: not found + /bin/sh: 1: llvm-readelf: not found', '', d)}"
 PNBLACKLIST[vboxguestdrivers] ?= "${@bb.utils.contains('ROS_WORLD_SKIP_GROUPS', 'world-issues', 'Fails to build with linux-yocto kernel: vboxguestdrivers/5.2.22-r0/vbox_module/vboxguest/r0drv/linux/time-r0drv-linux.c:175:5: error: implicit declaration of function ktime_get_real_ts; did you mean ktime_get_real_ns? [-Werror=implicit-function-declaration]', '', d)}"
-PNBLACKLIST[libmodulemd-v1] ?= "${@bb.utils.contains('ROS_WORLD_SKIP_GROUPS', 'world-issues', 'Fails to build with libyaml-vendor from meta-ros instead of regular libyaml: meson.build:54:0: ERROR:  Dependency yaml-0.1 not found, tried pkgconfig', '', d)}"
-PNBLACKLIST[libmodulemd] ?= "${@bb.utils.contains('ROS_WORLD_SKIP_GROUPS', 'world-issues', 'Fails to build with libyaml-vendor from meta-ros instead of regular libyaml: meson.build:54:0: ERROR:  Dependency yaml-0.1 not found, tried pkgconfig', '', d)}"
-PNBLACKLIST[libdnf] ?= "${@bb.utils.contains('ROS_WORLD_SKIP_GROUPS', 'world-issues', 'Depends on blacklisted libmodulemd', '', d)}"
-PNBLACKLIST[dnf] ?= "${@bb.utils.contains('ROS_WORLD_SKIP_GROUPS', 'world-issues', 'Depends on blacklisted libdnf->libmodulemd', '', d)}"
 PNBLACKLIST[dnfdragora] ?= "${@bb.utils.contains('ROS_WORLD_SKIP_GROUPS', 'world-issues', 'Depends on blacklisted dnf->libdnf->libmodulemd', '', d)}"
 PNBLACKLIST[dnf-plugin-tui] ?= "${@bb.utils.contains('ROS_WORLD_SKIP_GROUPS', 'world-issues', 'Depends on blacklisted dnf->libdnf->libmodulemd', '', d)}"
 PNBLACKLIST[createrepo-c] ?= "${@bb.utils.contains('ROS_WORLD_SKIP_GROUPS', 'world-issues', 'Depends on blacklisted libmodulemd', '', d)}"
diff --git a/meta-ros-common/recipes-devtools/libmodulemd-v1/libmodulemd-v1_%.bbappend b/meta-ros-common/recipes-devtools/libmodulemd-v1/libmodulemd-v1_%.bbappend
new file mode 100644
index 0000000..013afc1
--- /dev/null
+++ b/meta-ros-common/recipes-devtools/libmodulemd-v1/libmodulemd-v1_%.bbappend
@@ -0,0 +1,3 @@
+PREFERRED_PROVIDER_libyaml_pn-${PN}-native = "libyaml"
+PREFERRED_PROVIDER_libyaml-dev_pn-${PN}-native = "libyaml-dev"
+PREFERRED_PROVIDER_libyaml-dbg_pn-${PN}-native = "libyaml-dbg"
diff --git a/meta-ros2-foxy/conf/ros-distro/include/foxy/ros-distro-preferred-providers.inc b/meta-ros2-foxy/conf/ros-distro/include/foxy/ros-distro-preferred-providers.inc
index 4fd2f2b..ad7620c 100644
--- a/meta-ros2-foxy/conf/ros-distro/include/foxy/ros-distro-preferred-providers.inc
+++ b/meta-ros2-foxy/conf/ros-distro/include/foxy/ros-distro-preferred-providers.inc
@@ -5,6 +5,6 @@
 # Set PREFERRED_PROVIDER_<PN> here for non-platform packages and to override those set in
 # ros-distro-platform-preferred-providers.inc .

-PREFERRED_PROVIDER_libyaml = "libyaml-vendor"
-PREFERRED_PROVIDER_libyaml-dev = "libyaml-vendor-dev"
-PREFERRED_PROVIDER_libyaml-dbg = "libyaml-vendor-dbg"
+#PREFERRED_PROVIDER_libyaml = "libyaml-vendor"
+#PREFERRED_PROVIDER_libyaml-dev = "libyaml-vendor-dev"
+#PREFERRED_PROVIDER_libyaml-dbg = "libyaml-vendor-dbg"
diff --git a/meta-ros2-foxy/conf/ros-distro/include/foxy/ros-distro-recipe-blacklist.inc b/meta-ros2-foxy/conf/ros-distro/include/foxy/ros-distro-recipe-blacklist.inc
index a6b55c6..a7e3a91 100644
--- a/meta-ros2-foxy/conf/ros-distro/include/foxy/ros-distro-recipe-blacklist.inc
+++ b/meta-ros2-foxy/conf/ros-distro/include/foxy/ros-distro-recipe-blacklist.inc
@@ -2,8 +2,6 @@
 #
 # Copyright (c) 2020 LG Electronics, Inc.

-PNBLACKLIST[libyaml] ?= "Provided by libyaml-vendor"
-
 PNBLACKLIST[ament-clang-format-native] ?= "${@bb.utils.contains('ROS_WORLD_SKIP_GROUPS', 'clang', 'clang: depends on clang-format', '', d)}"
 PNBLACKLIST[ament-clang-format] ?= "${@bb.utils.contains('ROS_WORLD_SKIP_GROUPS', 'clang', 'clang: depends on clang-format', '', d)}"
[... 34 lines cut ...]
 SYSROOT_DIRS_append = " ${prefix}/cmake"

-PROVIDES += "libyaml"
-RPROVIDES_${PN} += "libyaml"
+#PROVIDES += "libyaml"
+#RPROVIDES_${PN} += "libyaml"

 # Instead of fetching
 # https://github.com/yaml/libyaml/archive/10c907871f1ccd779c7fccf6b81a62762b5c4e7b.zip
diff --git a/meta-ros2/recipes-devtools/foonathan-memory/foonathan-memory_git.bb b/meta-ros2/recipes-devtools/foonathan-memory/foonathan-memory_git.bb
index 5f86846..01cf62d 100644
--- a/meta-ros2/recipes-devtools/foonathan-memory/foonathan-memory_git.bb
+++ b/meta-ros2/recipes-devtools/foonathan-memory/foonathan-memory_git.bb
@@ -75,3 +75,10 @@ FILES_${PN}-dev += "${datadir}/foonathan_memory/cmake"
 FILES_${PN}-doc += "${datadir}/foonathan_memory"

 BBCLASSEXTEND = "native"
+
+# Error do not properly handle security.conf: disable PIE CFLAGS
+# | FAILED: tool/nodesize_dbg
+#| : && /workdir/build/tmp/work/aarch64-droneos-linux/foonathan-memory/0.6.2+gitAUTOINC+8f6a027d47-r0/recipe-sysroot-native/usr/bin/aarch64-droneos-linux/aarch64-droneos-linux-g++  -march=armv8-a+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/workdir/build/tmp/work/aarch64-droneos-linux/foonathan-memory/0.6.2+gitAUTOINC+8f6a027d47-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/workdir/build/tmp/work/aarch64-droneos-linux/foonathan [... cut ...] dynamic -static tool/CMakeFiles/foonathan_memory_node_size_debugger.dir/node_size_debugger.cpp.o  -o tool/nodesize_dbg   && :
+#| /workdir/build/tmp/work/aarch64-droneos-linux/foonathan-memory/0.6.2+gitAUTOINC+8f6a027d47-r0/recipe-sysroot-native/usr/bin/aarch64-droneos-linux/../../libexec/aarch64-droneos-linux/gcc/aarch64-droneos-linux/7.3.0/ld: cannot find rcrt1.o: No such file or directory
+#| collect2: error: ld returned 1 exit status
+SECURITY_CFLAGS_pn-${PN} = "${SECURITY_NOPIE_CFLAGS}"
```

shr-project, 2020-07-09:
> gstreamer1.0-plugins-bad: seems to be a license problem in your case, I did not encountered any yocto error but I am not sure that this is relevant for you

Are you building with meta-raspberrypi? See the comment above world-license-faad section:
```
# faad2 is enabled only for rpi MACHINEs in meta-raspberrypi/recipes-multimedia/gstreamer/gstreamer1.0-plugins-bad_%.bbappend
# but we're blacklisting it for all to keep this simpler
```

If you don't care about faad2, then you can just remove "world-license-faad" from ROS_WORLD_SKIP_GROUPS like meta-ros-webos does:
https://github.com/ros/meta-ros-webos/commit/2eb6d31be422e83361d0da976a72488f1a4770de

victor-uavia, 2020-07-09:
> If you don't care about faad2, then you can just remove "world-license-faad" from ROS_WORLD_SKIP_GROUPS like meta-ros-webos does:

Thank you for this tip, indeed I do not build with meta-raspberrypi, doing as it is done in `meta-ros-webos` works for me.

For my use case, with the above patch, I was able to remove "world-issues" from `ROS_WORLD_SKIP_GROUPS` and doing so building `libmodulemd-v1` `libmodulemd` `libdnf` `dnf`.

Since it looks for me that you blacklisted those recipe due to some build issues, do you think that part of this patch should be merged ?
However, if you do think that these recipe should stay blacklisted, you can close this issue.

shr-project, 2020-07-09:
For libyaml issue see the commits linked from this ticket, e.g.
https://github.com/shr-project/meta-ros/commit/27230cc632b4413a25f59ef02790ea89123a9a26
I haven't finished testing it properly, but if that works I'll create PR with that.

Let me know if that works for you as well.

closed 2020-07-28 by shr-project

shr-project, 2020-07-28:
Fix was merged into all active branches (thud, warrior, zeus, dunfell, master).