ros/meta-ros #716: Use colcon with meta-ros

christian-nils, 2020-08-19:
Hello,

Sorry for this naive question, but I cannot find a clear answer on internet. When using meta-ros in my Yocto project, is it possible to have the build tool _colcon_ on the target? I cannot find any reference to that on the meta-ros repository, what repositories to add via IMAGE_INSTALL_append?

If this use case seems to not be correct to you, would you mind detailing a better practice to create a ROS package that will be used in a custom hardware using meta-ros for building the image?

Thank you in advance.

niederha, 2021-06-04:
Hello @christian-nils !

I'm interested in this topic... would it be possible to know why it was closed? Is there an answer? Is it not a correct usecase? Is it not the correct place to ask this question?

windelbouwman, 2021-11-19:
I created some recipes to build colcon, and include it in an SDK, or on target:

https://github.com/ros/meta-ros/pull/977

robwoolley, 2023-12-02:
The code was merged, but we should document this functionality, how to use it, and how to test it.

annto99, 2024-05-03:
At the end did you manage to document it somewhere?

OznOg, 2024-11-22:
Would it be possible to enable the use of colcon in the do_compile() of a bitbake recipe? I have a project that builds this was on a fedora, and integrating it on yocto become really painful because I cannot have colcon to be used during the build. Having it in the sdk is not really helpful because it would need to split my build in at least 3 phases which seems overcomplicated.

robwoolley, 2024-11-23:
Can you provide more details about your project and where you ran into problems integrating it into Yocto?

OznOg, 2024-11-23:
Thank you for your answer;
I'm working on a huge application that uses some ROS1 components.
As there are quite a lot of ROS projects (~50), we created a metapackage that aggregates all of them, and I relied on catkin (tool) to resolve dependencies and build. As colcon seems to be the uptodate way of doing that, I moved to colcon to build them on fedora (the distro we use for dev/early testing).
I would like to have the same flow in yocto: build the metapackage as "one big dependency" without having to enter into the details of how to build each of them nor resolve their dependencies.
That's how I came to trying to use colcon inside the do_compile, and the purpose of my question.
I was puzzled because having colcon in the sdk would mean that it can be used in the yocto build (so colcon seem to work with yocto libraries etc.) do *inside* yocto build.
Please let me know if there is a "standard" way to build metapackages in yocto, maybe I'm just trying to reinvent the wheel.

windelbouwman, 2024-11-23:
You could have a look at superflore, this is a tool which sort of takes package.xml descriptions, and spits out yocto bitbake recipes.

See also:
https://github.com/ros-infrastructure/superflore?tab=readme-ov-file#openembedded-usage

Maybe you can use this, instead of bundling a colcon build package amalgamation into a single bitbake recipe?

Btw, if you want, you can use colcon in a recipe, it's just yet another build tool. If you specify it as a native dependency, you can use it from your recipe.