ros/meta-ros #725: {melodic} cv-bridge broken

maroessler, 2020-08-25:
Hi,

I am currently working with ros1-melodic thud. I'm having some problems getting the cv-bridge to work. The package builds fine but it is having problems executing.

I managed to reproduce the error by trying to import:
`from cv_bridge.boost.cv_bridge_boost import getCvType`
in a Python script.

It throws this error:
`ImportError: /opt/ros/melodic/lib/python2.7/site-packages/cv_bridge/boost/cv_bridge_boost.so: undefined symbol: _ZN5boost6python6detail11init_moduleER11PyModuleDefPFvvE`

I suspect it has something to do with compiling cv-bridge with libboost_python37 instead of a Python 2 version.

Can anyone verify that this error exists and is not a problem on my part?

Regards
Michael

shr-project, 2020-09-08:
Hi,

> I suspect it has something to do with compiling cv-bridge with libboost_python37 instead of a Python 2 version.

Yes, this is very similar to what I was seeing with pinocchio in 2b77b41d04

> Can anyone verify that this error exists and is not a problem on my part?

I don't think this is issue in your build, cv-bridge from melodic isn't tested in runtime at all by our CI (only build testing it), with dunfell I see an error as well:

```
root@qemux86-64:~# PYTHONPATH=/opt/ros/melodic/lib/python2.7/site-packages LD_LIBRARY_PATH=/opt/ros/melodic/lib/ python -c "from cv_bridge.boost.cv_bridge_boost import getCvType"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: /usr/lib/libboost_python38.so.1.72.0: undefined symbol: _Py_tracemalloc_config
```

shr-project, 2020-09-08:
Fails similarly with gatesgarth melodic:
```
root@qemux86-64:~# PYTHONPATH=/opt/ros/melodic/lib/python2.7/site-packages LD_LIBRARY_PATH=/opt/ros/melodic/lib/ python -c "from cv_bridge.boost.cv_bridge_boost import getCvType"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: /usr/lib/libboost_python38.so.1.74.0: undefined symbol: _Py_tracemalloc_config
```

With Foxy on Gatesgarth this import test works OK.

robwoolley, 2024-04-10:
Closing as Melodic is no longer supported.

closed 2024-04-10 by robwoolley