ros/meta-ros #727: meta-ros foxy build failure

avinashiitk, 2020-08-28:
ERROR: foonathan-memory-native-0.6.2+gitAUTOINC+8f6a027d47-r0 do_configure: Execution of '/home/kumara2/WW32-2/tgr-build-setup/build/ecs-base-poky/build/tmp/work/x86_64-linux/foonathan-memory-native/0.6.2+gitAUTOINC+8f6a027d47-r0/temp/run.do_configure.17866' failed with exit code 1:
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /home/kumara2/WW32-2/tgr-build-setup/build/ecs-base-poky/build/tmp/hosttools/gcc
-- Check for working C compiler: /home/kumara2/WW32-2/tgr-build-setup/build/ecs-base-poky/build/tmp/hosttools/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Configuring incomplete, errors occurred!
| See also "/home/kumara2/WW32-2/tgr-build-setup/build/ecs-base-poky/build/tmp/work/x86_64-linux/foonathan-memory-native/0.6.2+gitAUTOINC+8f6a027d47-r0/build/CMakeFiles/CMakeOutput.log".
| WARNING: /home/kumara2/WW32-2/tgr-build-setup/build/ecs-base-poky/build/tmp/work/x86_64-linux/foonathan-memory-native/0.6.2+gitAUTOINC+8f6a027d47-r0/temp/run.do_configure.17866:1 exit 1 from 'cmake -G 'Ninja' -DCMAKE_MAKE_PROGRAM=ninja $oecmake_sitefile /home/kumara2/WW32-2/tgr-build-setup/build/ecs-base-poky/build/tmp/work/x86_64-linux/foonathan-memory-native/0.6.2+gitAUTOINC+8f6a027d47-r0/git -DCMAKE_INSTALL_PREFIX:PATH=/home/kumara2/WW32-2/tgr-build-setup/build/ecs-base-poky/build/tmp/work/x86_64-linux/foonathan-memory-native/0.6.2+gitAUTOINC+8f6a027d47-r0/recipe-sysroot-native/usr -DCMAKE_INSTALL_BINDIR:PATH=bin -DCMAKE_INSTALL_SBINDIR:PATH=sbin -DCMAKE_INSTALL_LIBEXECDIR:PATH=libexec -DCMAKE_INSTALL_SYSCONFDIR:PATH=/home/kumara2/WW32-2/tgr-build-setup/build/ecs-base-poky/build/tmp/work/x86_64-linux/foonathan-memory-native/0.6.2+gitAUTOINC+8f6a027d47-r0/recipe-sysroot-native/etc -DCMAKE_INSTALL_SHAREDSTATEDIR:PATH=../com -DCMAKE_INSTALL_LOCALSTATEDIR:PATH=/home/kumara2/WW32-2/tgr-build-setup/build/ecs-base-poky/build/tmp/work/x86_64-linux/foonathan-memory-native/0.6.2+gitAUTOINC+8f6a027d47-r0/recipe-sysroot-native/var -DCMAKE_INSTALL_LIBDIR:PATH=lib -DCMAKE_INSTALL_INCLUDEDIR:PATH=include -DCMAKE_INSTALL_DATAROOTDIR:PATH=share -DPYTHON_EXECUTABLE:PATH=/usr/bin/python3 -DPython_EXECUTABLE:PATH=/usr/bin/python3 -DPython3_EXECUTABLE:PATH=/usr/bin/python3 -DLIB_SUFFIX= -DCMAKE_INSTALL_SO_NO_EXE=0 -DCMAKE_TOOLCHAIN_FILE=/home/kumara2/WW32-2/tgr-build-setup/build/ecs-base-poky/build/tmp/work/x86_64-linux/foonathan-memory-native/0.6.2+gitAUTOINC+8f6a027d47-r0/toolchain.cmake -DCMAKE_NO_SYSTEM_FROM_IMPORTED=1 -DCMAKE_CROSSCOMPILING_EMULATOR="qemu-x86_64" -DCOMP_CMAKE_PATH=/home/kumara2/WW32-2/tgr-build-setup/build/ecs-base-poky/build/tmp/work/x86_64-linux/foonathan-memory-native/0.6.2+gitAUTOINC+8f6a027d47-r0/git/cmake/comp -Wno-dev'
| 
ERROR: Task (virtual:native:/home/kumara2/WW32-2/tgr-build-setup/build/ecs-base-poky/layers/meta-ros/meta-ros2/recipes-devtools/foonathan-memory/foonathan-memory_git.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3237 tasks of which 3200 didn't need to be rerun and 1 failed

shr-project, 2020-08-28:
What host OS are you using?

Check CMakeOutput.log to see the actual error, looks like your host's gcc in /home/kumara2/WW32-2/tgr-build-setup/build/ecs-base-poky/build/tmp/hosttools/gcc doesn't work for some reason. But hard to guess why without the log with error.

avinashiitk, 2020-08-29:
Sharing build log:
build is for Yocto Dunfell.
I am using Ubuntu 18.04 host OS.
build for ros dashing and eloquent works fine.
Issue is faced only with foxy.

meta-ros 
[log.do_configure.24123.txt](https://github.com/ros/meta-ros/files/5145161/log.do_configure.24123.txt)

-Avinash

shr-project, 2020-08-29:
The previous log was from foonathan-memory-native, this one is from foonathan-memory.

foxy uses the same foonathan-memory recipe as dashing and eloquent, my guess is that "catch: "Timeout was reached" Trying 151.101.64.133:443" was only temporary issue with your network?

Can you please debug what it was trying to fetch in do_configure from 151.101.64.133:443 ? comp, catch and catch.hpp should be already fetched by bitbake fetcher before do_configure.

shr-project, 2020-09-08:
Fixed in milestone 14.

closed 2020-09-08 by shr-project

avinashiitk, 2020-09-09:
Taken latest patch, still "bitbake foonathan-memory" is failing with below issue.
 qemu: uncaught target signal 4 (Illegal instruction) - core dumped
| ninja: build stopped: subcommand failed.

Attaching detailed log file.
-Avinash
[log.do_compile.19603.txt](https://github.com/ros/meta-ros/files/5193459/log.do_compile.19603.txt)