ros/meta-ros #729: configure error with libg2o

willofdiamond, 2020-08-30:
Hi I am trying to integrate Meta-ros foxy with my yocto build using CORE_IMAGE_EXTRA_INSTALL += "  ros-core  packagegroup-ros-world-foxy" in my local.conf for raspberry pi 3. By adding packagegroup-ros-world-foxy to CORE_IMAGE_EXTRA_INSTALL  I am receiving a configure error with libg2o. Not sure how to fix this issue

```
ERROR: libg2o-2020.5.29-1-r0 do_configure: Execution of '/home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/libg2o/2020.5.29-1-r0/temp/run.do_configure.5736' failed with exit code 1:
-- The C compiler identification is GNU 10.1.0
-- The CXX compiler identification is GNU 10.1.0
-- Check for working C compiler: /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/libg2o/2020.5.29-1-r0/recipe-sysroot-native/usr/bin/aarch64-oe-linux/aarch64-oe-linux-gcc
-- Check for working C compiler: /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/libg2o/2020.5.29-1-r0/recipe-sysroot-native/usr/bin/aarch64-oe-linux/aarch64-oe-linux-gcc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/libg2o/2020.5.29-1-r0/recipe-sysroot-native/usr/bin/aarch64-oe-linux/aarch64-oe-linux-g++
-- Check for working CXX compiler: /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/libg2o/2020.5.29-1-r0/recipe-sysroot-native/usr/bin/aarch64-oe-linux/aarch64-oe-linux-g++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Compiling on Unix
-- Found CHOLMOD: /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/libg2o/2020.5.29-1-r0/recipe-sysroot/usr/include/suitesparse
-- Looking for sgemm_
-- Looking for sgemm_ - not found
-- Looking for sgemm
-- Looking for sgemm - not found
-- Looking for sgemm_
-- Looking for sgemm_ - found
-- A library with BLAS API found.
-- Looking for cheev_
-- Looking for cheev_ - found
-- A library with LAPACK API found.
-- Found CHOLMOD and its dependencies
-- Found CSPARSE: /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/libg2o/2020.5.29-1-r0/recipe-sysroot/usr/include/suitesparse
-- Found OpenGL: /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/libg2o/2020.5.29-1-r0/recipe-sysroot/usr/lib/libGL.so
-- Compiling with OpenGL support
CMake Warning at cmake_modules/FindQGLViewer.cmake:1 (find_package):
  By not providing "FindQt5.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt5", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt5" with any of
  the following names:

    Qt5Config.cmake
    qt5-config.cmake

  Add the installation prefix of "Qt5" to CMAKE_PREFIX_PATH or set "Qt5_DIR"
  to a directory containing one of the above files.  If "Qt5" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:174 (find_package)

Qt5 not found. Install it and set Qt5_DIR accordingly
-- Could NOT find QGLVIEWER (missing: QGLVIEWER_INCLUDE_DIR QGLVIEWER_LIBRARY)
-- Compiling g2o apps
-- Compiling g2o examples
-- Compiling with GCC
-- Found Eigen3: /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/libg2o/2020.5.29-1-r0/recipe-sysroot/usr/include/eigen3 (Required is at least version "2.91.0")
-- Configuring done
CMake Error at EXTERNAL/freeglut/CMakeLists.txt:1 (add_library):
  Target "freeglut_minimal" links to target "OpenGL::GLU" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at EXTERNAL/freeglut/CMakeLists.txt:1 (add_library):
  Target "freeglut_minimal" links to target "OpenGL::GLU" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    BUILD_TESTING
    CMAKE_INSTALL_BINDIR
    CMAKE_INSTALL_DATAROOTDIR
    CMAKE_INSTALL_INCLUDEDIR
    CMAKE_INSTALL_LIBDIR
    CMAKE_INSTALL_LIBEXECDIR
    CMAKE_INSTALL_LOCALSTATEDIR
    CMAKE_INSTALL_SBINDIR
    CMAKE_INSTALL_SHAREDSTATEDIR
    CMAKE_INSTALL_SYSCONFDIR
    LIB_SUFFIX
    PYTHON_EXECUTABLE
    PYTHON_SOABI
    Python3_EXECUTABLE
    Python_EXECUTABLE

CMake Generate step failed.  Build files cannot be regenerated correctly.
WARNING: /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/libg2o/2020.5.29-1-r0/temp/run.do_configure.5736:1 exit 1 from 'cmake -G 'Ninja' -DCMAKE_MAKE_PROGRAM=ninja $oecmake_sitefile /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/libg2o/2020.5.29-1-r0/git -DCMAKE_INSTALL_PREFIX:PATH=/usr -DCMAKE_INSTALL_BINDIR:PATH=bin -DCMAKE_INSTALL_SBINDIR:PATH=sbin -DCMAKE_INSTALL_LIBEXECDIR:PATH=libexec -DCMAKE_INSTALL_SYSCONFDIR:PATH=/etc -DCMAKE_INSTALL_SHAREDSTATEDIR:PATH=../com -DCMAKE_INSTALL_LOCALSTATEDIR:PATH=/var -DCMAKE_INSTALL_LIBDIR:PATH=lib -DCMAKE_INSTALL_INCLUDEDIR:PATH=include -DCMAKE_INSTALL_DATAROOTDIR:PATH=share -DPYTHON_EXECUTABLE:PATH=/home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/libg2o/2020.5.29-1-r0/recipe-sysroot-native/usr/bin/python3-native/python3 -DPython_EXECUTABLE:PATH=/home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/libg2o/2020.5.29-1-r0/recipe-sysroot-native/usr/bin/python3-native/python3 -DPython3_EXECUTABLE:PATH=/home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/libg2o/2020.5.29-1-r0/recipe-sysroot-native/usr/bin/python3-native/python3 -DLIB_SUFFIX= -DCMAKE_INSTALL_SO_NO_EXE=0 -DCMAKE_TOOLCHAIN_FILE=/home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/libg2o/2020.5.29-1-r0/toolchain.cmake -DCMAKE_NO_SYSTEM_FROM_IMPORTED=1 -DBUILD_TESTING=OFF -DPYTHON_SOABI=cpython-38-aarch64-linux-gnu -Wno-dev'

ERROR: Logfile of failure stored in: /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/libg2o/2020.5.29-1-r0/temp/log.do_configure.5736
Log data follows:
```

shr-project, 2020-09-04:
Please always include at least "Build Configuration:" section shown at the beginning of the build and better describe the build configuration you've used.

libg2o isn't included in the default builds we're regularly testing (as defined by .mcf files and described in documentation). After adding opengl to DISTRO_FEATURES I was able to reproduce this issue and will add a fix for it.

Also do you really need all of packagegroup-ros-world-foxy in your image?

shr-project, 2020-09-08:
Fixed in milestone 14.

closed 2020-09-08 by shr-project

willofdiamond, 2020-09-17:
I can confirm that libg2o worked with  milestone 14.